espressif/esp-idf #6979: ESP32-C3 i2s_basic example cannot be compiled

zenghi, 2021-05-04:
The following error is displayed
```
/root/esp-c3/esp-idf/components/driver/i2s.c:1017: undefined reference to `rtc_clk_apll_enable'
/root/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/driver/libdriver.a(i2s.c.obj): in function `i2s_set_clk':
/root/esp-c3/esp-idf/components/driver/i2s.c:397: undefined reference to `rtc_clk_apll_enable'
/root/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/hal/libhal.a(i2s_hal.c.obj): in function `i2s_hal_init':
/root/esp-c3/esp-idf/components/hal/i2s_hal.c:273: undefined reference to `I2S0'
/root/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /root/esp-c3/esp-idf/components/hal/i2s_hal.c:273: undefined reference to `I2S0'
```

nicholasdkunes, 2021-06-03:
Hi @zenghi, were you ever able to resolve this? Running into the same issue...

L-KAYA, 2021-12-21:
For v4.4 and master branch, it should be able  to compile now. But for the earlier IDF version, I2S on ESP32-C3 is not supported.

closed 2021-12-21 by Alvin1Zhang