espressif/esp-idf #6980: Where can I found "Server-Sent Events" example in ESP-IDF git repo?

kunjanrshah, 2021-05-04:
Where can I found "Server-Sent Events" example in ESP-IDF Git Repo? over the internet lot of examples using Arduino but I need with ESP-IDF Framework example. could you please help with this?

shubhamkulkarni97, 2021-07-16:
Hi @kunjanrshah, thanks for raising this request.

Could you please elaborate the requirement? Do you need support for server sent events for esp_http_client or esp_http_server?

Also, please share some pointers to the Arduino examples which use server sent events.

Alvin1Zhang, 2021-07-28:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates, thanks.

closed 2021-07-28 by Alvin1Zhang

aquiot-inc, 2023-06-01:
Server-Sent Events (SSE) is needed for the esp_http_server so that the ESP32 web server can notify a web app that data has changed etc.

SSE is supported in the Arduino project https://github.com/me-no-dev/ESPAsyncWebServer#async-event-source-plugin

An ESP-IDF implementation of this feature is desirable.