espressif/esp-idf #6982: Additional LWIP IPv6 Flag Configuration

Avi-Singh12, 2021-05-04:
**Is your feature request related to a problem? Please describe.**

I am testing the IPv6 capabilities provided in ESP-IDF v4.2.   The ESP32 is connecting to a WPA2 network that has DHCPv4 disabled, so no IPv4 address will be assigned.  I had to develop a solution single-stack IPv6 solution that can function without needing an IPv4 address.  The code below is an adaptation of the wpa2_enterprise example with an additional POST request from the esp_http_client example.  

**Describe the solution you'd like**

Make more LWIP flags configurable through `idf.py menuconfig` so that a single-stack IPv6 solution is possible. If this request is more suited for the `esp-lwip` repo, let me know and I can make the request there instead.

**Describe alternatives you've considered**

There were lwip flags that needed to be modified, but some were not accessible using `idf.py menuconfig`. 

- SLAAC had to be enabled (this was configurable through the `idf.py menuconfig` command at the path `Component config -> LWIP -> Enable IPV6 stateless address autoconfiguration`)
- `LWIP_IPV6` had to be set to 1 in [`esp-idf/components/lwip/lwip/src/include/lwip/opt.h`](https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/include/lwip/opt.h#L2413-L2417) to enable IPv6 capabilities (this is not configurable in `idf.py menuconfig`)
- `LWIP_IPV4` had to be set to 0 in [`esp-idf/components/lwip/lwip/src/include/lwip/opt.h`](https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/include/lwip/opt.h#L727-L731) to force the [DNS resolver to only resolve IPv6 addresses](https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/include/lwip/dns.h#L56-L69) (this is not configurable in `idf.py menuconfig`)
- `LWIP_ND6_RDNSS_MAX_DNS_SERVERS` had to be set to >0 in [`esp-idf/components/lwip/lwip/src/include/lwip/opt.h`](https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/include/lwip/opt.h#L2711-L2713) so that DNS servers can be fetched through Router Advertisements (this is not configurable in `idf.py menuconfig`)

**Additional context**

```
/* WiFi Connection Example using WPA2 Enterprise
 *
 * Original Copyright (C) 2006-2016, ARM Limited, All Rights Reserved, Apache 2.0 License.
 * Additions Copyright (C) Copyright 2015-2016 Espressif Systems (Shanghai) PTE LTD, Apache 2.0 License.
 *
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "esp_http_client.h"
#include "esp_tls.h"

#define EXAMPLE_WIFI_SSID "SSID"
#define EXAMPLE_EAP_ID "EAP_ID"
#define MAX_HTTP_OUTPUT_BUFFER 2048

static EventGroupHandle_t wifi_event_group;

static esp_netif_t *sta_netif = NULL;
const int CONNECTED_BIT = BIT0;
static const char *TAG = "ipv6-testing";

extern uint8_t client_crt_start[] asm("_binary_wpa2_client_crt_start");
extern uint8_t client_crt_end[] asm("_binary_wpa2_client_crt_end");
extern uint8_t client_key_start[] asm("_binary_wpa2_client_key_start");
extern uint8_t client_key_end[] asm("_binary_wpa2_client_key_end");
extern uint8_t ca_pem_start[] asm("_binary_wpa2_ca_pem_start");
extern uint8_t ca_pem_end[] asm("_binary_wpa2_ca_pem_end");

[... 109 lines cut ...]

            esp_err_t err = esp_http_client_perform(client);
            if (err == ESP_OK)
            {
                ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %d",
                         esp_http_client_get_status_code(client),
                         esp_http_client_get_content_length(client));
                ESP_LOGI(TAG, "Response: %s", local_response_buffer);
            }
            else
            {
                ESP_LOGE(TAG, "HTTP POST request failed: %s", esp_err_to_name(err));
            }
        }
        ESP_LOGI(TAG, "~~~~~~~~~~~");
    }
}

void app_main(void)
{
    ESP_ERROR_CHECK(nvs_flash_init());
    initialise_wifi();
    xTaskCreate(&wpa2_enterprise_example_task, "wpa2_enterprise_example_task", 16000, NULL, 5, NULL);
}
```

ESP-YJM, 2021-05-06:
@Avi-Singh12 To support single-stack IPv6, we maybe do some modifition in ESP-IDF. As you mention, need add some lwip flags to configure. 
* For `LWIP_IPV6 `, it is already enabled in `lwipopts.h`. So `esp-idf/components/lwip/lwip/src/include/lwip/opt.h` make it to 0 will make no sense. In the master and release/v4.4, we offer the menuconfig to config IPv6(Component config -> LWIP -> Enable IPV6).
* For `LWIP_IPV4 `, i think it is difficult to disable it. Although lwip uses good code habits, the functions of ipv4 are distinguished by using `LWIP_IPV4`. Component code in ESP-IDF regards the `LWIP_IPV4` always enabled, so lots of code not use `LWIP_IPV4` to distinguish. If disable `LWIP_IPV4`, will make code build fail. And it is a huge work to distinguish code from IPv4. I think it is easy to remove `ESP_NETIF_DHCP_CLIENT` flag in `ESP_NETIF_INHERENT_DEFAULT_WIFI_STA` to make no dhcp action to achieve IPv4 address.
* For `LWIP_ND6_RDNSS_MAX_DNS_SERVERS `, i agree with your point to enable it. But for LWIP, maybe the fault design in LWIP, it will cover IPv4 DNS server when this option enable. We will add other action to avoid it in future.

After this flag set, i think your code also need some modifition. Please follow this code and test.
```
+ else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP6)
{
+   ESP_LOGI(TAG, "Got IP6 Event");
+   esp_ip6_addr_t addr;
+   if (esp_netif_get_ip6_global(sta_netif, &addr) == ESP_OK) {
+       ESP_LOGI(TAG, "Got IP6 Global Event");
+       xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
+   }
-    xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
}

....
while (1)
    {
-        vTaskDelay(2000 / portTICK_PERIOD_MS);
+       xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);
```

Avi-Singh12, 2021-05-06:
I tried testing your recommend changes with `v4.4-dev-1254-g639e7ad49`.  
- Removing the `ESP_NETIF_DHCP_CLIENT` flag in `ESP_NETIF_INHERENT_DEFAULT_WIFI_STA` still does not solve the issue of the the [DNS resolver](https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/core/dns.c#L1599-L1600) not trying to resolve the host to an IPv4 address first.  Hardcoding (`dns_addrtype = LWIP_DNS_ADDRTYPE_IPV6;`) in the `dns_gethostbyname_addrtype()` function is how I tested this.  You say that disabling `LWIP_IPV4` will be require a big change, so is there any other way to control the default [DNS resolution address type](https://github.com/espressif/esp-lwip/blob/2c9c531f0a7e0ee536db9de4f9dc54e453712087/src/include/lwip/dns.h#L56-L69)?
- `IP_EVENT_STA_GOT_IP6` does not exist
- Blocking call to `xEventGroupWaitBits` never finished

ESP-YJM, 2021-05-07:
@Avi-Singh12 
* To solve the DNS resolve, you can call `dns_gethostbyname_addrtype(hostname, addr, found, callback_arg, LWIP_DNS_ADDRTYPE_IPV6)` or use getaddrinfo and make `ai_family = AF_INET6`. We can not modify the default DNS resolution address type. I think it is a good way to resolve the DNS in your own code. And transfere the IP address to our IDF code. 
* Sorry for that, I made a spelling error. The event is `IP_EVENT_GOT_IP6 `.
* Make sure that your device whether support achieve global IPv6 address.

Avi-Singh12, 2021-05-07:
I tried calling `dns_gethostbyname_addrtype(hostname, addr, found, callback_arg, LWIP_DNS_ADDRTYPE_IPV6)` to convert the hostname in the url to an IPv6 addr that can be used when creating the `esp_http_client_config_t `, but the `esp_http_client` tries to resolve the address again, which leads to issues.  Do you have pointers on how to fix this?

ESP-YJM, 2021-05-08:
@Avi-Singh12 I think you can follow this step. First, resolve url to IPv6 address. For example, if you want to get `https://test.com:443/inform`, you first need resolve `test.com` to IPv6 address as `2409:8C38:202:1B:2424:0:1234:5`, and then you need combine the URL to `https://[2409:8C38:202:1B:2424:0:1234:5]:443/inform`. Then you transfer this URL to `esp_http_client_config_t `, the `esp_http_client` will not solve this address again. Because this address is already not a domain address.

ESP-YJM, 2021-06-21:
@Avi-Singh12 Is there any update or comment on this issue? If no, please help close this issue. Thanks.

Avi-Singh12, 2021-06-21:
I was able to follow the instructions in your last comment to resolve a URL to an IPv6 address before setting the `esp_http_client_config_t`.  As discussed previously, the `LWIP_ND6_RDNSS_MAX_DNS_SERVERS` field still needed to be set in the `lwip/opt.h` file and this is not currently configurable using `idf.py menuconfig`.

ESP-YJM, 2021-06-22:
@Avi-Singh12 Ok, i will add Kconfig for `LWIP_ND6_RDNSS_MAX_DNS_SERVERS`. Thanks for your suggestion.

closed 2021-06-28 by espressif-bot