espressif/esp-idf #6983: Failed to resolve component 'mbedtls'.

cyberpsyche, 2021-05-05:
How to resolve component 'mbedtls' ? 
I'm a newbie on esp32. I tried to install esp-idf on my Mac. My idf.py build command output list as follows. Thanks! 

```
~/e/hello_world (master)> idf.py build
Executing action: all (aliases: build)
Running cmake in directory /Users/sam/esp/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /Users/sam/esp/hello_world"...
-- Found Git: /usr/bin/git (found version "2.24.3 (Apple Git-128)") 
-- Component directory /Users/sam/esp/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/sam/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/sam/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/sam/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /Users/sam/esp/esp-idf/tools/cmake/build.cmake:194 (message):
  Failed to resolve component 'mbedtls'.
Call Stack (most recent call first):
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:225 (__build_resolve_and_add_req)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:226 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:226 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:221 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/build.cmake:508 (__build_expand_requirements)
  /Users/sam/esp/esp-idf/tools/cmake/project.cmake:361 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/sam/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

igrr, 2021-05-05:
Hi @cyberpsyche, looks like something went wrong when installing esp-idf. The `components/mbedtls` directory should contain, among other files, but looking at the log,
```
Component directory /Users/sam/esp/esp-idf/components/mbedtls does not contain a CMakeLists.txt file. No component will be added
```
the file is missing. Please run `git status` in the esp-idf directory and see if some files have been deleted or modified. Then you can run `git reset --hard master` (assuming you are using the `master` branch) and `git submodule update --init`. Alternatively you can delete your copy of esp-idf and then clone the repository again.

cyberpsyche, 2021-05-06:
Thanks @igrr, your suggestion works for me.  :)  Besides the 'mbedtls' problem, when I recompile the hello_world example, it shows other unhandled mistakes.  I think it may related to the git problem.

As I re-pulled the git repo, the git status shows:

```
~/e/esp-idf (master) [1]> git pull
Already up to date.
``` 

But when I execute git status, it shows:

```
~/e/esp-idf (master) [1]> git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   components/bt/controller/lib_esp32 (modified content)
	modified:   components/bt/controller/lib_esp32c3_family (new commits)
	modified:   components/cbor/tinycbor (modified content)
	modified:   components/esp_wifi/lib (new commits)
	modified:   components/lwip/lwip (modified content)
	modified:   examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib (modified content)

no changes added to commit (use "git add" and/or "git commit -a")
```

As I have re run the 'git pull' many times, it seems all the contents have already downloaded. How can I get the listed modified content above re-pulled ? Thanks!

cyberpsyche, 2021-05-06:
Reply to my post by myself.  If somebody meet this kind of problems. One could remove the listed as 'modified' directory and re-run the commands :

closed 2021-05-06 by cyberpsyche