espressif/esp-idf #6985: Corrupted Bluetooth HID data on fast transmissions

demik, 2021-05-05:
## Environment

- Development Kit: none
- Module or chip used: 
- IDF version: v4.2.1
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description
Trying to use a Bluetooth Mouse with ESP32 as HID Host. Sometimes the HID data is corrupted. It only happens when there is "a lot" of data transmitted (example: when moving the mouse fast enough)

Issue was reproduced using an ESP32-CAM module (ESP-WROOM-32) and a custom ESP32-PICO-D4 board.
Issue was reproduced with multiple mouses (Apple Magic Mouse 2 (x2), HP Z5000)

### Expected Behavior
No data corruption

### Actual Behavior
A few data is corrupted

### Steps to reproduce
The example is easily reproduced using a Bluetooth mouse and USB example

1. use esp_hid_host example
2. pair mouse
3. move mouse fast enough

### Code to reproduce this issue

examples/bluetooth/esp_hid_host/*

## Debug Logs

Here is an example with only moving the mouse. The mouse is sending 5 bytes INPUT MOUSE REPORT packets, first byte is buttons status, last one is wheel scrolling. I'm neither clicking nor scrolling.

```
I (31) boot: ESP-IDF v4.2.1 2nd stage bootloader
I (31) boot: compile time 12:37:40
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x233a0 (144288) map
I (169) esp_image: segment 1: paddr=0x000333c8 vaddr=0x3ffbdb60 size=0x032cc ( 13004) load
I (175) esp_image: segment 2: paddr=0x0003669c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/demik/github/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (176) esp_image: segment 3: paddr=0x00036aa8 vaddr=0x40080404 size=0x09570 ( 38256) load
I (201) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x8e5d4 (583124) map
0x400d0020: _stext at ??:?

I (424) esp_image: segment 5: paddr=0x000ce5fc vaddr=0x40089974 size=0x0b984 ( 47492) load
0x40089974: ld_acl_frm_isr at ??:?

I (456) boot: Loaded app from partition at offset 0x10000
I (456) boot: Disabling RNG early entropy source...
I (457) cpu_start: Pro cpu up.
I (460) cpu_start: Application information:
[... 159 lines cut ...]
I (14596) ESP_HIDH_DEMO: 00 fb ef f9 00 
```
Good
```
I (14596) ESP_HIDH_DEMO: 98:fd:b4:7d:b3:98 INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
I (14606) ESP_HIDH_DEMO: 3f 28 d1 0e 40
```
Corrupted. Invalid click / scroll wheel / movements

## Other items if possible
When running esp_hid_host example with CONFIG_HEAP_POISONING_COMPREHENSIVE=y, the corrupted data is still there but the previously good data is replaced by "fe fe fe fe fe":

```
I (19628) ESP_HIDH_DEMO: 98:fd:b4:7d:b3:98 INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
I (19638) ESP_HIDH_DEMO: fe fe fe fe fe 
I (19908) ESP_HIDH_DEMO: 98:fd:b4:7d:b3:98 INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
I (19918) ESP_HIDH_DEMO: fe fe fe fe fe 
I (19928) ESP_HIDH_DEMO: 98:fd:b4:7d:b3:98 INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
I (19928) ESP_HIDH_DEMO: fe fe fe fe fe 
I (19938) ESP_HIDH_DEMO: 98:fd:b4:7d:b3:98 INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
I (19938) ESP_HIDH_DEMO: fe fe fe fe fe 
I (19948) ESP_HIDH_DEMO: 98:fd:b4:7d:b3:98 INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
I (19948) ESP_HIDH_DEMO: fe fe fe fe fe 
I (19958) ESP_HIDH_DEMO: 98:fd:b4:7d:b3:98 INPUT:    MOUSE, MAP:  0, ID:   2, Len: 5, Data:
```

arlenn, 2021-09-16:
I think this is similar to what I am seeing, trying to work with a clone PS4 controller, I suspect analog stick data blasting out values causing a stack overflow and crash. It doesn't always report the stack overflow, but always hangs after a few seconds

As the esp_hid_host_main test application launches it does report two warnings that are not seen when testing with a Wii controller which does not appear to broadcast its gyro data over HID.
`W (8965) BT_SDP: process_service_attr_rsp
W (9305) BT_SDP: process_service_attr_rsp`

`I (18325) ESP_HIDH_DEMO: 98:b6:e9:2f:6e:84 INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:
I (18335) ESP_HIDH_DEMO: 00 35 0c 00 00 00 00 00 00
I (18335) ESP_HIDH_DEMO: 98:b6:e9:2f:6e:84 INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:
I (18345) ESP_HIDH_DEMO: 00 b4 4b fd 3f 94 51 fd 3f
I (18355) ESP_HIDH_DEMO: 98:b6:e9:2f:6e:84 INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:
I (18365) ESP_HIDH_DEMO: 00 35 0c 00 00 00 00 00 00
I (18365) ESP_HIDH_DEMO: 98:b6:e9:2f:6e:84 INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:
I (18375) ESP_HIDH_DEMO: ff 9f 80 80 08 00 4c 00 00
I (18385) ESP_HIDH_DEMO: 98:b6:e9:2f:6e:84 INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:
I (18395) ESP_HIDH_DEMO: 00 04 49 fd 3f 24 51 fd 3f
I (18395) ESP_HIDH_DEMO: 98:b6:e9:2f:6e:84 INPUT:  GAMEPAD, MAP:  0, ID:   1, Len: 9, Data:
I (18405) ESP_HIDH_DEMO: 40 50 72 fd 3f 00 a4 00 00`

boblane1, 2021-09-16:
Hi @demik, @arlenn,

Please try the following patch to see if the issue still exists. Thanks!
[0001-fix-deep-copy-bugs-for-esp_hid-component.zip](https://github.com/espressif/esp-idf/files/7176142/0001-fix-deep-copy-bugs-for-esp_hid-component.zip)