espressif/esp-idf #6988: ESP32 using RMT for neopixel LED's

krupis, 2021-05-06:
Hello. I am learning how to write my own various drivers for IC's and chips. I have started developing a simple neopixel driver. I am fully aware that the drivers already exist, the purpose of me writing such driver is to learn more about programming. Please do not bother commenting "just use the existing libraries".  I have read the espressif documentation and found this git library:
https://github.com/espressif/esp-idf/tree/639e7ad/examples/peripherals/rmt/led_strip

I have read the .c file and cant really understand.  There must be a bunch of missing code that is not showing up in .main folder. There seems to be undefined functions such as led_strip_new_rmt_ws2812().. And also, I do not see how the timings are being handled. 

Can someone explain to me whether there is an issue with this code or I am just confused?

suda-morris, 2021-05-06:
The "missing" function is in another [component](https://github.com/espressif/esp-idf/tree/master/examples/common_components/led_strip). 
> I do not see how the timings are being handled.

LED timing is handled in this function: https://github.com/espressif/esp-idf/blob/master/examples/common_components/led_strip/led_strip_rmt_ws2812.c#L66

krupis, 2021-05-06:
Oh. Thanks! After reading through the source code  :
 https://github.com/espressif/esp-idf/blob/master/examples/common_components/led_strip/led_strip_rmt_ws2812.c , I still have some questions regarding the timing setup. 

Looking at the function : ws2812_rmt_adapter() , I can see that the following variables are used for timing : 
```
    const rmt_item32_t bit0 = {{{ ws2812_t0h_ticks, 1, ws2812_t0l_ticks, 0 }}}; //Logical 0
    const rmt_item32_t bit1 = {{{ ws2812_t1h_ticks, 1, ws2812_t1l_ticks, 0 }}}; //Logical 1
```

As far as I can see, these values are being configured in the other function  :  
```
led_strip_t *led_strip_new_rmt_ws2812(const led_strip_config_t *config)
{
    led_strip_t *ret = NULL;
    STRIP_CHECK(config, "configuration can't be null", err, NULL);

    // 24 bits per led
    uint32_t ws2812_size = sizeof(ws2812_t) + config->max_leds * 3;
    ws2812_t *ws2812 = calloc(1, ws2812_size);
    STRIP_CHECK(ws2812, "request memory for ws2812 failed", err, NULL);

    uint32_t counter_clk_hz = 0;
    STRIP_CHECK(rmt_get_counter_clock((rmt_channel_t)config->dev, &counter_clk_hz) == ESP_OK,
                "get rmt counter clock failed", err, NULL);
    // ns -> ticks
    float ratio = (float)counter_clk_hz / 1e9;
    ws2812_t0h_ticks = (uint32_t)(ratio * WS2812_T0H_NS);
    ws2812_t0l_ticks = (uint32_t)(ratio * WS2812_T0L_NS);
    ws2812_t1h_ticks = (uint32_t)(ratio * WS2812_T1H_NS);
    ws2812_t1l_ticks = (uint32_t)(ratio * WS2812_T1L_NS);

    // set ws2812 to rmt adapter
    rmt_translator_init((rmt_channel_t)config->dev, ws2812_rmt_adapter);

    ws2812->rmt_channel = (rmt_channel_t)config->dev;
    ws2812->strip_len = config->max_leds;

    ws2812->parent.set_pixel = ws2812_set_pixel;
    ws2812->parent.refresh = ws2812_refresh;
    ws2812->parent.clear = ws2812_clear;
    ws2812->parent.del = ws2812_del;

    return &ws2812->parent;
err:
    return ret;
}
```
Particularly, I am most interested in the following  : 

```
    uint32_t counter_clk_hz = 0;
    STRIP_CHECK(rmt_get_counter_clock((rmt_channel_t)config->dev, &counter_clk_hz) == ESP_OK,
                "get rmt counter clock failed", err, NULL);
    // ns -> ticks
    float ratio = (float)counter_clk_hz / 1e9;
    ws2812_t0h_ticks = (uint32_t)(ratio * WS2812_T0H_NS);
    ws2812_t0l_ticks = (uint32_t)(ratio * WS2812_T0L_NS);
    ws2812_t1h_ticks = (uint32_t)(ratio * WS2812_T1H_NS);
    ws2812_t1l_ticks = (uint32_t)(ratio * WS2812_T1L_NS);
```

I am not certain how is this ratio calculated and how the conversion is made between time in nanoseconds to ticks

suda-morris, 2021-05-06:
Because we defined WS2812 timings in nanoseconds, but in datasheets, it's usually in microseconds. So we need to divide this 1000 ratio when calculating the "rmt ticks". That explains why we `counter_clk_hz / 1e9` instead of `counter_clk_hz / 1e6`

krupis, 2021-05-06:
Thanks for the. But why all this extra code then? What is the point of :
```
 STRIP_CHECK(rmt_get_counter_clock((rmt_channel_t)config->dev, &counter_clk_hz) == ESP_OK,
```

For example, if my APB clock is set to 80MHZ, 1 tick will be 12,5 nanoseconds. So what is the issue defining the ticks as following:

#define WS2812_T0H_NS (400)
#define WS2812_T0L_NS (850)
#define WS2812_T1H_NS (800)
#define WS2812_T1L_NS (450)

    ws2812_t0h_ticks = WS2812_T0H_NS / 12.5;
    ws2812_t0l_ticks = WS2812_T0L_NS / 12.5;
    ws2812_t1h_ticks = WS2812_T0H_NS /12.5;
    ws2812_t1l_ticks = WS2812_T0H_NS / 12.5;

That should give me how many 12.5 nano second ticks it will need to generate required 400ns, 800ns and 850ns timings.

negativekelvin, 2021-05-06:
If you can guarantee for your use case that your clock will always be 80mhz then you can choose to hardcode the timing values.  But to be more flexible they should be derived and really there should be more code that checks if the timing requirements can be satisfied by the clock value.

krupis, 2021-05-06:
In my ws2812b program, I have replaced:

```
uint32_t counter_clk_hz = 0;

    // this emulates the rmt_get_counter_clock() function from ESP-IDF 3.4

    if (RMT_LL_HW_BASE->conf_ch[config.channel].conf1.ref_always_on == RMT_BASECLK_REF)
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = REF_CLK_FREQ / (div);
    }
    else
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = APB_CLK_FREQ / (div);
    }

    // NS to tick converter
    float ratio = (float)counter_clk_hz / 1e9;

    t0h_ticks = (uint32_t)(ratio * WS2812_T0H_NS);
    t0l_ticks = (uint32_t)(ratio * WS2812_T0L_NS);
    t1h_ticks = (uint32_t)(ratio * WS2812_T1H_NS);
    t1l_ticks = (uint32_t)(ratio * WS2812_T1L_NS);

```

With 

```
    t0h_ticks = (uint32_t)(WS2812_T0H_NS / 25);
    t0l_ticks = (uint32_t)(WS2812_T0L_NS / 25);
    t1h_ticks = (uint32_t)(WS2812_T1H_NS / 25);
    t1l_ticks = (uint32_t)(WS2812_T1L_NS / 25);
```

It actually turned out that I must divide by 25 instead of 12.5 which means that my APB clock is 80mhz.

I am not fully understanding how this code works:

```
 if (RMT_LL_HW_BASE->conf_ch[config.channel].conf1.ref_always_on == RMT_BASECLK_REF)
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = REF_CLK_FREQ / (div);
    }
    else
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = APB_CLK_FREQ / (div);
    }
```

Is there really no easier way to find APB clock frequency of my ESP32 device?
From a quick search on the internet I have found a following function which apparently should return APB clock 

rtc_clk_apb_freq_get();

Wouldnt it make more sense to use something like that: 

```
uint32_t current_apb_frequency = rtc_clk_apb_freq_get();

    float tick_ns = 1/current_apb_frequency;
    t0h_ticks = (uint32_t)(WS2812_T0H_NS / tick_ns );
    t0l_ticks = (uint32_t)(WS2812_T0L_NS / tick_ns );
    t1h_ticks = (uint32_t)(WS2812_T1H_NS / tick_ns );
    t1l_ticks = (uint32_t)(WS2812_T1L_NS / tick_ns );
```

Please do not get me wrong! I am not trying to be picky here or anything! I am just learning and trying to understand the code and what it does.

krupis, 2021-05-06:
I have did some further testing and digging and learnt some additional things. In my rmt config :  
```
rmt_config_t config = {
        .rmt_mode = RMT_MODE_TX,
        .channel = channel,
        .gpio_num = pin,
        .clk_div = 2,
        .mem_block_num = 1,
        .tx_config = {
            //.carrier_freq_hz = 38000,
            //.carrier_level = RMT_CARRIER_LEVEL_HIGH,
            .idle_level = RMT_IDLE_LEVEL_LOW,
            //.carrier_duty_percent = 33,
            .carrier_en = false,
            .loop_en = false,
            .idle_output_en = true,
        }};
    rmt_config(&config);
    rmt_driver_install(config.channel, 0, 0);
```

Note that clk_div is set to 2. Thats why I have to swamp 12.5 nanoseconds to 25 nanoseconds.

I want to understand this clk_div a little more. What are the benefits of using a fastest clock possible ( clk_div  = 1 ) apart from the obvious fact that it has more resolution. Are there any drawbacks? Why would I not want to use the full resolution

suda-morris, 2021-05-06:
No drawbacks I think if you can set the clk_div=1. But in this case (for led strip), clk_div=2 is enough.

krupis, 2021-05-06:
I see now. From what I understand a function can be used rmt_get_counter_clock();

If that is the case, can my code:
```
 if (RMT_LL_HW_BASE->conf_ch[config.channel].conf1.ref_always_on == RMT_BASECLK_REF)
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = REF_CLK_FREQ / (div);
    }
    else
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = APB_CLK_FREQ / (div);
    }
```
be replaced with  rmt_get_counter_clock(); ?

closed 2021-06-01 by Alvin1Zhang