espressif/esp-idf #6989: ESP32-C3 build fail

Rokachy, 2021-05-06:
esp-idf/esp-idf-v4.4/components/soc/include/soc/mcpwm_periph.h:16:10: fatal error: soc/mcpwm_reg.h: No such file or directory
 #include "soc/mcpwm_reg.h"
          ^~~~~~~~~~~~~~~~~

The example of mcpwm_brushed_dc_control can be compiled on ESP32 but fail on ESP32C3 see above error.

Yehuda

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

suda-morris, 2021-05-06:
Yes, ESP32-C3 doesn't have an `MCPWM` peripheral. And in that example's README, we have mentioned that supported target is only `esp32` for now.

closed 2021-05-06 by Alvin1Zhang

Hezhexi2002, 2021-07-24:
> Yes, ESP32-C3 doesn't have an `MCPWM` peripheral. And in that example's README, we have mentioned that supported target is only `esp32` for now.

so it means the ESP32-S2 isn't be supported yet?

suda-morris, 2021-10-11:
MCPWM hardware only exists on esp32 and esp32-s3 up to now. 
> i am using ESP32-S2 and need PWM to control servo motor. is this possible?

have you tried RMT or LEDC peripheral, both of them can generate a PWM wave.