espressif/esp-idf #6990: Internal_communication example without router needed

L30n4rd0-code, 2021-05-06:
I'm testing the `Internal_communication` example to create a ESP-MESH network and I see at the serial monitor that the nodes are always trying to connect to a router with the SSID and the password I configured, however, the network I intend to build shouldn't have a router in it, just nodes that communicate between them. I think this is possible, but I'm far from knowing how to make it.

So, I would like to know if there's a example to create a ESP-MESH network without needing a router to communicate. Because in my research I haven't something alike.

After creating the ESP-MESH network, I have to manage to make that this nodes work with WiFi and with BLE, not simultaneously, but periodically with each communication protocol. Is this possible? And if so, could I have some guidance of some examples or docs that talk about it?

shenjun7, 2021-05-08:
You can use esp_mesh_fix_root(true) to establish ESP-Mesh without a router, and make sure a root node is designated for the network by esp_mesh_set_type(MESH_ROOT) or esp_mesh_set_parent().

L30n4rd0-code, 2021-05-10:
Thanks @shenjun7, that's exactly what I did last friday. I had just some problems because I was designating all nodes as root nodes, after changing that, everything works as desired.

shenjun7, 2021-05-11:
The esp_mesh_fix_root(true) should be just included in the main program for all nodes.

closed 2021-06-18 by Alvin1Zhang