espressif/esp-idf #6991: Cannot use rtc_clk_apb_freq_get() function in esp-idf project

krupis, 2021-05-06:
Hello. I have setup esp-idf project using vscode for programming ESP32 devices. I am working on neopixel driver which requires me to know APB clock frequecncy. After doing a quick internet search, I have found that I can find APB clock by calling a function rtc_clk_apb_freq_get();

I have included a header file 
#include "driver/rmt.h"

I am trying the following code:

[code]
/* Hello World Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

#include "driver/rmt.h"

void app_main(void)
{
    printf("Hello world!\n");
    uint32_t apb_freq;
    apb_freq = rtc_clk_apb_freq_get();
}
[/code]

The error I am getting:
[code]
../main/hello_world_main.c:21:16: error: implicit declaration of function 'rtc_clk_apb_freq_get' [-Werror=implicit-function-declaration] 
     apb_freq = rtc_clk_apb_freq_get();
                ^~~~~~~~~~~~~~~~~~~~
../main/hello_world_main.c:20:14: warning: variable 'apb_freq' set but not used [-Wunused-but-set-variable]
     uint32_t apb_freq;
              ^~~~~~~~
[/code]

Which suggests that the function is not recognised.

Can someone suggest me what is the issue and how to solve it? Are there any other methods of finding out your APB clock frequency?

I am writing my driver code in platformio, but I am not sure whether this command is only available for esp-idf. 
Any advice is appreciated.

negativekelvin, 2021-05-06:
`#include "soc/rtc.h"`

Eventually you are going to end up back at rmt_get_counter_clock

Please remember to close your issues when answered

krupis, 2021-05-06:
Okay I see now. It is starting to make more sense. In my program, I have the following:

```
    uint32_t counter_clk_hz = 0;

    // this emulates the rmt_get_counter_clock() function from ESP-IDF 3.4
    if (RMT_LL_HW_BASE->conf_ch[config.channel].conf1.ref_always_on == RMT_BASECLK_REF)
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = REF_CLK_FREQ / (div);
    }
    else
    {
        uint32_t div_cnt = RMT_LL_HW_BASE->conf_ch[config.channel].conf0.div_cnt;
        uint32_t div = div_cnt == 0 ? 256 : div_cnt;
        counter_clk_hz = APB_CLK_FREQ / (div);
    }
```

Can this code be further simplified by using rmt_get_counter_clock() ?

suda-morris, 2022-05-16:
Yes, using `rmt_get_counter_clock` can solve the problem.

closed 2022-05-16 by suda-morris