espressif/esp-idf #6992: Couldn't connect to wifi using esp-idf above 4.0

ErwinTheRabbit, 2021-05-06:
## Environment

Module or chip used: ESP32-D0WDQ6 or ESP32-D0WD
Any IDF after 4.0 (4.0 is the latest working SDK for this module)
Current testing is v4.4-dev-1254-g639e7ad49

## Problem Description

Can't connect to WiFi or Bluetooth

### Steps to reproduce

Just build getting-started/station example. It won't work.
I have tried a lot of sets of configurations, but even on default it doesn't work. 

## Debug Logs

```
I (0) cpu_start: App cpu up.
I (1095) cpu_start: Pro cpu start user code
I (1095) cpu_start: cpu freq: 160000000
I (1095) cpu_start: Application information:
I (1099) cpu_start: Project name:     fr_esp32
I (1104) cpu_start: App version:      1
I (1109) cpu_start: Compile time:     May  6 2021 12:31:04
I (1115) cpu_start: ELF file SHA256:  0135a9deb691125f...
I (1121) cpu_start: ESP-IDF:          v4.4-dev-1254-g639e7ad49
I (1128) heap_init: Initializing. RAM available for dynamic allocation:
I (1135) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1141) heap_init: At 3FFB7628 len 000289D8 (162 KiB): DRAM
I (1147) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1154) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1160) heap_init: At 400947E0 len 0000B820 (46 KiB): IRAM
I (1168) spi_flash: detected chip: mxic
I (1171) spi_flash: flash io: dio
I (1176) sleep: Configure to isolate all GPIO pins in sleep state
I (1182) sleep: Enable automatic switching of GPIO sleep configuration
I (1190) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1220) wifi station: ESP_WIFI_MODE_STA
I (1240) wifi:wifi driver task: 3ffb8fd8, prio:23, stack:6656, core=0
I (1240) system_api: Base MAC address is not set
I (1240) system_api: read default base MAC address from EFUSE
I (1250) wifi:wifi firmware version: 44799ff
I (1250) wifi:wifi certification version: v7.0
I (1250) wifi:config NVS flash: enabled
I (1250) wifi:config nano formating: disabled
I (1260) wifi:Init data frame dynamic rx buffer num: 32
I (1260) wifi:Init management frame dynamic rx buffer num: 32
I (1270) wifi:Init management short buffer num: 32
I (1270) wifi:Init dynamic tx buffer num: 32
I (1280) wifi:Init static rx buffer size: 1600
I (1280) wifi:Init static rx buffer num: 10
I (1290) wifi:Init dynamic rx buffer num: 32
I (1290) wifi_init: rx ba win: 6
I (1290) wifi_init: tcpip mbox: 32
I (1300) wifi_init: udp mbox: 6
I (1300) wifi_init: tcp mbox: 6
I (1300) wifi_init: tcp tx win: 5744
I (1310) wifi_init: tcp rx win: 5744
I (1310) wifi_init: tcp mss: 1440
I (1320) wifi_init: WiFi IRAM OP enabled
I (1320) wifi_init: WiFi RX IRAM OP enabled
I (1350) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (1350) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1510) wifi:mode : sta (30:ae:a4:8a:53:90)
I (1510) wifi:enable tsf
I (1510) wifi station: wifi_init_sta finished.
I (1640) wifi:new:<2,0>, old:<1,0>, ap:<255,255>, sta:<2,0>, prof:1
I (2390) wifi:state: init -> auth (b0)
[... 98 lines cut ...]
I (47710) wifi station: connect to the AP fail
I (49760) wifi station: retry to connect to the AP
I (49760) wifi station: connect to the AP fail
I (49760) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (49770) wifi:state: init -> auth (b0)
I (49810) wifi:state: auth -> assoc (0)
I (49850) wifi:state: assoc -> run (10)
I (52960) wifi:state: run -> init (fc0)
I (52960) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (52960) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (52970) wifi station: retry to connect to the AP
I (52970) wifi station: connect to the AP fail
I (55020) wifi station: retry to connect to the AP
I (55020) wifi station: connect to the AP fail
I (55030) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (55030) wifi:state: init -> auth (b0)
I (55070) wifi:state: auth -> assoc (0)
I (55110) wifi:state: assoc -> run (10)
I (58220) wifi:state: run -> init (fc0)
I (58220) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (58220) wifi:new:<2,0>, old:<2,0>, ap:<255,255>, sta:<2,0>, prof:1
I (58230) wifi station: retry to connect to the AP
I (58230) wifi station: connect to the AP fail
I (60280) wifi station: connect to the AP fail
```

ErwinTheRabbit, 2021-05-06:
Here is a part of schematics:
[image]

ErwinTheRabbit, 2021-05-08:
Here is full log:
````
I (1107) cpu_start: Pro cpu up.
D (1107) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (1107) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (1113) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (1120) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (1127) cpu_start: Starting app cpu, entry point is 0x400810c4
I (0) cpu_start: App cpu up.
D (1142) clk: RTC_SLOW_CLK calibration value: 3571968
I (1154) cpu_start: Pro cpu start user code
I (1155) cpu_start: cpu freq: 160000000
I (1155) cpu_start: Application information:
I (1159) cpu_start: Project name:     fr_esp32
I (1164) cpu_start: App version:      1
I (1169) cpu_start: Compile time:     May  8 2021 15:24:00
I (1175) cpu_start: ELF file SHA256:  63b4a7f41a7d1b7b...
I (1181) cpu_start: ESP-IDF:          v4.4-dev-1254-g639e7ad49
D (1188) memory_layout: Checking 7 reserved memory ranges:
D (1193) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (1200) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb7688
D (1206) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (1213) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (1219) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (1226) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (1232) memory_layout: Reserved memory range 0x40080000 - 0x40094840
D (1239) memory_layout: Building list of available memory regions:
D (1245) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (1252) memory_layout: Available memory region 0x3ffb7688 - 0x3ffb8000
D (1258) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
D (1265) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
D (1272) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
D (1278) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
D (1285) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (1292) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (1298) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (1305) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (1312) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (1319) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (1325) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (1332) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (1339) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (1345) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (1352) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (1359) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (1365) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (1372) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe3f20
D (1379) memory_layout: Available memory region 0x3ffe4350 - 0x3ffe8000
D (1385) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (1392) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (1399) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (1405) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (1412) memory_layout: Available memory region 0x40094840 - 0x40096000
D (1419) memory_layout: Available memory region 0x40096000 - 0x40098000
D (1425) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (1432) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (1439) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (1446) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (1452) heap_init: Initializing. RAM available for dynamic allocation:
D (1460) heap_init: New heap initialised at 0x3ffae6e0
I (1465) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (1471) heap_init: New heap initialised at 0x3ffb7688
I (1476) heap_init: At 3FFB7688 len 00028978 (162 KiB): DRAM
I (1482) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1489) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (1495) heap_init: New heap initialised at 0x40094840
I (1501) heap_init: At 40094840 len 0000B7C0 (45 KiB): IRAM
D (1507) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (1513) FLASH_HAL: extra_dummy: 1
D (1516) spi_flash: trying chip: issi
[... 688 lines cut ...]
D (22221) wifi:handoff_cb: status=0
D (22231) wifi:clear blacklist
D (22231) wifi:clear rc list
D (22231) wifi:clear blacklist
D (22231) wifi:send disconnect event
D (22241) wifi:connect status 1 -> 3
D (22241) wifi:disable connect timer
D (22241) wifi:clear scan ap list
D (22251) event: running post WIFI_EVENT:5 with handler 0x400ee398 and context 0x3ffbe560 on loop 0x3ffbd32c

D (22261) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbe258 from event_id=5
D (22271) esp_netif_lwip: check: remote, if=0x3ffbe258 fn=0x400ed1ec

D (22271) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbe258
D (22281) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbe258
D (22291) esp_netif_lwip: if0x3ffbe258 start ip lost tmr: no need start because netif=0x3ffbe2d8 interval=120 ip=0
D (22301) esp_netif_lwip: check: local, if=0x3ffbe258 fn=0x400ed8fc

D (22301) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbe258
D (22311) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (22311) event: running post WIFI_EVENT:5 with handler 0x400d3de4 and context 0x3ffbee44 on loop 0x3ffbd32c

I (22321) wifi station: connect to the AP fail
I (22331) wifi station: Failed to connect
````

negativekelvin, 2021-05-08:
Do you have a full log from a working 4.0 version?  Do you have any dev kits or only your custom hardware?  If so do the dev kits work?  Is it only with a specific router or any router?  Have you validated your antenna design and matching?

ErwinTheRabbit, 2021-05-11:
I hardly can reproduced it on my only Olimex esp32-gateway with ESP-WROOM-32 with some old routers, but in general it looks fine.
About my hardware: with some routers/access points it works fine, with some not. Some devices connect better, some worse. With IDF v4.0 every my device works fine, but with newer it looks like random number generator. Antenna design was copied from dev kit.
v4.0 log:
```
I (267) cpu_start: Pro cpu up.
I (267) cpu_start: Application information:
I (267) cpu_start: Project name:     fr_esp32
I (270) cpu_start: App version:      1
I (274) cpu_start: Compile time:     May 11 2021 11:12:39
I (280) cpu_start: ELF file SHA256:  c713b18fc051d074...
I (286) cpu_start: ESP-IDF:          v4.0-dirty
I (291) cpu_start: Single core mode
D (296) memory_layout: Checking 5 reserved memory ranges:
D (301) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (307) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffc5a40
D (314) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (320) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (327) memory_layout: Reserved memory range 0x40080000 - 0x40093ea8
D (333) memory_layout: Building list of available memory regions:
D (339) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (346) memory_layout: Available memory region 0x3ffb0000 - 0x3ffb8000
D (352) memory_layout: Available memory region 0x3ffb8000 - 0x3ffbdb60
D (359) memory_layout: Available memory region 0x3ffc5a40 - 0x3ffc6000
D (366) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (372) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (379) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (385) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (392) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (399) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (405) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (412) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (418) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (425) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (432) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (438) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (445) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (451) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe4000
D (458) memory_layout: Available memory region 0x3ffe4000 - 0x3ffe8000
D (465) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (471) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (478) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (484) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (491) memory_layout: Available memory region 0x40078000 - 0x40080000
D (498) memory_layout: Available memory region 0x40093ea8 - 0x40094000
D (504) memory_layout: Available memory region 0x40094000 - 0x40096000
D (511) memory_layout: Available memory region 0x40096000 - 0x40098000
D (517) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (524) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (531) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (537) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (544) heap_init: Initializing. RAM available for dynamic allocation:
D (551) heap_init: New heap initialised at 0x3ffae6e0
I (556) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
D (562) heap_init: New heap initialised at 0x3ffc5a40
I (567) heap_init: At 3FFC5A40 len 0001A5C0 (105 KiB): DRAM
I (574) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
D (580) heap_init: New heap initialised at 0x40078000
I (585) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
D (591) heap_init: New heap initialised at 0x40093ea8
I (596) heap_init: At 40093EA8 len 0000C158 (48 KiB): IRAM
I (603) cpu_start: Pro cpu start user code
D (608) clk: waiting for 32k oscillator to start up
D (918) clk: RTC_SLOW_CLK calibration value: 15995668
D (924) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (925) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (926) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (930) cpu_start: Starting scheduler on PRO CPU.
D (935) heap_init: New heap initialised at 0x3ffe0440
D (939) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (945) partition: Loading the partition table
[... 2049 lines cut ...]
D (30372) wifi:connect status 5 -> 4
D (30372) wifi:sta leave
I (30372) wifi:pm stop, total sleep time: 6054532 us / 7461821 us

D (30382) wifi:stop CSA timer
D (30382) wifi:remove 30:a2:c2:ce:03:0a from rc list
I (30392) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
D (30392) wifi:filter: set rx policy=8
D (30402) wifi:Send disconnect event, reason=6, AP number=0
D (30402) event: running post WIFI_EVENT:5 with handler 0x400ee38c and context 0x3ffbe580 on loop 0x3ffbd34c
D (30412) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbe278 from event_id=5
D (30422) esp_netif_lwip: check: remote, if=0x3ffbe278 fn=0x400ed1e0

D (30432) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbe278
D (30432) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbe278
D (30442) esp_netif_lwip: if0x3ffbe278 start ip lost tmr: no need start because netif=0x3ffbe2f8 interval=120 ip=0
D (30452) esp_netif_lwip: check: local, if=0x3ffbe278 fn=0x400ed8f0

D (30462) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbe278
D (30462) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (30472) event: running post WIFI_EVENT:5 with handler 0x400d3de4 and context 0x3ffbee64 on loop 0x3ffbd34c
I (30482) wifi station: connect to the AP fail
I (30482) wifi station: Failed to connect

```

ErwinTheRabbit, 2021-05-12:
I just found a particular commit after which my device's wifi stops working. It's https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96 .
21 Dec 2019 9432ebd v4.1-dev-1635-g9432ebddf-dirty log:
```
I (306) cpu_start: Pro cpu up.
I (306) cpu_start: Application information:
I (306) cpu_start: Project name:     fr_esp32
I (308) cpu_start: App version:      1
I (313) cpu_start: Compile time:     May 12 2021 12:01:03
I (319) cpu_start: ELF file SHA256:  87343099b8f9eb9a...
I (325) cpu_start: ESP-IDF:          v4.1-dev-1635-g9432ebddf-dirty
I (332) cpu_start: Single core mode
D (336) memory_layout: Checking 5 reserved memory ranges:
D (341) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (348) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffc5bd8
D (354) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (360) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (367) memory_layout: Reserved memory range 0x40080000 - 0x40093fc8
D (373) memory_layout: Building list of available memory regions:
D (380) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (386) memory_layout: Available memory region 0x3ffb0000 - 0x3ffb8000
D (393) memory_layout: Available memory region 0x3ffb8000 - 0x3ffbdb60
D (399) memory_layout: Available memory region 0x3ffc5bd8 - 0x3ffc6000
D (406) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (412) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (419) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
D (426) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
D (432) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (439) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (445) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
D (452) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
D (459) memory_layout: Available memory region 0x3ffd6000 - 0x3ffd8000
D (465) memory_layout: Available memory region 0x3ffd8000 - 0x3ffda000
D (472) memory_layout: Available memory region 0x3ffda000 - 0x3ffdc000
D (478) memory_layout: Available memory region 0x3ffdc000 - 0x3ffde000
D (485) memory_layout: Available memory region 0x3ffde000 - 0x3ffe0000
D (492) memory_layout: Available memory region 0x3ffe0440 - 0x3ffe4000
D (498) memory_layout: Available memory region 0x3ffe4000 - 0x3ffe8000
D (505) memory_layout: Available memory region 0x3ffe8000 - 0x3fff0000
D (511) memory_layout: Available memory region 0x3fff0000 - 0x3fff8000
D (518) memory_layout: Available memory region 0x3fff8000 - 0x3fffc000
D (525) memory_layout: Available memory region 0x3fffc000 - 0x40000000
D (531) memory_layout: Available memory region 0x40078000 - 0x40080000
D (538) memory_layout: Available memory region 0x40093fc8 - 0x40094000
D (544) memory_layout: Available memory region 0x40094000 - 0x40096000
D (551) memory_layout: Available memory region 0x40096000 - 0x40098000
D (558) memory_layout: Available memory region 0x40098000 - 0x4009a000
D (564) memory_layout: Available memory region 0x4009a000 - 0x4009c000
D (571) memory_layout: Available memory region 0x4009c000 - 0x4009e000
D (577) memory_layout: Available memory region 0x4009e000 - 0x400a0000
I (584) heap_init: Initializing. RAM available for dynamic allocation:
D (591) heap_init: New heap initialised at 0x3ffae6e0
I (596) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
D (602) heap_init: New heap initialised at 0x3ffc5bd8
I (608) heap_init: At 3FFC5BD8 len 0001A428 (105 KiB): DRAM
I (614) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
D (620) heap_init: New heap initialised at 0x40078000
I (625) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
D (632) heap_init: New heap initialised at 0x40093fc8
I (637) heap_init: At 40093FC8 len 0000C038 (48 KiB): IRAM
I (643) cpu_start: Pro cpu start user code
D (648) clk: waiting for 32k oscillator to start up
W (1051) clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator
D (1085) clk: RTC_SLOW_CLK calibration value: 3529470
D (1091) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (1092) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (1092) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (1097) cpu_start: Starting scheduler on PRO CPU.
D (1103) heap_init: New heap initialised at 0x3ffe0440
D (1107) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (1113) partition: Loading the partition table
[... 1089 lines cut ...]

I (27620) wifi: find the 50:d2:f5:b9:17:60 in blacklist.

I (28746) wifi: find the 50:d2:f5:b9:17:60 in blacklist.

I (29645) wifi: filter: set rx policy=4
I (29645) wifi: handoff_cb: status=0
I (29645) wifi: clear blacklist
I (29646) wifi: clear rc list
I (29646) wifi: clear blacklist
I (29649) wifi: send disconnect event
I (29653) wifi: connect status 1 -> 3
I (29656) wifi: disable connect timer
D (29660) event: running post WIFI_EVENT:5 with handler 0x40137554 on loop 0x3ffb4330
D (29667) esp_netif_handlers: esp_netif action disconnected with netif0x3ffb5178 from event_id=5
D (29676) esp_netif_lwip: check: remote, if=0x3ffb5178 fn=0x400d2f4c

D (29683) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffb5178
D (29689) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffb5178
D (29696) esp_netif_lwip: if0x3ffb5178 start ip lost tmr: no need start because netif=0x3ffb51f8 interval=120 ip=0
D (29706) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (29712) event: running post WIFI_EVENT:5 with handler 0x40118c8c on loop 0x3ffb4330
I (29720) wifi station: connect to the AP fail
I (29725) wifi station: Failed to connect to SSID:ErwinAP
```

AxelLin, 2021-05-12:
> I just found a particular commit after which my device's wifi stops working. It's [1e47b01](https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96) .

To esp-idf support team:
Is it possible to improve the commit messages?
I understand it's closed source library so you don't need to tell implemenation detail.
But please at least let people know the **user impact**, e.g. generally why the commit improves wifi performance.
In additional, since it's regarding performance, it would be great if you add some performance numbers.
Otherwise, the commit "improve wifi performance" really means nothing to people out-side of espressif.

oneineight, 2021-09-22:
Any news on this?

I get the same problem on my TTGO T-WATCH

ErwinTheRabbit, 2021-09-22:
I checked 1 month ago - same situation.

AxelLin, 2021-11-09:
> I just found a particular commit after which my device's wifi stops working. It's [1e47b01](https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96) . 21 Dec 2019 [9432ebd](https://github.com/espressif/esp-idf/commit/9432ebddf919228244e3c7064c4fc15b50a98bb8) v4.1-dev-1635-g9432ebddf-dirty log:

@Alvin1Zhang 
The bad commit is in closed source library (esp_wifi lib)
Anyone from espressif can help to check this?

AxelLin, 2021-11-09:
@ErwinTheRabbit 
Any chance to provide a working log for comparison? (i.e. the log before commit 1e47b01)

ErwinTheRabbit, 2021-11-09:
> @ErwinTheRabbit Any chance to provide a working log for comparison? (i.e. the log before commit [1e47b01](https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96))

Here it is, two logs with 1 day difference between esp-idf commits: https://github.com/espressif/esp-idf/issues/6992#issuecomment-839620246
Or you want something else?

AxelLin, 2021-11-09:
Hi @ErwinTheRabbit 

According to your log: https://github.com/espressif/esp-idf/issues/6992#issuecomment-839620246
failure version: ESP-IDF: v4.1-dev-1633-g1e47b0156-dirty
working version: ESP-IDF: v4.1-dev-1635-g9432ebddf-dirty

i.e. After commit 1e47b01, it's working.

So I'm confused, it's look like commit 1e47b01 fixed the issue rather than commit 1e47b01 causes the issue.
Did I misunderstand something?

ErwinTheRabbit, 2021-11-09:
working version: ESP-IDF: v4.1-dev-1635-g9432ebddf-dirty (21 Dec 2019 https://github.com/espressif/esp-idf/commit/9432ebddf919228244e3c7064c4fc15b50a98bb8 )
failure version: ESP-IDF: v4.1-dev-1633-g1e47b0156-dirty (22 Dec 2019 https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96 )

You mixed up dates. Working one was before failure one.

AxelLin, 2021-11-10:
> You mixed up dates. Working one was before failure one.

Ah, I see. I thought v4.1-dev-1635 is newer than v4.1-dev-1633 but it's not true.

commit 1e47b015633cc5cc66e52798399ee11fd9351f96 **(bad)**
Author: Jack <jack@espressif.com>
Date:   Sun Dec 22 19:34:53 2019 +0800

    improve wifi performance

commit 9432ebddf919228244e3c7064c4fc15b50a98bb8 **(good)**
Author: KonstantinKondrashov <konstantin@espressif.com>
Date:   Thu Dec 19 21:59:39 2019 +0800

    esp32: Add UT for DPORT

AxelLin, 2021-11-10:
Add @jack0c

sagb2015, 2021-11-11:
@ErwinTheRabbit  Is it possible to get the sniffer captures? Most likely this is rate related interop issue. It will also help to know exact model of AP used for the failure case.

AxelLin, 2021-11-11:
@sagb2015 
If it is rate related interop issue, why it always works when build with IDF-v4.0?

mhdong, 2021-11-11:
hi @ErwinTheRabbit could you provide the settings of the router about pmf. 
D (1901) nvs: nvs_get ap.pmf_e 1
D (1901) nvs: nvs_get ap.pmf_r 1
As can be seen from log, sta Protected Management Frames (PMF) is required。Depending on what AP side PMF Mode is, the resulting connnection will behave differently.

sagb2015, 2021-11-11:
> If it is rate related interop issue, why it always works when build with IDF-v4.0?

Hi @AxelLin, it is because there are rate related changes in the commit that is mentioned as bad ( https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96). I can see 2nd msg from STA (first data packet after assoc) is not going through in failure case.

AxelLin, 2021-11-11:
> As can be seen from log, sta Protected Management Frames (PMF) is required。Depending on what AP side PMF Mode is, the resulting connnection will behave differently.

@mhdong @sagb2015 

A question regarding the usage of pmf_cfg capable/required flags.
Normal users know nothing about the pmf_cfg capable/required flags,
they just set SSID and password and expect device to connect an AP.

So in what kind of case I should set pmf_cfg capable/required flags for my esp32 device (as STA)?
What happen if pmf_cfg capable/required flags are set wrongly?
I'm confused how do you expect esp32 developer to set these capable/required flags because normal users won't config these flag settings.

AxelLin, 2021-11-11:
> > If it is rate related interop issue, why it always works when build with IDF-v4.0?
> 
> Hi @AxelLin, it is because there are rate related changes in the commit that is mentioned as bad ( [1e47b01](https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96)). I can see 2nd msg from STA during 4-way handshake(first data packet after assoc) is not going through in failure case.

That commit just never mention that change. Please consider improving the commit log in the future.
https://github.com/espressif/esp-idf/issues/6992#issuecomment-839637270

sagb2015, 2021-11-11:
> @mhdong @sagb2015
>
> A question regarding the usage of pmf_cfg capable/required flags. Normal users know nothing about the pmf_cfg capable/required flags, they just set SSID and password and expect [...]

Hi @AxelLin, please refer to the documentation [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi-security.html#protected-management-frames-pmf). By default it is recommended to keep the device PMF `capable` but not `required`. This will allow it to connect to non-PMF SSID as well if AP doesn't support PMF, and it will always connect using PMF if AP supports it.

sagb2015, 2021-12-13:
@ErwinTheRabbit Any update on the sniffer capture?

ErwinTheRabbit, 2021-12-14:
@sagb2015 I'm collecting more information from some devices right now because it looks like esp32's behavor differs a lot if i use different routers and SDK versions. I'll post result here in about 2 days.

AxelLin, 2021-12-27:
@ErwinTheRabbit 
Any update?

ErwinTheRabbit, 2021-12-28:
@AxelLin Sad, nothing interesting enough.
Just a bit of information that applicable to all devices and any IDF after 4.1:
Device could receive packets, so PHY's RX works.
Some (random) seconds after scan PHY's TX works terrible (looks like very weak).
After this period TX works fine.
All packets that could be gathered by sniffer (WiFi module in sniffing mode) look fine.

Situation 1:
Router with no encryption, device with static IP (no DHCP), so no TX data required for connection.
Associate well, after association device can receive any packet from router.
However device can't send any packet within random time (seconds)
After this period everything is OK

Situation 2:
Router with encryption (or router without encryption and with DHCP)
Usually (99% chance) can't associate (or send DHCP request), router doesn't receive packets from device.
If association was ok everything works fine.

I tried to somehow measure TX signal level, time after start, add delays, disassemble old libphy.bin and try to find differences etc, but i still can't find out any dependence between environment or wifi parameters of opensource code part and "TX disable time".

ErwinTheRabbit, 2021-12-28:
Right now i'm digging into disassembling differences in this commit: https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96 Looks like there is a lot of changes, so it's not that simple. I hope i could find solution simple enough to implement it into new IDF.

AxelLin, 2021-12-31:
@ErwinTheRabbit
Could you provide sniffer capture for espressif support team to check?
BTW, what is the brand/model of the router that can reproduce the issue?
Can you also check if recent esp-idf update works or not which includes some wifi bug fixes?

AxelLin, 2022-01-14:
> Right now i'm digging into disassembling differences in this commit: [1e47b01](https://github.com/espressif/esp-idf/commit/1e47b015633cc5cc66e52798399ee11fd9351f96) Looks like there is a lot of changes, so it's not that simple. I hope i could find solution simple enough to implement it into new IDF.

IMHO, had better let espressif's support team to handle this rather than disassembling closed source binary by yourself.

AxelLin, 2022-02-28:
@ErwinTheRabbit 
How is the status of this issue now?
If you still need support from espressif, you need to provide sniffer log for analysis.

AxelLin, 2022-05-10:
@ErwinTheRabbit 

Any update?
I'm wondering if you want to dig this issue by yourself (and close the issue).
Or you can provide sniffer capture for the support team to help on this issue.

AxelLin, 2022-12-18:
@ErwinTheRabbit 
Do you still hit the issue with latest 4.4.3 release? (Assume you are testing v4.4 branch)

Alvin1Zhang, 2023-02-20:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2023-02-20 by Alvin1Zhang

malachib, 2023-02-20:
Just a ponder, I wonder if this relates to https://github.com/espressif/arduino-esp32/issues/2144  .  I am experiencing a nearly identical issue and it is indeed on a WEMOS D1 ESP32