espressif/esp-idf #6993: SD SPI Driver fails after reading > 100 sectors

nikhilavrio, 2021-05-06:
You can look at many bug reports raised by me related to this.

SPI Frequency - 20Mhz

Tried 10K pullup with and without on CLK pin.

igrr, 2021-05-06:
Hi @nikhilavrio, I'm afraid without more details on how the issue can be reproduced, it will be very hard to help you troubleshoot this. Does the issue happen for the specific hardware design, or you also see it on some known-to-work hardware, like the ESP-WROVER-KIT board? Does this issue happen for a single SD card, several specific SD cards, or all SD cards? Is there some configuration which works well for you, and if so, can you try to bisect the differences between it and the non-working configuration? Furthermore, you can try to use a logic analyzer to record the bus activity when the issue happens. 0x107 is ESP_ERROR_TIMEOUT, which means that the driver didn't receive the command or data response from the card in time. If, for example, you see from the logic analyzer trace that the card did in fact respond, then the issue should be in the driver. If, on the other hand, the card didn't respond, it may be either an incorrect request from the host (so again issue in the driver) or a hardware problem (card browning out, signal integrity issue, etc).

nikhilavrio, 2021-05-06:
Hi @igrr ,

I hooked up the 3V3 of SD Card to Programmable DC Power Supply and ran the code. It looks like SD card was browning out.

I will check again tomorrow if by putting more capacitors the issue is solved and SD Card is working stable in SPI Mode.

Alvin1Zhang, 2021-06-01:
@nikhilavrio Thanks for reporting, would you please help share if any updates for this issue? Thanks.

vamshi51, 2022-03-17:
Hi @nikhilavrio, **esp-idf/examples/storage/sd_card/sdspi** example is modified to write more than 100 sectors with different patterns and read back to verify the same, works fine on ESP-WROVER-KIT. Please provide additional details as suggested  by @igrr

closed 2022-04-11 by igrr