espressif/esp-idf #6996: Audio samples droped in HFP (CVSD codec, VHCI)

RyanYaoHZ, 2021-05-07:
INSTRUCTIONS
============
We found strange phenomena when using HFP( cvsd codec,vhci)，it is simular to issue IDFGH-731 #3171

[image]

## Environment

- Development Kit:      ESP32-LyraT4.3
- Kit version 
- Module or chip used:  [ESP32-WROVER]
- IDF version 4.2 and 4.2.1

## Problem Description

Run hfp_client example in ESP32-LyraT4.3 and record the voice in handset. The recoded waveform lost 60sample every few seconds.

RyanYaoHZ, 2021-05-07:
We can see the issue in both directions. It happens more frequently in HFP client incoming side

xiongweichao, 2021-05-08:
Hi, @RyanYaoHZ 

Could you please provide the log?

Thanks.

RyanYaoHZ, 2021-05-13:
@xiongweichao 

Add "pkt_status" in btm_route_sco_data:

BTM_TRACE_WARNING("SCO: hdl %x, len %d, pkt_sz %d, pkt_status %d\n", handle, p_msg->len, pkt_size, pkt_status);

log attathed. some packets are in status(2) or status(1). 
.
(25021) BT_BTM: SCO: hdl 180, len 16379, pkt_sz 120, pkt_status 1
.
.
(25511) BT_BTM: SCO: hdl 180, len 16379, pkt_sz 120, pkt_status 2

The explanation in Bluetooth spec " No data received. All data from the baseband received during the (e)SCO interval(s) corresponding to the HCI Synchronous Data packet have beenmarked as "lost data" by the baseband. The Payload data octets shall be set to 0."

Update:  A bug found in our project. The outgoing (hf client --> phone) has no such phenomena. This issue exist only in incoming (phone----> hf client)  direction.
[Receive_20210513130401.zip](https://github.com/espressif/esp-idf/files/6471599/Receive_20210513130401.zip)

RyanYaoHZ, 2021-05-13:
sdkconfig attached
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/6471652/sdkconfig.zip)

xiongweichao, 2021-07-02:
Hi, @RyanYaoHZ 

Due to clock drift, the clocks of master and slave are inconsistent, resulting in the phenomenon that the data packet is not received.It's a compatibility issue.

Thanks

Wth-Esp, 2021-12-13:
Hi, @RyanYaoHZ 

We have provided an fix in https://github.com/espressif/esp-idf/issues/3171 and seems that customer feels good with that fix.
Could you please update your IDF and try that commit?

Thanks

closed 2021-12-23 by Alvin1Zhang