espressif/esp-idf #6997: unable to provision the node in the ble mesh in esp idf 4.2.1

ankit-thealchemist, 2021-05-07:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version DevKitC4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    //v4.2.1
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [|Yes (please give details)] vs code
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
 
Hello I upgraded the esp idf from v4.2 to v4.2.1. I was working on the esp ble mesh on the espidf version 4.2 and every thing goes fine. I upgraded to the stable version v4.2.1. It broke the esp ble mesh. Now i am unable to provision the node using the nrf mesh app. Then I downgraded the esp idf to v4.2 and every thing goes fine. Then I  compile the on off server from the examples on the version 4.2.1 and found that it also not provisioning the node.

### Expected Behavior

The node should be provisioned using the pb gatt. I am using the nrf mesh app.
### Actual Behavior

The node has not been provisioned using the nrf mesh app.

### Steps to reproduce

1. step1 compile the on off server example using the esp idf v4.2.1
2. Try to provision the node using the nrf mesh app. 

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

I (14807) bleApp: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
D (15667) BLE_MESH: write, handle 18, len 3, data 030005
D (15667) BLE_MESH: Mesh Provisioning PDU
D (15667) BLE_MESH: 2 bytes: 0005
D (15667) BLE_MESH: Attention Duration: 5 seconds
W (15677) BLE_MESH: No Health Server context provided
D (15677) BLE_MESH: conn 0x3ffc2094 type 0x03 len 12: 010100010000000000000000
D (15687) BLE_MESH: 13 bytes: 03010100010000000000000000
GATT procedure initiated: notify; att_handle=20
D (16697) BLE_MESH: write, handle 18, len 7, data 03020000000000
D (16697) BLE_MESH: Mesh Provisioning PDU
D (16697) BLE_MESH: 6 bytes: 020000000000
I (16697) BLE_MESH: Algorithm:   0x00
I (16697) BLE_MESH: Public Key:  0x00
I (16707) BLE_MESH: Auth Method: 0x00
I (16707) BLE_MESH: Auth Action: 0x00
I (16717) BLE_MESH: Auth Size:   0x00
D (16837) BLE_MESH: write, handle 18, len 20, data 43032f58b9bda5ec60dd142fa503395071d45fd1
D (16837) BLE_MESH: write, handle 18, len 20, data 83a74cf753b413a2f3b4be3d48f7d89548971bec
D (16847) BLE_MESH: write, handle 18, len 20, data 83f29ac1509f6c1e0c85b738a0a2ec7f240a29e1
W (36857) BLE_MESH: Proxy SAR timeout

InfiniteYuan, 2021-05-12:
Hi @ankit-thealchemist 

Please confirm again whether you have switched to v4.2.1 and updated the submodules.

This is test result on v4.2.1(21ecef5c56e01d3cccf0648c869e3fa9248bf3d4).

```
I (31) boot: ESP-IDF v4.2-475-g21ecef5c5 2nd stage bootloader
I (31) boot: compile time 09:58:42
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x282a4 (164516) map
I (177) esp_image: segment 1: paddr=0x000382cc vaddr=0x3ffbdb60 size=0x03e3c ( 15932) load
I (184) esp_image: segment 2: paddr=0x0003c110 vaddr=0x40080000 size=0x00404 (  1028) load
I (185) esp_image: segment 3: paddr=0x0003c51c vaddr=0x40080404 size=0x03afc ( 15100) load
I (199) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x97d54 (621908) map
I (438) esp_image: segment 5: paddr=0x000d7d7c vaddr=0x40083f00 size=0x11404 ( 70660) load
I (481) boot: Loaded app from partition at offset 0x10000
I (481) boot: Disabling RNG early entropy source...
I (482) cpu_start: Pro cpu up.
I (485) cpu_start: Application information:
I (490) cpu_start: Project name:     onoff_server
I (496) cpu_start: App version:      v4.2-475-g21ecef5c5
I (502) cpu_start: Compile time:     May 12 2021 09:58:38
I (508) cpu_start: ELF file SHA256:  06380bd26eef1d49...
I (514) cpu_start: ESP-IDF:          v4.2-475-g21ecef5c5
I (520) cpu_start: Starting app cpu, entry point is 0x40081880
I (0) cpu_start: App cpu up.
I (530) heap_init: Initializing. RAM available for dynamic allocation:
I (537) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (543) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (549) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (555) heap_init: At 3FFCC180 len 00013E80 (79 KiB): DRAM
I (561) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (568) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (574) heap_init: At 40095304 len 0000ACFC (43 KiB): IRAM
I (580) cpu_start: Pro cpu start user code
I (599) spi_flash: detected chip: generic
I (599) spi_flash: flash io: dio
W (599) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (610) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (621) EXAMPLE: Initializing...
I (661) BTDM_INIT: BT controller compile version [a4c513b]
I (661) system_api: Base MAC address is not set
I (661) system_api: read default base MAC address from EFUSE
I (671) phy_init: phy_version 4660,0162888,Dec 23 2020
W (671) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
I (1271) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (1291) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (1291) EXAMPLE: BLE Mesh Node initialized
I (22291) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (22651) BLE_MESH: No Health Server context provided
W (25151) BLE_MESH: No Health Server context provided
I (25171) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (25181) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (25181) EXAMPLE: net_idx: 0x0000, addr: 0x000c
I (25181) EXAMPLE: flags: 0x00, iv_index: 0x00000000
I (43691) EXAMPLE: ESP_BLE_MESH_MODEL_OP_APP_KEY_ADD
I (43691) EXAMPLE: net_idx 0x0000, app_idx 0x0000
I (43691) AppKey: cc dc 3c f0 32 7a b6 0a df ef c0 c1 04 56 5a 6d 
I (53381) EXAMPLE: ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND
I (53381) EXAMPLE: elem_addr 0x000c, app_idx 0x0000, cid 0xffff, mod_id 0x1000
I (57421) EXAMPLE: event 0x00, opcode 0x8203, src 0x0001, dst 0x000c
I (57431) EXAMPLE: ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT
I (57431) EXAMPLE: onoff 0x01
```

ankit-thealchemist, 2021-05-13:
@InfiniteYuan Sorry for the delayed reply. I have update to the esp idf to version 4.2.1  and positively forced updated the modules. example is working fine with the bluedroid.  I have trouble while provisioning the node using the nimble.

Please checkout the below logs. 
[image]

git submodule update -f  --init --recursive
Submodule path 'components/asio/asio': checked out '3b66e5b051381fb70de9c2791df70a06181c64e3'
Submodule path 'components/bootloader/subproject/components/micro-ecc/micro-ecc': checked out 'd037ec89546fad14b5c4d5456c2e23a71e554966'
Submodule path 'components/bt/controller/lib': checked out 'c2e961d07cba8e5a08749e1f1d5c6ffe76d4b115'
Submodule path 'components/bt/host/nimble/nimble': checked out 'ba72e37ac40c13b66ad49c3178c0dbc9588add61'
Submodule path 'components/cbor/tinycbor': checked out 'd2dd95cb8841d88d5a801e3ef9c328fd6200e7bd'
Submodule path 'components/coap/libcoap': checked out '98954eb30a2e728e172a6cd29430ae5bc999b585'
Submodule path 'components/coap/libcoap/ext/tinydtls': checked out '7f8c86e501e690301630029fa9bae22424adf618'
Submodule path 'components/esp_wifi/lib': checked out '1c4c5396cab728b3524fc4a21b324307cd972d0c'
Submodule path 'components/esptool_py/esptool': checked out '4fa0bd7b0d1f69f5ff22b043adc07c5e562a8931'
Submodule path 'components/expat/expat': checked out '968b8cc46dbee47b83318d5f31a8e7907199614b'
Submodule path 'components/json/cJSON': checked out '3c8935676a97c7c97bf006db8312875b4f292f6c'
Submodule path 'components/libsodium/libsodium': checked out '70170c28c844a4786e75efc626e1aeebc93caebc'
Submodule path 'components/lwip/lwip': checked out '7285b846a22ebd02377bd9c67e7f9eaaf842492b'
Submodule path 'components/mbedtls/mbedtls': checked out '99c88bb7bd7c2d91c6873abd6e6d5ee04f4f164c'
Submodule path 'components/mqtt/esp-mqtt': checked out 'da850b0add1e71b3659bfac5d797cc834dc3e89b'
Submodule path 'components/nghttp/nghttp2': checked out '3bcc416e13cc790e2fb45fcfe9111d38609c5032'
Submodule path 'components/nghttp/nghttp2/third-party/mruby': checked out '22464fe5a0a10f2b077eaba109ce1e912e4a77de'
Submodule path 'components/nghttp/nghttp2/third-party/neverbleed': checked out 'da5c2ab419a3bb8a4cc6c37a6c7f3e4bd4b41134'
Submodule path 'components/protobuf-c/protobuf-c': checked out 'dac1a65feac4ad72f612aab99f487056fbcf5c1a'
Submodule path 'components/spiffs/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
Submodule path 'components/tinyusb/tinyusb': checked out '28f89e13473d40637574bcbfe4142633b39899fd'
Submodule path 'components/unity/unity': checked out '7d2bf62b7e6afaf38153041a9d53c21aeeca9a25'
Submodule path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2': checked out '7e8e249990ec491ec15990cf95b6d871a66cf64a'
Submodule path 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib': checked out 'c3d3a69021cfec3236ca2c0b63be4048ec6643a4'

Also please confirm that you are using the **nimble** not the bludroid for the ble mesh.

InfiniteYuan, 2021-05-14:
@ankit-thealchemist  Please provide more details on how you use the ble mesh example or provide your test project directly.

ankit-thealchemist, 2021-05-14:
> @ankit-thealchemist Please provide more details on how you use the ble mesh example or provide your test project directly.

Hello @InfiniteYuan 

The process  I followed is the I go to the esp-idf\examples\bluetooth\esp_ble_mesh\ble_mesh_node\onoff_server.  Then I open the menuconfig. In menuconfig I set the bluetooth host as nimble as shown below. 
[image]

Then I open the esp ble mesh android application and then try to provision the node. The node doesnt get provisioned.

ankit-thealchemist, 2021-05-16:
Hello @InfiniteYuan 

I tried to build the example \esp-idf\examples\bluetooth\nimble\blehr what I found that the example is not working. the device is not able to connect to mobile phone, while the example is perfectly working with version v4.2.

--- idf_monitor on \\.\COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (12) boot: ESP-IDF v4.2.1-dirty 2nd stage bootloader
I (12) boot: compile time 16:59:07
I (13) boot: chip revision: 1
I (15) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (22) boot.esp32: SPI Speed      : 40MHz
I (26) boot.esp32: SPI Mode       : DIO
I (31) boot.esp32: SPI Flash Size : 4MB
I (35) boot: Enabling RNG early entropy source...
I (41) boot: Partition Table:
I (44) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (59) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  2 factory          factory app      00 00 00010000 00100000
I (74) boot: End of partition table
I (78) boot_comm: chip revision: 1, min. application chip revision: 0
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0c380 ( 50048) map
I (113) esp_image: segment 1: paddr=0x0001c3a8 vaddr=0x3ffbdb60 size=0x0343c ( 13372) load
I (119) esp_image: segment 2: paddr=0x0001f7ec vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at D:/IOTCode/framework/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (121) esp_image: segment 3: paddr=0x0001fbf8 vaddr=0x40080404 size=0x00420 (  1056) load
I (130) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x50564 (329060) map
0x400d0020: _stext at ??:?

I (263) esp_image: segment 5: paddr=0x0007058c vaddr=0x40080824 size=0x14b88 ( 84872) load
I (312) boot: Loaded app from partition at offset 0x10000
I (312) boot: Disabling RNG early entropy source...
I (313) cpu_start: Pro cpu up.
I (316) cpu_start: Application information:
I (321) cpu_start: Project name:     blehr
I (326) cpu_start: App version:      v4.2.1
I (331) cpu_start: Compile time:     May 16 2021 16:59:29
I (337) cpu_start: ELF file SHA256:  ef4874ea0525ddb3...
I (343) cpu_start: ESP-IDF:          v4.2.1
I (349) cpu_start: Starting app cpu, entry point is 0x4008188c
0x4008188c: call_start_cpu1 at D:/IOTCode/framework/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (359) heap_init: Initializing. RAM available for dynamic allocation:
I (366) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (372) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (378) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (384) heap_init: At 3FFC3938 len 0001C6C8 (113 KiB): DRAM
I (391) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (397) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (403) heap_init: At 400953AC len 0000AC54 (43 KiB): IRAM
I (410) cpu_start: Pro cpu start user code
I (428) spi_flash: detected chip: generic
I (429) spi_flash: flash io: dio
I (429) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (458) BTDM_INIT: BT controller compile version [a4c513b]
I (458) system_api: Base MAC address is not set
I (458) system_api: read default base MAC address from EFUSE
I (468) phy_init: phy_version 4660,0162888,Dec 23 2020
I (808) NimBLE_BLE_HeartRate: BLE Host Task Started
Device Address: a4:cf:12:6a:5e:2a
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
connection established; status=0
disconnect; reason=520
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0

[image]

what i think is somehow nimble broke down while migrating from version 4.2 to 4.2.1

RahulEspressif, 2021-05-17:
Hi @ankit-thealchemist  , thanks for reporting this issue. We are internally working on fixing this. In meantime, you can disable Host Flow control ( menuconfig ---> Nimble Options --> Deselect "Enable Host Flow Control" ) . This is just a workaround to unblock you as we provide correct fix for this issue.

dhrishi, 2021-06-18:
@ankit-thealchemist  ESP-IDF v4.2.2 is released that has the fix for NimBLE and it is functional as expected. You can find it here: https://github.com/espressif/esp-idf/releases/tag/v4.2.2

Please give it a try and let us know in case of any issues

Alvin1Zhang, 2021-07-14:
@ankit-thealchemist Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

ankit-thealchemist, 2021-08-16:
Yes it was working... Srry forgot to close the  issue.

closed 2021-08-16 by ankit-thealchemist