espressif/esp-idf #7000: Bug in HCD pipe alloc, should be BULK not ISO

chipweinberger, 2021-05-09:
```

esp_err_t hcd_pipe_alloc(hcd_port_handle_t port_hdl, const hcd_pipe_config_t *pipe_config, hcd_pipe_handle_t *pipe_hdl)
{
....
    switch (type) {
        case USB_XFER_TYPE_CTRL:
            hal_type = USB_PRIV_XFER_TYPE_CTRL;
            break;
        case USB_XFER_TYPE_ISOCHRONOUS:
            hal_type = USB_PRIV_XFER_TYPE_ISOCHRONOUS;
            break;
        case USB_XFER_TYPE_BULK:
            hal_type = USB_PRIV_XFER_TYPE_ISOCHRONOUS; //BUG! Should be USB_PRIV_XFER_TYPE_BULK
            break;
        default:    //USB_XFER_TYPE_INTR
            hal_type = USB_PRIV_XFER_TYPE_INTR;
            break;
    }
```

See above. There is a clear copy past bug for USB_XFER_TYPE_BULK.

link: https://github.com/espressif/esp-idf/blob/9d34a1cd42f6f63b3c699c3fe8ec7216dd56f36a/components/usb/hcd.c

chegewara, 2021-05-09:
https://github.com/espressif/esp-idf/issues/6887#issuecomment-828373381

Dazza0, 2021-06-08:
Closed by 7f421048936c6007186db00ed382cd989c4b905e

closed 2021-06-08 by Dazza0