espressif/esp-idf #7004: Setting Max AP Connexions to a custom value yields no node connexions to a parent node

L30n4rd0-code, 2021-05-10:
I'm trying to reduce the Max AP connexions to crate more than layeres, since I only have 6 modules, and the default value is 6. So I modified it to 2, after to 3, but after doing it, I don't see in the monitor the messages logs of child node connexions. So I assume that they are not working as they should. 
I'm using the exact same example, I just added the two following lines, to create a routerless network (with the `ESP_ERROR_CHECK(esp_mesh_set_type(MESH_ROOT));` only for the root node, and not for the rest) 

With the default configuration that works as it should, or that's what I think, however, the child nodes just connect to the root node, but they never send messages at all. I would like to know if somebody could help me giving me some information about the configuration need to do so.

shenjun7, 2021-05-11:
In routerless internal_communication example, root node has no parent, so esp_mesh_comm_p2p_start() can't be started through MESH_EVENT_PARENT_CONNECTED. You need call esp_mesh_comm_p2p_start() after setting the root node to send messages.

L30n4rd0-code, 2021-05-11:
Okey, I understand, I will try so, I'm going to call that function just right after these lines of code, since it's right here where I set the root node:
```
ESP_ERROR_CHECK(esp_mesh_set_type(MESH_ROOT));
ESP_ERROR_CHECK_WITHOUT_ABORT(esp_mesh_fix_root(true));
```

shenjun7, 2021-05-11:
esp_mesh_fix_root(true) need to be called before esp_mesh_set_config(&cfg).

shenjun7, 2021-05-31:
esp_mesh_set_type(MESH_ROOT) need to be called after esp_mesh_start() to avoid the issue that mesh hasn't been started. 
In internal_communication example, the root node sends data to the child nodes by routing table. First confirming whether the routing table of the root node contains all child nodes. Data transmission can be confirmed by adding log above and below esp_mesh_send(), and analyze the reason through return value. If ps mode is configured, please update IDF version to v4.2(d9ec7df3) or v4.3(639e7ad4).

Alvin1Zhang, 2021-12-13:
@L30n4rd0-code Thanks for reporting, would you please help share if any further updates? Thanks.