espressif/esp-idf #7005: gpio_set_level for GPIO_NUM_16 & GPIO_NUM_25 is not working for me.

Dineshkumar-J, 2021-05-10:
## Environment
- Module or chip used:  ESP32-WROOM-32
- IDF version : v4.2.1
- Build System: idf.py
- Compiler version : 8.4.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes, VSCode IDE
- Power Supply:        USB

## Problem Description

//Detailed problem description goes here.
I have on going project which has binary size of 1457 KB.
When I try to gpio_set_level(GPIO_NUM_16,1) after initializing everything properly, that GPIO 16 it is not set to High.

### Expected Behavior
When I call for gpio_set_level(GPIO_NUM_16,1). It has to set to High.

### Actual Behavior
It is not Set to High. But when I call it in While loop without any delay or any other function it is set to high.

### Steps to reproduce
I have tried to modify the gpio example code. it works perfectly. but when I tried to add it to my project it is not working.
I even tried to disable entire program logic. and just used the same code as like as example. But result is same.

And I tried to call the gpio_set_level in a loop which makes it work

This is happening only for the GPIO_NUM_16 and GPIO_NUM_25. other GPIOs are working perfectly.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/gpio.h"

#define GPIO_OUTPUT_IO_0    16
#define GPIO_OUTPUT_IO_1    25
#define GPIO_OUTPUT_PIN_SEL  ((1ULL<<GPIO_OUTPUT_IO_0) | (1ULL<<GPIO_OUTPUT_IO_1))
#define GPIO_INPUT_IO_0     4
#define GPIO_INPUT_IO_1     5
#define GPIO_INPUT_PIN_SEL  ((1ULL<<GPIO_INPUT_IO_0) | (1ULL<<GPIO_INPUT_IO_1))
#define ESP_INTR_FLAG_DEFAULT 0

#define HIGH 1
#define LOW 0

void app_main(void)
{
    gpio_config_t io_conf;
    //disable interrupt
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_OUTPUT;
    //bit mask of the pins that you want to set,e.g.GPIO18/19
    io_conf.pin_bit_mask = GPIO_OUTPUT_PIN_SEL;
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);

[... 162 lines cut ...]
D (1149) APP: [APP] IDF version: v4.2
D (1149) partition: Loading the partition table
D (1169) event: running task for loop 0x3ffbd778
D (1169) event: created task for loop 0x3ffbd778
D (1169) event: created event loop 0x3ffbd778
D (1169) APP: Heaps Are Valid
D (1179) SLEEP: Not a deep sleep reset

D (1179) history: Initializing SPIFFS
I (1209) history: Partition size: total: 233681, used: 502
D (1209) ADC: ADC Calibrartion Type 0

I (1209) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1219) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1229) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1239) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1249) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1259) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1269) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1269) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1279) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
D (1289) intr_alloc: Connected src 22 to int 13 (cpu 0)
D (1299) intr_alloc: Connected src 35 to int 17 (cpu 0)
D (1299) intr_alloc: Connected src 49 to int 18 (cpu 0)
```

Dineshkumar-J, 2021-05-11:
## Problem Description

Hey, Today I found something strange. Could someone help me with this. Even though I disable All GPIO Configs in my Code. Somehow intr_alloc is adding specific GPIO 16 & 25 to interrupt logics.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main(void)
{
    // gpio_config_t io_conf;
    // //disable interrupt
    // io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    // //set as output mode
    // io_conf.mode = GPIO_MODE_OUTPUT;
    // //bit mask of the pins that you want to set,e.g.GPIO18/19
    // io_conf.pin_bit_mask = GPIO_OUTPUT_PIN_SEL;
    // //disable pull-down mode
    // io_conf.pull_down_en = 0;
    // //disable pull-up mode
    // io_conf.pull_up_en = 0;
    // //configure GPIO with the given settings
    // gpio_config(&io_conf);

    // //interrupt of rising edge
    // io_conf.intr_type = GPIO_PIN_INTR_POSEDGE;
    // //bit mask of the pins, use GPIO4/5 here
    // io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
    // //set as input mode    
    // io_conf.mode = GPIO_MODE_INPUT;
    // //enable pull-up mode
    // io_conf.pull_up_en = 1;
    // gpio_config(&io_conf);

    //change gpio intrrupt type for one pin
    // gpio_set_intr_type(GPIO_INPUT_IO_0, GPIO_INTR_ANYEDGE);

    //create a queue to handle gpio event from isr
    // gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
    // //start gpio task
    // xTaskCreate(gpio_task_example, "gpio_task_example", 2048, NULL, 10, NULL);

    //install gpio isr service
    // gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
    //hook isr handler for specific gpio pin
    // gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
    // //hook isr handler for specific gpio pin
    // gpio_isr_handler_add(GPIO_INPUT_IO_1, gpio_isr_handler, (void*) GPIO_INPUT_IO_1);

    //remove isr handler for gpio number.
    // gpio_isr_handler_remove(GPIO_INPUT_IO_0);
    // //hook isr handler for specific gpio pin again
    // gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);

    int cnt = 0;
	// while (1){
	gpio_set_level(BM_LITE_RST_N,HIGH);
	ESP_LOGD(TAG,"BM_LITE_IRQ GPIO Level: %d",gpio_get_level(BM_LITE_IRQ));
	ESP_LOGD(TAG,"LOCK_PLATE_FEEDBACK GPIO Level: %d",gpio_get_level(LOCK_PLATE_FEEDBACK));

	// }
    // while(1) {
    //     printf("cnt: %d\n", cnt++);
    //     vTaskDelay(1000 / portTICK_RATE_MS);
    //     gpio_set_level(GPIO_OUTPUT_IO_0, cnt % 2);
    //     gpio_set_level(GPIO_OUTPUT_IO_1, cnt % 2);
    // }
}
[... 80 lines cut ...]
I (644) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (651) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (657) heap_init: New heap initialised at 0x400891e4
I (662) heap_init: At 400891E4 len 00016E1C (91 KiB): IRAM
I (668) cpu_start: Pro cpu start user code
D (673) clk: waiting for 32k oscillator to start up
D (709) clk: RTC_SLOW_CLK calibration value: 11948147
D (715) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (715) intr_alloc: Connected src 17 to int 3 (cpu 0)
D (716) intr_alloc: Connected src 24 to int 9 (cpu 0)
D (721) FLASH_HAL: extra_dummy: 1
D (724) spi_flash: trying chip: issi
D (728) spi_flash: trying chip: gd
D (731) spi_flash: trying chip: mxic
D (735) spi_flash: trying chip: generic
I (739) spi_flash: detected chip: generic
I (744) spi_flash: flash io: qio
D (747) chip_generic: set_io_mode: status before 0x200
I (753) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (0) cpu_start: Starting scheduler on APP CPU.
D (768) heap_init: New heap initialised at 0x3ffe0440
D (778) heap_init: New heap initialised at 0x3ffe4350
D (778) intr_alloc: Connected src 16 to int 12 (cpu 0)
```

negativekelvin, 2021-05-11:
`D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)`

This has nothing to do with gpios, interrupt sources have their own numbering system

Dineshkumar-J, 2021-05-11:
But As I Can see clearly, I can't able set the level for these GPIOs to High without Calling it in loop. But Other GPIOs are working fine. I even tried to declared them as a Inputs. In that case it works fine. What may be reason?.

ryan4volts, 2021-05-14:
Try something more rudimentary like the below.

```
const int io_16 = 16;
gpio_pad_select_gpio(io_16);
ESP_ERROR_CHECK(gpio_set_direction((gpio_num_t)io_16 , (gpio_mode_t)GPIO_MODE_OUTPUT));
ESP_ERROR_CHECK(gpio_set_level((gpio_num_t)io_16 , 1));
```

Dineshkumar-J, 2021-05-24:
> Try something more rudimentary like the below.
> 
> ```
> const int io_25 = 25;
> gpio_pad_select_gpio(io_25);
> ESP_ERROR_CHECK(gpio_set_direction((gpio_num_t)io_25 , (gpio_mode_t)GPIO_MODE_OUTPUT));
> ESP_ERROR_CHECK(gpio_set_level((gpio_num_t)io_25 , 1));
> ```
> 
> Note the following IOs are used for flash/ram and SHOULD not be used for general ios:
> 
> * IO11
> * IO7
> * IO16
> * IO17
> * IO6
> * IO8

I tried this. But it didn't show any progress. The issue is still there.
Also, As PSRAM is Optional one, I didn't use that in my design.

Dineshkumar-J, 2021-05-24:
> `D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)`
> 
> This has nothing to do with gpios, interrupt sources have their own numbering system

But I can't able use those pins as output. Particularly , GPIO16, 17 and 25. Which is connected to the intr_alloc. Can you please help me with this.?

Dineshkumar-J, 2021-07-31:
@negativekelvin Can I get any suggestion regard this. I am still getting this issue. even after I update esp-idf to v4.3.

ginkgm, 2021-08-02:
Hi @Dineshkumar-J ,

Your issue looks like a concurrency issue, I wonder if there are too many controller for the same GPIO. Are you using gpio_set_level in different threads for GPIO16 and GPIO25 respectively? btw, using interrupt will not reduce your problem, unless you are moving all the code into the interrupt, otherwise you are just adding one possible controller for the GPIO to make it more uncontrollable.

Another thought, as you say, if your code is simpled enough (only one task), and looks very like the example but still can't work. The only difference between your code and the example is the sdkconfig. Maybe you can compare the sdkconfig of the example, and your project.

ginkgm, 2022-04-17:
Hi @mkeyno ,
I don't know too much about 3rd party module/devkits. But basically which pins are occupied by Flash and PSRAM is determined by:
1. the efuse inside your module. (decides CLK, CS, MOSI, MISO, HD, package)
2. bootloader sdkconfig (decides WP, will also check the package above)
3. PSRAM sdkconfig (decides CLK and CS of PSRAM, will also check the package above)

But before continue debugging, I suggest check if those pins are break out on your devkit first. If you are sure your devkit vendor supports that, please come up with:
1. The sdkconfig you are using
2. The efuse inside your chip (by calling `espefuse.py summary`)

ginkgm, 2022-04-17:
Hi @mkeyno ,
I don't know too much about 3rd party module/devkits. But basically which pins are occupied by Flash and PSRAM is determined by:
1. the efuse inside your module. (decides CLK, CS, MOSI, MISO, HD, package)
2. bootloader sdkconfig (decides WP, will also check the package above)
3. PSRAM sdkconfig (decides CLK and CS of PSRAM, will also check the package above)

But before continue debugging, I suggest check if those pins are break out on your devkit first. If you are sure your devkit vendor supports that, please come up with:
1. The sdkconfig you are using
2. The efuse inside your chip (by calling `espefuse.py summary`)

btw, I think this issue is case by case, next time please open a new issue, instead of replying under the same topic. This topic will be closed as soon as this discussion is resolved.

Alvin1Zhang, 2023-06-14:
Thanks for reporting, will close due to short of feedback, feel free to reopen.

closed 2023-06-14 by Alvin1Zhang

L-KAYA, 2023-06-14:
For now, there is a function to check whether a pin on the current SoC is reserved for other purpose, for example:
```c
#include "esp_gpio_reserve.h"
...
uint32_t pin = 16;
bool is_reserved = esp_gpio_is_pin_reserved(pin);
```

vvb333007, 2024-12-25:
Hello from 2025! They did change location of the include file as well as the function name.