espressif/esp-idf #7006: Bad RTC Crystal causes esp_restart() function to fail

ryan4volts, 2021-05-11:
My esp32 fails to restart correctly after a crash or by calling the `esp_restart()` function. It simply "stops".

Looking more closely at the sdkconfig, from the default settings to reproduce this, simply enable the external 32khz crystal in the rtc-clock source section.

The first hint for this was the line on boot:
```
clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator
```

On my hardware I originally had poorly chosen load capacitors, this meant the crystal would never go into oscillation. This should easily be repeatable by reducing any load capacitors on existing boards to test against. The issue was confirmed by simply placing my oscilloscope probe (~7pF) onto the XP pin. This increased the capacitance to roughly the target value and brought the crystal into oscillation. This was futher verified by using the get-started/hello-world which does the constant restart. Here the app stops after the first restart attempt, stuck (I believe) at the shutdown, rather than the boot up. Once I place my scope probe on the XP capacitor, I can see the crystal comes into oscillation, the software continues to run and restarts as expected.

Current software work around is to simply increase the log verbosity to either DEBUG or VERBOSE. The device then operates as expected.

A few details
- esp-idf v4.2.1 
- esp32-pico32-d4 
- custom hardware

KonstantinKondrashov, 2022-03-23:
Hi @ryan4volts 

Could you try this option `ESP32_RTC_EXT_CRYST_ADDIT_CURRENT`  https://github.com/espressif/esp-idf/blob/release/v4.2/components/esp32/Kconfig#L567-L578? I think it could help to fix this issue.

ryan4volts, 2022-03-23:
Unfortunately I fixed all my boards since this occurred and don't have anything to test this on. Though if I see the issue in the future i'll try this fix and report back!

closed 2022-11-17 by Alvin1Zhang