espressif/esp-idf #7008: [TensorFlowLite] Deploy issue: micro_speech example project using platformIO on Visual Studio Code

vanessastad, 2021-05-11:
Hello, I'm trying to create a project on Visual Studio Code using platformIO and I want to develop it on a ESP32 board but I'm having some problems. Can you help me?

## Environment

- Development Kit:  ESP32-DevKitC
- Kit version DevKitC: v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version : v4.2.1
- Build System:   CMake
- Compiler version : 8.4.0
- Operating System:     Windows
- Windows  environment type: ESP Command Prompt and PowerShell.
- Using an IDE?: Yes -> Visual Studio Code
- Power Supply:   USB

## Problem Description

I want to load the speech_recognition example of tensorflow lite on my board (esp32 wroom 32d). I can build and load the code but in the serial monitor I visualize:

"Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1252
load:0x40078000,len:12692
load:0x40080400,len:3100
entry 0x400805ec
assertion "buf" failed: file "src/esp/ringbuf.c", line 51, function: rb_init

abort() was called at PC 0x400e82a3 on core 1"

[micro_speech.zip](https://github.com/espressif/esp-idf/files/6459278/micro_speech.zip)

vikramdattu, 2021-05-17:
Hi @vanessastad can you please provide more info on this?

What's the TFLite version/commit you are using?
If you keep getting errors, can you please enable gdb stub and provide the logs?

PS: TFLite would be better place for the issue.

vikramdattu, 2021-05-20:
Hello again,

Okay, this is due to memory allocation is failing there. That's why the assert.
Try to optimize memory footprint and check if it works.

Decrease task stack for main task. Change it to 8KB or something. This will save ~24KB.
You can use APIs from esp_heap_caps.h to check available memory at particular instance.

Thanks.

vanessastad, 2021-05-20:
Thank you very much @vikramdattu.
Do you think the use of a partition table could help?
I experienced this problem also on a different code that was working before to integrate some lines for a Bluetooth application. Could the creation of two different tasks (one for each core) fix memory allocation problems?? 

Thanks again for your time.

vikramdattu, 2021-05-20:
No problem.

The change in partition table or running tasks on different cores should not have impact on memory available.

It's just that the tasks created in this application have been assigned 32 KB size lazily.

Those can easily fit in much lesser memory (8KB) each.

There are two such tasks in following files:

1. app_main.c
2. audio_provider.c

Change the argument to xTaskCreate to 8KB (Currently it is 32K).

This will save 48K in total.

Thanks.

vanessastad, 2021-05-21:
Hi @vikramdattu, 

I have followed your suggestion but unfortunately without results. I still get: "

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00       
mode:DIO, clock div:2
load:0x3fff0030,len:1252
load:0x40078000,len:12692
load:0x40080400,len:3100
entry 0x400805ec
assertion "buf" failed: file "src/esp/ringbuf.c", line 51, function: rb_init

abort() was called at PC 0x400e8faf on core 1

Backtrace:0x400dc819:0x3ffcc8f00x40086bc1:0x3ffcc910 0x4008be11:0x3ffcc930 0x400e8faf:0x3ffcc9b0 0x400d1999:0x3ffcc9e0 0x400d1859:0x3ffcca00 0x400d18b2:0x3ffcca30 0x400d1d7c:0x3ffcca50 0x400d23c0:0x3ffcca90 0x400d1926:0x3ffccac0      
  #0  0x400dc819:0x3ffcc8f00 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/build/../esp-idf/components/esp_system/panic.c:365

ELF file SHA256: 0000000000000000"

The weird thing is that when I upload the code the memory doesn't seem to be full. In the terminal appears: "

> RAM:   [=         ]   8.1% (used 26476 bytes from 327680 bytes)

Flash: [=         ]   9.8% (used 308216 bytes from 3145728 bytes)"

and this doesn't change if I use 1024*32 or 1024*8 in xTaskCreate.

vikramdattu, 2021-05-21:
Hi @vanessastad I think you are confused between flash, IRAM and heap memory.
We are trying to optimize for heap memory here.
Static analysis won't show this.

1. Did you change the stack sizes for both the tasks mentioned to 8KB?
2. You will need to regenerate the example for changes in examples/ to take effect in generated example present in ...prj/micro_speech/esp-idf directory.

If that's not sufficient do change ringbuffer size which is set to 80000 to 40000 and check again.
(kAudioCaptureBufferSize variable in esp/audio_provider.cc)

BTW, did you attach some microphone module to the module?

vanessastad, 2021-05-21:
I did what you said changing both tasks size and ringbuffer size but it didn't work. I didn't perform andyregeneration of the example because I don't know what you mean. Yes, I have a max4466 microphone attached to my esp32. What I don't get at all is why this code works on esp-idf but not on platformIO

vikramdattu, 2021-05-22:
PlatformIO should not make much difference.
The example has been tested for esp-eye board, which has 4MB of external SPIRAM and hence there is no memory crunch on that board.
Example generation process of TFLite is bit tedious to understand. It copies files to different folder and from there you build the example. Hence the changes you make in sources won't reflect automatically.

Follow following steps do regenerate the example... ([link to README](https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/examples/micro_speech#deploy-to-esp32))

1. Remove generated example: `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp clean`
2. Make the changes discussed above in `tensorflow/lite/micro/examples/micro_speech` example
3. Re-generate the example: `make -f tensorflow/lite/micro/tools/make/Makefile TARGET=esp generate_micro_speech_esp_project`

You are now ready to build and flash the example again.

If you keep getting issue. Please use [`heap_caps_get_free_size`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html#_CPPv423heap_caps_get_free_size8uint32_t) API to query available memory before allocation of the ringbuffer which is getting assert.

Do make sure to regenerate the example when you make any changes.

Thanks.

Alvin1Zhang, 2021-05-31:
@vanessastad Thanks for reporting and sharing progress, would you please help share if any updates for the issue? Thanks.

vanessastad, 2021-06-21:
Good morning,
Unfortunately I was not able to fix this problem using platformIO and neither esp-idf so that I moved to Arduino IDE.

Thank you very much for your time.

closed 2021-06-21 by vanessastad