espressif/esp-idf #7009: [ESP32-C3] Setting UART1 to 9600 baud rate actually set a higher baud rate

rafaeldantaas, 2021-05-11:
## Environment

- Development Kit: ESP32-C3
- Kit version: DevKitM-1: v1
- Module or chip used:  ESP32-C3-MINI-1
- IDF version: v3.2-dev-10751-gc78d859f1
- Build System: idf.py
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System: macOS
- Using an IDE?: Yes (Visual Studio Code)
- Power Supply: USB

## Problem Description

I'm trying to get my UART1 working with 9600 baud rate, without success... I can see with an analyzer my packages on TX pin but with a wrong baud rate (~500k). It is working fine for 115200 and 57600 baud rate.

Tried to use the ESP32-WROOM-32D to validate the steps of installing the driver, set pins and send command... Working fine too!

### Expected Behavior
Transmit frames over 9600 baud rate with ESP32-C3

### Actual Behavior
Transmitting frames over a wrong baud rate with ESP32-C3

### Steps to reproduce

1. Set UART1 pins tx and rx (GPIO 7 and GPIO 9);
2. Configure uart_config_t structure;
3. Install drive and load parameters;
4. Transmit some bytes;

Logic analyzer image with auto baud detect...
[image]

### Code to reproduce this issue

```cpp

#include "esp_log.h"
#include "driver/uart.h"

void uart_transmit(char* frame, size_t size){
    uart_write_bytes(UART_NUM_1, (const char*)frame, size);
}

void app_main()
{
    esp_err_t ret;

    uart_config_t uart_config = {
        .baud_rate = 9600,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
    };

    // Install UART driver using an event queue here
    ret = uart_driver_install(UART_NUM_1, CTRL32_UART_BUFFER_SIZE, \
                              CTRL32_UART_BUFFER_SIZE, 0, NULL, 0);
    ESP_ERROR_CHECK(ret);

    // Configure UART parameters
    ret = uart_param_config(UART_NUM_1, &uart_config);
    ESP_ERROR_CHECK(ret);

    // Set UART pins
    ret = uart_set_pin(UART_NUM_1, 7, 9, 18, 19);
    ESP_ERROR_CHECK(ret);

    SysInfo("UART driver sucessfully instaled");

    uart_transmit("55AA", 4);
}

```

igrr, 2021-05-11:
This looks like an issue that was fixed in https://github.com/espressif/esp-idf/commit/fc61e60948e22e5e866e7a3e12e0dc6959dc09a1...

Could you please confirm which commit you are using?  Commit `c78d859f1` mentioned in your `git describe` output doesn't exist in this repository, perhaps you have made some commits locally?

It looks like it is some older commit than the one where the fix got added. The fix commit is `v3.2-dev-11143-gfc61e60948e` if we `git describe` it relative to `v3.2-dev`, which means 11143 commits after v3.2-dev; whereas yours is 10751 commits after `v3.2-dev`.

Alvin1Zhang, 2021-05-27:
@rafaeldantaas Thanks for reporting, would you please help share if any updates for this issue? Thanks.

rafaeldantaas, 2021-05-27:
The commit [fc61e60](https://github.com/espressif/esp-idf/commit/fc61e60948e22e5e866e7a3e12e0dc6959dc09a1) fixed my issue, I just had to update my esp-idf, thanks for the support!

closed 2021-05-27 by igrr