espressif/esp-idf #7010: Where is the esp_websocket_client_send_text() buffer?

0xFEEDC0DE64, 2021-05-12:
The ESP32 is connected to a wifi station and connects to a backend server via websocket client.

When I unplug the internet uplink ethernet cable from the wifi access point, I expect all sent websocket messages fail to send, because no acknoledge can come back. But instead, the esp_websocket_client_send_text() function returns the correct size of the sent message even 10 seconds after unplugging the cable. And the free heap memory goes low and sometimes even memory allocations start to fail and crash.

Physical setup:
Internet <- cable -> Modem/NAT <- ethernet cable -> WiFi AccessPoint <- Wireless -> ESP32 (in station mode)

Software setup:
The main task starts the websocket client and another websocket send task. Every second, the main task checks if the websocket client is connected, and if it is, pushes a new message to send into my FIFO buffer. My websocket send task then takes the message from the FIFO buffer calls esp_websocket_client_send_text() from the esp-idf and loggs success or fail.

For my log output below:
`cloud_send()` is called from the main thread and enqueues the message to send
`cloud_send_task()` is the separate send task and wakes up every time a message is ready to send
`websocketClient.send_text()` is my C++ wrapper for esp_websocket_client_send_text()
`<--- ---->` are my hand written comments to the logoutput

My questions are:
Where are the sending buffers?
How can I turn them off?
When I unplug the internet uplink I want the esp_websocket_client_send_text() to get stuck immediatly and not only after 10s or so.
How can I get a negative return value indicating the send failed? 

```
D (1817735) CLOUD cloud.cpp:196 cloud_send(): size=201
D (1817735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=201
D (1817745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=201 took 10ms
D (1818125) CLOUD cloud.cpp:489 cloudEventHandler(): WEBSOCKET_EVENTS event_id=WEBSOCKET_EVENT_DATA op_code=HTTPD_WS_TYPE_PONG data_len=0 payload_len=0 payload_offset=0
D (1818735) CLOUD cloud.cpp:196 cloud_send(): size=222
D (1818735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=222
D (1818745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=222 took 9ms
D (1819735) CLOUD cloud.cpp:196 cloud_send(): size=188
D (1819735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=188
D (1819745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=188 took 9ms
D (1820735) CLOUD cloud.cpp:196 cloud_send(): size=189
D (1820735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=189
D (1820745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=189 took 10ms
 <---- HERE I UNPLUG THE ETHERNET CABLE ---->
D (1821735) CLOUD cloud.cpp:196 cloud_send(): size=242
D (1821735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=242
D (1821745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=242 took 6ms
<---- AND STILL SENDING SUCCEEDS AND RETURNS THE RIGHT SIZE VERY QUICKLY 6ms ---->
D (1822735) CLOUD cloud.cpp:196 cloud_send(): size=254
D (1822735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=254
D (1822745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=254 took 6ms
D (1823735) CLOUD cloud.cpp:196 cloud_send(): size=189
D (1823735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=189
D (1823745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=189 took 6ms
D (1824735) CLOUD cloud.cpp:196 cloud_send(): size=200
D (1824735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=200
D (1824745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=200 took 5ms
D (1825735) CLOUD cloud.cpp:196 cloud_send(): size=242
D (1825735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=242
D (1825745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=242 took 6ms
D (1826735) CLOUD cloud.cpp:196 cloud_send(): size=296
D (1826735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=296
D (1826745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=296 took 5ms
W (1827205) wifi:<ba-del>idx
D (1827735) CLOUD cloud.cpp:196 cloud_send(): size=296
D (1827735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=296
D (1827745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=296 took 5ms
D (1828735) CLOUD cloud.cpp:196 cloud_send(): size=304
D (1828735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=304
D (1828745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=304 took 6ms
D (1829735) CLOUD cloud.cpp:196 cloud_send(): size=242
D (1829735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=242
D (1829745) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=242 took 6ms
D (1830735) CLOUD cloud.cpp:196 cloud_send(): size=285
D (1830735) CLOUD cloud.cpp:338 cloud_send_task(): websocketClient.send_text() for size=285
D (1831735) CLOUD cloud.cpp:196 cloud_send(): size=283
 <---- HERE I GET FIRST SIGNS OF A BROKEN CONNECTION? ---->
E (1832735) TRANSPORT_WS transport_ws.c:243 _ws_write(): Error transport_poll_write
E (1832735) WEBSOCKET_CLIENT esp_websocket_client.c:872 esp_websocket_client_send_with_opcode(): Network error: esp_transport_write() returned 0, errno=0
D (1832735) CLOUD cloud.cpp:196 cloud_send(): size=303
I (1832745) WEBSOCKET_CLIENT esp_websocket_client.c:165 esp_websocket_client_abort_connection(): Reconnect after 10000 ms
W (1832765) CLOUD cloud.cpp:477 cloudEventHandler(): WEBSOCKET_EVENTS event_id=WEBSOCKET_EVENT_DISCONNECTED
I (1832775) CLOUD cloud.cpp:478 cloudEventHandler(): cloud queue size 2
D (1832785) CLOUD cloud.cpp:345 cloud_send_task(): websocketClient.send_text() for size=285 took 2043ms
I (1832785) CMDHANDLER cmdhandler.cpp:2304 cmdhandler_update(): cmdhandler_update() type=Disconnected socketIdentifier=cloud
E (1832785) CLOUD cloud.cpp:353 cloud_send_task(): websocket sent size mismatch, sent=0, expected=285
 <---- RETURN VALUE DOES NOT TELL ME ERROR, BUT SIZE 0? ---->
W (1851145) wifi:<ba-add>idx:0 (ifx:0, 30:b5:c2:c2:e2:c6), tid:0, ssn:2336, winSize:64
```

0xFEEDC0DE64, 2021-05-13:
Looks like lwip is (by default) configured to have lots of outgoing buffers. And by disabling them I could fix many problems with unstable internet connection uplinks on the esp32 :)

AxelLin, 2021-05-13:
Hi @0xFEEDC0DE64 
For better understanding the issue, can you list the difference of default setting v.s. your working setting for comparison?

Alvin1Zhang, 2021-08-31:
@0xFEEDC0DE64 Would you please help share if any updates for this issue? Thanks.

Alvin1Zhang, 2021-12-07:
Thanks for reporting, would close due to short of feedback, feel free to reopen with more updates.

closed 2021-12-07 by Alvin1Zhang