espressif/esp-idf #7011: WPA2 Enterprise TTLS+PAP Phase 2 authentication fails due to unsupported PAP

jmgraeffe, 2021-05-12:
- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): not sure right now
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    // v4.4-dev-1254-g639e7ad49
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

### Expected Behavior

Connection to **WPA2 Enterprise** network can be established successfully using **TTLS** and **PAP** protocol (it's listed in the `wpa2_enterprise` example configuration options for Phase2).

### Actual Behavior

The ESP32 considers PAP (even though it's available in example) to be unsupported and then switches to PEAP+MSCHAPv2 automatically:

```
...
I (44134) wpa: TLS: Unsupported Phase2 EAP method 'PAP'
...
D (5040) wpa: EAP-PEAP: Start (server ver=0, own ver=1)
D (5050) wpa: EAP-PEAP: Using PEAP version 0
...
```

Our network does support PEAP+MSCHAPv2 but it uses different Kerberos systems, thus some accounts are only available with TTLS+PAP. Yes, it's stupid, but well.

### Steps to reproduce

1. go to `examples/wifi/wpa2_enterprise`
2. `idf.py menuconfig` and do all example-related settings, especially TTLS+PAP
3. `idf.py build flash monitor`

### Code to reproduce this issue

Example in `examples/wifi/wpa2_enterprise` was used.

## Debug Logs

[ESP32 log with Phase 2 failure](https://github.com/espressif/esp-idf/files/6467829/esp32_ttls_pap_miotr_failure.txt) (it fails because it switches to PEAP+MSCHAPv2 which does not work with the specified login credentials)

[ESP32 log with successful connection due to switch to PEAP](https://github.com/espressif/esp-idf/files/6467852/esp32_pap_unsupported_peap_miotr_success.txt) (the specified login credentials are present on both Kerberos systems...)

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder
- [ ] coredump (This provides stacks of tasks.)

If the other things are needed, I'll provide them.

Alvin1Zhang, 2021-05-13:
Thanks for reporting and sharing the details, we will look into.

hpdhabe, 2021-06-01:
Hi @jmgraeffe,

I tried reproducing the issue on a local hostapd/freeradius server and was able to reproduce it only when the server is set to TLS and the client is set to TTLS/PAP. This is what I guess is happening, in case the client does not support the method proposed by the server, the client device will send a NAK which has to be populated with the best available authentication method. Which in your case is PEAP and not TTLS/PAP. PEAP gets registered because it only requires username and password. Same as TTLS.

So you will either have to set the server to explicitly use TTLS as default EAP method ,which is unsafe.
OR,  you can simply comment out the code that adds EAP/PEAP as supported method. (Just change this line https://github.com/espressif/esp-idf/blob/c13afea635adec735435961270d0894ff46eef85/components/wpa_supplicant/src/eap_peer/eap.c#L216
to `#if 0`)

Please let us know if this works. Regards.

negativekelvin, 2021-06-02:
@hpdhabe there should be user api to set methods at runtime

jmgraeffe, 2021-06-07:
> was able to reproduce it only when the server is set to TLS and the client is set to TTLS/PAP

That's interesting. The server should not support TLS (if you mean PKI authentication with that) at all. Maybe the server just tells PEAP is the default at the beginning, and the client just chooses that one since it was the proposed one? Because the example's code has nothing to set any priority or enforcement of any protocol whatsoever.

> OR, you can simply comment out the code that adds EAP/PEAP as supported method.

@hpdhabe This works fine, yes. Thanks for the effort!

However, I agree with @negativekelvin although it does not necessarily need to be at runtime, a CMake option or a way to enable/disable protocols over defines would suffice. Or would it make sense to just try out all "matching" protocol combinations to ensure that everything was attempted?

Also the example's configuration may be a bit misleading as setting TTLS+PAP does not actually enforce or prioritize the protocol combination, but that's probably nitpicking and I see that the program still needs to know the TTLS phase 2 protocol.

sagb2015, 2021-06-09:
@negativekelvin @jmgraeffe - Thanks for your inputs! We will add a way to configure the authentication method preference on device side since upstream supplicant seems to have it.

jmgraeffe, 2021-06-14:
Okay, thanks for your help. It would be nice if the issue is mentioned in the commit once the changes are upstream.

closed 2021-06-14 by jmgraeffe

CarlosGS, 2021-06-22:
@sagb2015 Would this allow to manually specify PAP as the phase2 authentication?
Currently esp-idf always defaults to MSCHAPv2 even though the server doesn't allow it, so the connection times out.
Is there any way at all to achieve PAP with the current version?

sagb2015, 2021-06-23:
Hi @CarlosGS, you can use `esp_wifi_sta_wpa2_ent_set_ttls_phase2_method(ESP_EAP_TTLS_PHASE2_PAP);` to set phase2 method as PAP.

CarlosGS, 2022-03-12:
Hi! Make sure to use a recent version of the IDF and also `#include <esp_wpa2.h>`, that should do it.

I can confirm the method works very well, allowing us to deploy many sensors in an enterprise network (thanks Sagar! :tada:)