espressif/esp-idf #7018: Terminal crashes when I run source vendors/espressif/esp-idf/install.sh

OzzyGrammer, 2021-05-14:
on my ubuntu 18.04 whenever I run the install.sh script my terminal crashes

source vendors/espressif/esp-idf/install.sh
. vendors/espressif/esp-idf/install.sh
vendors/espressif/esp-idf/install.sh

Alvin1Zhang, 2021-06-07:
Thanks for reporting, will close due to short of details, feel free to reopen with more details. Thanks.

closed 2021-06-07 by Alvin1Zhang

aaronjudd, 2024-09-24:
this is actually a real issue - If you run  `$IDF_PATH/install.sh esp32` it will not crash.  if you run with source ` source $IDF_PATH/install.sh esp32` it will crash the terminal.

fhrbata, 2024-09-25:
> this is actually a real issue - If you run `$IDF_PATH/install.sh esp32` it will not crash. if you run with source ` source $IDF_PATH/install.sh esp32` it will crash the terminal.

Hello, the `install.sh` script should be executed normally, not sourced. The issue you're encountering arises because the script includes `set -e` (which causes the script to exit immediately if any command returns a non-zero status). The
 command `basedir=$(dirname "$0")` fails because `$0` does not contain the script name when it is sourced. As a result, the `install.sh` script exits, and since it is sourced, it also closes your current terminal session. This is equivalent to running `exit` in the terminal.