espressif/esp-idf #7019: Security status about wifi vulnerability: fragattacks

Rotzbua, 2021-05-14:
**Is your feature request related to a problem? Please describe.**

Currently there is no status page where I can lookup currently known security flaws with espressif products and the patch status.
I noticed this with the new wifi security flaw fragattacks which has an similar impact than KRACK.
https://www.fragattacks.com/

Is the esp affected by the fragattacks vulnerability?

**Describe the solution you'd like**

I clear webpage which shows CVEs, related software (idf) versions and patch status.

nishanth-radja, 2021-05-17:
we are looking into the testcases of frag attacks to see if we are vulnerable.

projectgus, 2021-05-20:
Hi @Rotzbua,

There is an Advisories section on our website here: https://www.espressif.com/en/support/documents/advisories . We will post an advisory there once we've finished analysis and patching, and also update here.

The vulnerability pertains to design flaw in the specification and therefore ubiquitous.  It involves toggling an unauthenticated flag in otherwise authenticated and encrypted packet for the purpose of changing parsing sequence to favor the attacker. The attacker also needs to be in physical proximity of the device under attack for the purpose of spoofing (man-in-the-middle).  We believe this vulnerability is difficult to exploit, however, we will be making corresponding patches and an advisory available soon.

ameuret, 2021-06-16:
> The attacker also needs to be in physical proximity

🤔The attacker simply has to be within WIFI range.

projectgus, 2021-06-29:
The advisory for this issue has been published on the [Espressif advisories page](https://www.espressif.com/en/support/documents/advisories) as [AR2021-003 (PDF format)](https://www.espressif.com/sites/default/files/advisory_downloads/AR2021-003%20Security%20Advisory%20for%20WFA%20vulnerability.pdf).

closed 2021-06-29 by projectgus

AxelLin, 2021-06-29:
Hi @projectgus 
When will Release v4.3.1 ( 46144f70 ) available?

barbudor, 2021-06-29:
Hi  
Could someone please help me understanding the advisory ?
Does it means that the listed releases will fix the 5 identified CVEs including those 3 that are related to the Wifi standard design ?
Thanks

projectgus, 2021-07-01:
Hi @barbudor,

That's correct. Some of the releases are future releases (not shown on [release page](https://github.com/espressif/esp-idf/) yet. In these cases if you check out a pre-release branch (like the `release/v4.3` branch) that includes the listed commit, then you'll have the fix ahead of the stable release.