espressif/esp-idf #7020: ESP32S2 Support second CDC channel

KlausPopp, 2021-05-14:
Hi, 
we'd like to have a second CDC channel over USB, to have separate communication channels between the host and the ESP32.
Is there a plan to support this?

Alvin1Zhang, 2021-05-17:
Thanks for raising this feature request.

moefear85, 2021-11-26:
would like to have this feature too. The USB protocol is capable of supporting up to 5 independent cdc ports

moefear85, 2021-12-05:
I added multi-cdc feature to esp-idf and tinyusb, in the form of patches. Since I'm not yet versed in git, please visit https://github.com/moefear85/esp-idf. If possible, please merge into the mainstream esp-idf.

KlausPopp, 2021-12-05:
Hello @moefear85 : Thanks for the information. We don't need this feature anymore. We took a different approach and use Ethernet-over USB. We implemented a driver for this, which you can find here: https://github.com/ci4rail/esp-idf/tree/update_mdns_comp/components/tinyusb/additions/src. Perhaps this might be also valuable for you.

closed 2022-01-05 by espressif-bot