espressif/esp-idf #7024: Component error - "project command is not scriptable"

qt1, 2021-05-15:
I have a project that contains reference to an external component. Running ```idf.py set-target esp32``` fails 

Reconstruct:

After installing esp-idf
    
    cp -r esp-idf/examples/get-started/hello_world/ .
    cd hello_world
    git init
    source ../esp-idf/export.sh
    idf.py set-target esp32
    rm -r build

So far so good, it works fine.
Now I am trying to add a component (still in hello_world).

    git submodule add https://github.com/bblanchon/ArduinoJson.git components/ArduinoJson
    
    
This fails. With the error:

````
CMake Error at /home/baruch/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  fatal: Needed a single revision

  CMake Error at
  /home/baruch/esp/hello_world/components/ArduinoJson/CMakeLists.txt:7
  (project):

    project command is not scriptable

  Call Stack (most recent call first):

    /home/baruch/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:97 (include)
    /home/baruch/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:115 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /home/baruch/esp/esp-idf/tools/cmake/build.cmake:487 (__component_get_requirements)
  /home/baruch/esp/esp-idf/tools/cmake/project.cmake:361 (idf_build_process)
  CMakeLists.txt:6 (project)
````    

This looks like cmake clash between esp-idf and a component's CMakeLists.txt
[CMakeLists.txt](https://github.com/espressif/esp-idf/files/6488108/CMakeLists.txt)

Before esp-idf 4 I have added a component.mk file that worked with the make build system. 
[component.mk.txt](https://github.com/espressif/esp-idf/files/6488109/component.mk.txt)

Please advise.

projectgus, 2021-05-17:
Hi,

Thanks for the clear explanation of the issue.

You're right, the root cause is that the ArduinoJson CMakeLists.txt file is not an ESP-IDF component CMakeLists.txt file - it's a CMake project by itself. Including a second CMake project directly inside your ESP-IDF CMake project is unfortunately not possible (it is possible using a CMake feature called "ExternalProject", but shouldn't be necessary to use that there.)

Unfortunately, due to the CMake restrictions both the project file and the component file need to be named CMakeLists.txt (unlike GNU Make build system where we had Makefile for the project and component.mk for the components).

You can add ESP-IDF support to the existing ArduinoJson CMakeLists.txt file by doing something like this (untested, this may require some tweaks):

```
cmake_minimum_required(VERSION 3.0)

if(ESP_PLATFORM)
   # Build ArduinoJson as an ESP-IDF component
   idf_component_register(INCLUDE_DIRS src)
   target_compile_definitions(${COMPONENT_LIB} ARDUINOJSON_EMBEDDED_MODE=1)
   return()
endif()

project(ArduinoJson VERSION 6.18.0)
# ... rest of original CMakeLists file goes here (the return() means it's ignored if ESP-IDF is being used)
```

(Note: the above sets ARDUINOJSON_EMBEDDED_MODE flag but I noticed the component.mk doesn't. Remove this line if you want to build the "big" version!)

qt1, 2021-05-17:
@projectgus Thanks! 

It did something.. 
I added a printout of arguments

````
if(ESP_PLATFORM)
   # Build ArduinoJson as an ESP-IDF component
   idf_component_register(INCLUDE_DIRS src)
   message(STATUS ${COMPONENT_LIB}  " " ARDUINOJSON_EMBEDDED_MODE=1)
   target_compile_definitions(${COMPONENT_LIB} ARDUINOJSON_EMBEDDED_MODE=1)
   return()
endif()
````

Now I get this error:

````
-- __idf_ArduinoJson ARDUINOJSON_EMBEDDED_MODE=1
CMake Error at components/ArduinoJson/CMakeLists.txt:11 (target_compile_definitions):
  target_compile_definitions called with invalid arguments
````

projectgus, 2021-05-17:
Apologies, I shouldn't have tried to write CMake from memory. :)

Replace the line with

```
target_compile_definitions(${COMPONENT_LIB} PUBLIC ARDUINOJSON_EMBEDDED_MODE=1)
```

qt1, 2021-05-18:
Hi, current error is:

````
CMake Error at components/ArduinoJson/CMakeLists.txt:10 (target_compile_definitions):
  target_compile_definitions may only set INTERFACE properties on INTERFACE
  targets
````

projectgus, 2021-05-18:
Oh, right. Because this is a header-only library! Change PUBLIC To INTERFACE.

Have stopped trying to do this in my head (sorry!) and actually cloned ArduinoJson into a components directory and built a CPP file that includes it. The diff that works is as follows:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index a42759e1..f349e413 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -4,6 +4,13 @@
 
 cmake_minimum_required(VERSION 3.0)
 
+if(ESP_PLATFORM)
+   # Build ArduinoJson as an ESP-IDF component
+   idf_component_register(INCLUDE_DIRS src)
+   target_compile_definitions(${COMPONENT_LIB} INTERFACE ARDUINOJSON_EMBEDDED_MODE=1)
+   return()
+endif()
+
 project(ArduinoJson VERSION 6.18.0)
 
 if(CMAKE_PROJECT_NAME STREQUAL PROJECT_NAME)
```

projectgus, 2021-05-19:
Good to hear!

closed 2021-05-19 by projectgus