espressif/esp-idf #7025: Support for BLE Object transfer service (OTS)

prasad-dot-ws, 2021-05-15:
**Support for BLE Object transfer service (OTS)**

Is there any way i can receive data over L2CAP channel? If this is currently not supported, then is there any future plans to provide support for L2CAP communication over BLE?

prasad-dot-ws, 2021-05-17:
Any feedback? @projectgus @igrr @me-no-dev

prasad-dot-ws, 2021-06-08:
Based on my findings the [command(**L2CAP_CMD_BLE_CREDIT_BASED_CONN_REQ** 0x14)](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/bt/host/bluedroid/stack/include/stack/l2cdefs.h#L44) for opening a side L2CAP channel is not implemented in `bt` component in the `ESP-IDF`. See below function from [l2c_ble.c](https://github.com/espressif/esp-idf/blob/1d7068e4be430edd92bb63f2d922036dcf5c3cc1/components/bt/host/bluedroid/stack/l2cap/l2c_ble.c#L676). Could someone from Espressif confirm this please? @Weijian-Espressif @wmy-espressif @Sushant-Espressif @dhrishi 

```
/*******************************************************************************
**
** Function         l2cble_process_sig_cmd
**
** Description      This function is called when a signalling packet is received
**                  on the BLE signalling CID
**
** Returns          void
**
*******************************************************************************/
void l2cble_process_sig_cmd (tL2C_LCB *p_lcb, UINT8 *p, UINT16 pkt_len)
{
    UINT8           *p_pkt_end;
    UINT8           cmd_code, id;
    UINT16          cmd_len;
    UINT16          min_interval, max_interval, latency, timeout;

    p_pkt_end = p + pkt_len;

    STREAM_TO_UINT8  (cmd_code, p);
    STREAM_TO_UINT8  (id, p);
    STREAM_TO_UINT16 (cmd_len, p);

    /* Check command length does not exceed packet length */
    if ((p + cmd_len) > p_pkt_end) {
        L2CAP_TRACE_WARNING ("L2CAP - LE - format error, pkt_len: %d  cmd_len: %d  code: %d", pkt_len, cmd_len, cmd_code);
        return;
    }
    
    L2CAP_TRACE_WARNING ("L2CAP - LE - command code: %d", cmd_code);

    switch (cmd_code) {
    case L2CAP_CMD_REJECT:
    case L2CAP_CMD_ECHO_RSP:
    case L2CAP_CMD_INFO_RSP:
        p += 2;
        break;
    case L2CAP_CMD_ECHO_REQ:
    case L2CAP_CMD_INFO_REQ:
        l2cu_send_peer_cmd_reject (p_lcb, L2CAP_CMD_REJ_NOT_UNDERSTOOD, id, 0, 0);
        break;

    case L2CAP_CMD_BLE_UPDATE_REQ:
        STREAM_TO_UINT16 (min_interval, p); /* 0x0006 - 0x0C80 */
        STREAM_TO_UINT16 (max_interval, p); /* 0x0006 - 0x0C80 */
        STREAM_TO_UINT16 (latency, p);  /* 0x0000 - 0x03E8 */
        STREAM_TO_UINT16 (timeout, p);  /* 0x000A - 0x0C80 */
        /* If we are a master, the slave wants to update the parameters */
        if (p_lcb->link_role == HCI_ROLE_MASTER) {
            if (min_interval < BTM_BLE_CONN_INT_MIN || min_interval > BTM_BLE_CONN_INT_MAX ||
                    max_interval < BTM_BLE_CONN_INT_MIN || max_interval > BTM_BLE_CONN_INT_MAX ||
                    latency  > BTM_BLE_CONN_LATENCY_MAX ||
                    timeout < BTM_BLE_CONN_SUP_TOUT_MIN || timeout > BTM_BLE_CONN_SUP_TOUT_MAX ||
                    /* The supervision_timeout parameter defines the link supervision timeout for the connection.
                       The supervision_timeout in milliseconds shall be large than (1 + latency) * max_interval * 2,
                       where max_interval is given in milliseconds. (See [Vol 6] Part B, Section 4.5.2).
                       supervision_timeout (mult of 10ms); conn_interval (mult of 1.25ms)
                       (max_interval * 1.25 * 2) replaced by ((max_interval * 5) >> 1).
                    */
                    ((timeout * 10) < ((1 + latency) *((max_interval * 5) >> 1))) ||
                    max_interval < min_interval) {
                l2cu_send_peer_ble_par_rsp (p_lcb, L2CAP_CFG_UNACCEPTABLE_PARAMS, id);

                L2CAP_TRACE_ERROR("slave connection parameters update failed, the parameters are out of range");

            } else {

                l2cu_send_peer_ble_par_rsp (p_lcb, L2CAP_CFG_OK, id);
                p_lcb->waiting_update_conn_min_interval = min_interval;
                p_lcb->waiting_update_conn_max_interval = max_interval;
                p_lcb->waiting_update_conn_latency = latency;
                p_lcb->waiting_update_conn_timeout = timeout;
                p_lcb->conn_update_mask |= L2C_BLE_NEW_CONN_PARAM;

                if (l2cble_start_conn_update(p_lcb) == TRUE) {
                    UINT32 time = CalConnectParamTimeout(p_lcb);
                    btu_start_timer(&p_lcb->upda_con_timer, BTU_TTYPE_L2CAP_UPDA_CONN_PARAMS, time);
                }
            }
        } else {
            l2cu_send_peer_cmd_reject (p_lcb, L2CAP_CMD_REJ_NOT_UNDERSTOOD, id, 0, 0);
        }
        break;

    case L2CAP_CMD_BLE_UPDATE_RSP: {
        UINT16 result = 0;
        STREAM_TO_UINT16(result, p); //result = 0 connection param accepted, result = 1 connection param rejected.
        UINT8 status = (result == 0) ? HCI_SUCCESS : HCI_ERR_PARAM_OUT_OF_RANGE;
        if (status != HCI_SUCCESS) {
            btu_stop_timer(&p_lcb->upda_con_timer);
            p_lcb->conn_update_mask &= ~L2C_BLE_UPDATE_PENDING;
            p_lcb->conn_update_mask &= ~L2C_BLE_UPDATE_PARAM_FULL;
            l2c_send_update_conn_params_cb(p_lcb, status);
        }
        break;
    }
    default:
        L2CAP_TRACE_WARNING ("L2CAP - LE - unknown cmd code: %d", cmd_code);
        l2cu_send_peer_cmd_reject (p_lcb, L2CAP_CMD_REJ_NOT_UNDERSTOOD, id, 0, 0);
        return;
    }
}
```

xiewenxiang, 2021-06-09:
this feature is not yet support, we will evaluate this feature internal.

prasad-dot-ws, 2021-06-09:
Thank you for the update @xiewenxiang. My attempt is to achieve a better data transfer rate when performing OTA firmware update over BLE. Do I have any other options to achieve my goal?

xiewenxiang, 2021-06-28:
At present, there is no other way. You can only read and weite the characteristic value or send indication and nitification.

prasad-dot-ws, 2021-06-28:
Currently, I'm writing to the characteristic without response. The only problem is with the iOS as it automatically negotiates MTU to 185 bytes.

xiewenxiang, 2021-06-28:
Do you think MTU size equal 185 bytes is too small? Currently ESP devices can support up to 517 bytes， is there an API at the APP side to exchange the MTU size?

prasad-dot-ws, 2021-06-28:
It's not terrible. However, Currently to transfer a 700Kb file in iOS takes about **2 mins 30 seconds**. In android, it transfers in **seconds**. The problem is iOS automatically negotiates the MTU and doesn't allow us to change it even though the ESP peripheral is supporting up to **517 bytes**. As of now, there is no API in IOS Core Bluetooth SDK to request desired MTU size. Instead, they encourage us to use the L2CAP channel for large file transfers.

In another product used L2CAP channel on nRF51 Series silicon and it worked perfectly on iOS devices without any packet limitations. I think with iOS this is the only way of achieving faster data rates.

mickeyl, 2022-05-27:
@prasad-dot-ws I‘m also looking to directly use an L2CAP channeL for transmitting to iOS, here, using an ESP32-C3. Did you get any futher with this?

prasad-dot-ws, 2022-05-28:
> @prasad-dot-ws I‘m also looking to directly use an L2CAP channeL for transmitting to iOS, here, using an ESP32-C3. Did you get any futher with this?

@mickeyl Sorry, no. I dropped the idea of using L2CAP with ESP silicon as it wasn't supported and I couldn't find much time to implement it by myself. I changed my design to the nRF51 series due to this issue.

mickeyl, 2022-05-29:
@xiewenxiang Before I lose time trying what still is not supported, could you give us an update on the l2cap situation, please?