espressif/esp-idf #7026: NimBLE appears broken in v4.2.1

DavidJRobertson, 2021-05-15:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it): v4.2.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:  Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

After updating from idf v4.2 to v4.2.1, my Android app was no longer able to connect to the device (which uses NimBLE). Additionally, android BLE debug tools (like nRF Connect and LightBlue) also failed to connect.

I then tried the idf examples to see if these worked properly (i.e. to eliminate problems in my own firmware code). Using one of the Bluedroid examples (`examples/bluetooth/bluedroid/ble/gatt_server`) worked as expected, but the corresponding NimBLE example (`examples/bluetooth/nimble/bleprph`) failed in exactly the same way as my own firmware did.

### Expected Behavior

Android device (running e.g. nRF Connect) can connect to esp32 running `examples/bluetooth/nimble/bleprph` example, discover the services, and read characteristics etc.

### Actual Behavior

Android device (running e.g. nRF Connect) connects to the esp32 running `examples/bluetooth/nimble/bleprph` example, but the service discovery appears to fail after a delay of a few seconds, and the device is disconnected with error code 0x8 (GATT CONN TIMEOUT).

## Debug Logs

```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7068
load:0x40078000,len:13212
ho 0 tail 12 room 4
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (31) boot: ESP-IDF v4.2.1 2nd stage bootloader
I (31) boot: compile time 22:33:01
I (31) boot: chip revision: 3
I (34) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 3, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0d2c4 ( 53956) map
I (134) esp_image: segment 1: paddr=0x0001d2ec vaddr=0x3ffbdb60 size=0x02d2c ( 11564) load
I (139) esp_image: segment 2: paddr=0x00020020 vaddr=0x400d0020 size=0x5a470 (369776) map
0x400d0020: _stext at ??:?

I (282) esp_image: segment 3: paddr=0x0007a498 vaddr=0x3ffc088c size=0x006b8 (  1720) load
I (283) esp_image: segment 4: paddr=0x0007ab58 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/david/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (289) esp_image: segment 5: paddr=0x0007af64 vaddr=0x40080404 size=0x15d44 ( 89412) load
I (348) boot: Loaded app from partition at offset 0x10000
I (349) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (353) cpu_start: Application information:
I (357) cpu_start: Project name:     bleprph
I (362) cpu_start: App version:      v4.2.1
I (367) cpu_start: Compile time:     May 15 2021 22:32:56
I (373) cpu_start: ELF file SHA256:  11949e2ce2dfca86...
I (379) cpu_start: ESP-IDF:          v4.2.1
I (384) cpu_start: Starting app cpu, entry point is 0x40081910
0x40081910: call_start_cpu1 at /home/david/esp/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (395) heap_init: Initializing. RAM available for dynamic allocation:
I (402) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (408) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (414) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (420) heap_init: At 3FFC3B70 len 0001C490 (113 KiB): DRAM
I (426) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (432) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (439) heap_init: At 40096148 len 00009EB8 (39 KiB): IRAM
I (445) cpu_start: Pro cpu start user code
I (463) spi_flash: detected chip: gd
I (464) spi_flash: flash io: dio
W (464) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (474) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (520) BTDM_INIT: BT controller compile version [a4c513b]
I (520) system_api: Base MAC address is not set
I (520) system_api: read default base MAC address from EFUSE
I (530) phy_init: phy_version 4660,0162888,Dec 23 2020
I (870) NimBLE_BLE_PRPH: BLE Host Task Started
I (870) uart: queue free spaces: 8
Device Address: a8:03:2a:f3:68:62
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
connection established; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=a8:03:2a:f3:68:62 our_id_addr_type=0 our_id_addr=a8:03:2a:f3:68:62 peer_ota_addr_type=1 peer_ota_addr=6e:22:1c:a0:83:d3 peer_id_addr_type=1 peer_id_addr=6e:22:1c:a0:83:d3 conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

connection updated; status=0 handle=0 our_ota_addr_type=0 our_ota_addr=a8:03:2a:f3:68:62 our_id_addr_type=0 our_id_addr=a8:03:2a:f3:68:62 peer_ota_addr_type=1 peer_ota_addr=6e:22:1c:a0:83:d3 peer_id_addr_type=1 peer_id_addr=6e:22:1c:a0:83:d3 conn_itvl=6 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

disconnect; reason=520 handle=0 our_ota_addr_type=0 our_ota_addr=a8:03:2a:f3:68:62 our_id_addr_type=0 our_id_addr=a8:03:2a:f3:68:62 peer_ota_addr_type=1 peer_ota_addr=6e:22:1c:a0:83:d3 peer_id_addr_type=1 peer_id_addr=6e:22:1c:a0:83:d3 conn_itvl=6 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
```

nRF Connect log:
```
nRF Connect, 2021-05-15
nimble-bleprph (A8:03:2A:F3:68:62)
I    22:55:52.142    [Server] Server started
V    22:55:52.143    [Server] No services
V    22:55:52.317    Connecting to A8:03:2A:F3:68:62...
D    22:55:52.317    gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D    22:55:53.119    [Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I    22:55:53.119    [Server] Device with address A8:03:2A:F3:68:62 connected
D    22:55:53.128    [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I    22:55:53.128    Connected to A8:03:2A:F3:68:62
V    22:55:53.225    Discovering services...
D    22:55:53.225    gatt.discoverServices()
D    22:55:53.244    [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I    22:55:53.656    Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D    22:56:03.829    [Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E    22:56:03.829    Error 8 (0x8): GATT CONN TIMEOUT
I    22:56:03.829    Disconnected
D    22:56:03.830    [Server callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I    22:56:03.830    [Server] Device disconnected
D    22:56:03.870    [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
```

justintconroy, 2021-05-17:
I'm also experiencing this issue. Tried some of the same things, with the same results. Service Discovery just isn't working anymore.

RahulEspressif, 2021-05-18:
Hi @justintconroy  @DavidJRobertson ,  Thanks for reporting this issue.  

We are internally working on fixing this. In meantime, you can disable Host Flow control ( menuconfig ---> Nimble Options --> Deselect "Enable Host Flow Control" ) . This is just a workaround to unblock you as we provide correct fix for this issue . 

Thanks,
 Rahul

justintconroy, 2021-05-18:
I tried this out today. With host flow control disabled, service discovery, notifications and writes seem to be working. Attempting to do any read operations looks like it does nothing at all, though.

AxelLin, 2021-05-21:
@justintconroy 
You may want to check with recently release/v4.2 branch update which includes fixes: 1c7340d229f5c398ae35d6335065de38da7aa3f1 and b3e1829f1058afa8728e6bcd9c59a19b99e5972f

justintconroy, 2021-05-21:
I'm using the tag `v4.2`. It's only tag `v4.2.1` that isn't working for me. I can live with it, there were just some other fixes in 4.2.1 that I wanted, too. 😸

dhrishi, 2021-05-21:
>I'll wait for v4.2.2 to hopefully get all I want.

@justintconroy Yes. That's right. The fix should be available in v4.2.2. We will try to release it soon.

AxelLin, 2021-05-22:
> I'm using the tag `v4.2`. It's only tag `v4.2.1` that isn't working for me. I can live with it, there were just some other fixes in 4.2.1 that I wanted, too. 
> 
> I'll wait for `v4.2.2` to hopefully get all I want.

Just in case if it does not fix your issue and you didn't verify the fix, you will have to wait yet another release to get real fix.
I would suggest you to test the fix in release/v4.2 branch and report test result.

ChrisEAlfred, 2021-06-10:
We have also observed that Nimble BLE does not connect on the following branches:

- develop
- v4.2
- v4.3

The only versions we have found that work are:

- v4.1
- v4.1-beta2

bencbyers, 2021-06-15:
@ChrisEAlfred was that the v4.2 tag that you used? If so, @AxelLin is that what you were saying we should test first in order to verify the fix works? Or is this deciding that the fix didn't work? We are using the workaround of disabling Host Flow Control in the nimBLE and that is working for us now, but anxiously following this.

ChrisEAlfred, 2021-06-16:
Yes, they are all tag names.

dhrishi, 2021-06-18:
@DavidJRobertson @justintconroy @ChrisEAlfred @bencbyers  ESP-IDF v4.2.2 is released that has the fix for NimBLE and it is functional as expected. You can find it here: https://github.com/espressif/esp-idf/releases/tag/v4.2.2 

Please give it a try and let us know in case of any issues

DavidJRobertson, 2021-06-20:
@dhrishi The v4.2.2 tag does not even build!

```
In file included from /home/david/esp/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:19,
                 from /home/david/esp/esp-idf/components/wpa_supplicant/src/utils/common.h:423,
                 from /home/david/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:17:
/home/david/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c: In function 'wpa_supplicant_gtk_in_use':
/home/david/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:884:120: error: 'hw_keyidx' undeclared (first use in this function); did you mean 'keyidx'?
             wpa_printf(MSG_DEBUG, "GTK %d is already in use in entry %d, it may be an attack, ignore it.", gd->keyidx, hw_keyidx);
                                                                                                                        ^~~~~~~~~
/home/david/esp/esp-idf/components/log/include/esp_log.h:329:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/home/david/esp/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:62:40: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define wpa_printf(level,fmt, args...) ESP_LOG_LEVEL_LOCAL(level, TAG, fmt, ##args)
                                        ^~~~~~~~~~~~~~~~~~~
/home/david/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:884:13: note: in expansion of macro 'wpa_printf'
             wpa_printf(MSG_DEBUG, "GTK %d is already in use in entry %d, it may be an attack, ignore it.", gd->keyidx, hw_keyidx);
             ^~~~~~~~~~
/home/david/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:884:120: note: each undeclared identifier is reported only once for each function it appears in
             wpa_printf(MSG_DEBUG, "GTK %d is already in use in entry %d, it may be an attack, ignore it.", gd->keyidx, hw_keyidx);
                                                                                                                        ^~~~~~~~~
/home/david/esp/esp-idf/components/log/include/esp_log.h:329:137: note: in definition of macro 'ESP_LOG_LEVEL'
         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag, ##__VA_ARGS__); } \
                                                                                                                                         ^~~~~~~~~~~
/home/david/esp/esp-idf/components/wpa_supplicant/include/utils/wpa_debug.h:62:40: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define wpa_printf(level,fmt, args...) ESP_LOG_LEVEL_LOCAL(level, TAG, fmt, ##args)
                                        ^~~~~~~~~~~~~~~~~~~
/home/david/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:884:13: note: in expansion of macro 'wpa_printf'
             wpa_printf(MSG_DEBUG, "GTK %d is already in use in entry %d, it may be an attack, ignore it.", gd->keyidx, hw_keyidx);
             ^~~~~~~~~~
[295/1196] Building C object esp-idf/wpa_supplicant/CMakeFiles/__idf_wpa_supplicant.dir/src/esp_supplicant/esp_wps.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

DavidJRobertson, 2021-06-20:
Update: I ran menuconfig, didn't change anything, and it now mysteriously builds. 

v4.2.2 appears to resolve my NimBLE issue.

closed 2021-06-20 by DavidJRobertson

bencbyers, 2021-06-21:
This also built for me and fixed the nimBLE. Thanks!