espressif/esp-idf #7027: Issues with mdns server and now with websocket server too

baldhead69, 2021-05-16:
Hi,

My mdns server and websocket server were working properly for a long time.
I ended up putting the project aside for few months because i am waiting for the launch of the esp32-s3 modules.

When i came back to the project i decided to do some tests, and the mdns server was behaving strangely.
Using wireshark i saw that the mdns server sent the ad for about 6 times and stopped(after manual reset).
The first 4 times every 1 seconds and the last 2 times every 2 seconds and stopped.
I configured mdns on menuconfig so that it would send the notification to the network every 1 second.

During this period we moved to an new internet provider and also to a new wifi router.
I thought the problem could be the new wifi router, but i haven't been able to find the problem yet.

So i decided to update the esp-idf.
I updated to v4.4-dev-1254-g639e7ad49.

After esp-idf update the mdns problem still persists.
And now the websocket server also bugged.

Using the idf.py monitor, i can see that websocket server start ok, but when i connect to the server through my android application, the following message appears:
W (00:00:08.807) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now
W (00:00:08.816) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now

Before the update of the esp idf, the websocket server worked ok, including "ESP-IDF Release v4.2.1".

Some suggestion on how to resolve the mdns server and now the websocket server bug ?

Note: i changed the httpd priority from "config.httpd.task_priority = tskIDLE_PRIORITY + 5" to "config.httpd.task_priority = configMAX_PRIORITIES".

@david-cermak

Thank's.

david-cermak, 2021-05-19:
Hi @baldhead69 

Could you please share more details:
* About your application: is it based on some example? how is the ws_transport handled? Does it use TLS?
* Share debug logs

The above warnings tell us, that an error occurred during tcp receive, enabling debug log would tell us, what kind of error it was (probably some kind of disconnection). Or maybe there already was an error on the `warn` level:
```cpp
ESP_LOGW(TAG, LOG_FMT("error in %s : %d"), ctx, errno);
```

Cannot help much without more information, as now this looks to me like you've changed your network configuration, your project stopped working, so you switched to `v4.4-dev` which didn't help so raised this issue.

If your project is based on some of the `httpd-ws` examples in IDF, very unlikely there'd be a regression, as they are continuously tested in the CI. We have some pending fixed to `httpd-ws`, but nothing that would cause interrupted connections.

baldhead69, 2021-05-19:
Hi @david-cermak ,

I already uninstalled the v4.4-dev-1254-g639e7ad49.
I installed v4.2.1 now.

I dont switch the versions on my computer.
I always need to edit my enviroment variables to esp-idf to work.

baldhead69, 2021-05-19:
Hi @david-cermak ,

Follow the attached responses:

> Hi @baldhead69
> 
> Could you please share more details:
> 
> * About your application: is it based on some example? how is the ws_transport handled? Does it use TLS?
**Yes, it use TLS.**
> * Share debug logs
> 
> The above warnings tell us, that an error occurred during tcp receive, enabling debug log would tell us, what kind of error it was (probably some kind of disconnection). Or maybe there already was an error on the `warn` level:
**The received data was all corrupted.**
> 
> ```c++
> ESP_LOGW(TAG, LOG_FMT("error in %s : %d"), ctx, errno);
> ```
> 
> Cannot help much without more information, as now this looks to me like you've changed your network configuration, your project stopped working, so you switched to `v4.4-dev` which didn't help so raised this issue.
**In version v4.2.1 the websocket server are OK.**
> 
> If your project is based on some of the `httpd-ws` examples in IDF, very unlikely there'd be a regression, as they are continuously tested in the CI. We have some pending fixed to `httpd-ws`, but nothing that would cause interrupted connections.
**The received data was all corrupted.**

david-cermak, 2021-05-20:
With the little info you've provided, we could only guess that the TLS handshake completed successfully and you're getting corrupted received data. No idea about the application, so assuming that you've registered a `ws_handler()` that getting the data, which are not exactly the ws-transport data, but something else. (this is strange, because previously you've reported interruption of the received data: `: Failed to read header byte (socket FD invalid), closing socket now` )

Listing the commits between [v4.2.1...v4.4-dev-1254](https://github.com/espressif/esp-idf/compare/v4.2.1...v4.4-dev-1254-g639e7ad49) I would fortune my guess to  this one:

* a1d5cfc260 example: Identify the callback whether is hanshake or frame-receive by req->method

Now the uri handler gets called immediately after the handshake. In the handler we can identify that this was the handshake by checking req->method as it is still a GET from the first part of the handshake the client has sent. On a normal websocket-frame-receive (when normally a websocket uriHandler gets called) this field is set to 0 Closes https://github.com/espressif/esp-idf/issues/6597

Could you please try to check for the `req->method` in your ws-handler and make sure the received data are not part of the WS handshake?

```patch
 static esp_err_t ws_handler(httpd_req_t *req)
 {
+    if (req->method == HTTP_GET) {
+        ESP_LOGI(TAG, "Handshake done, the new connection was opened");
+        return ESP_OK;
+    }
     ...
}
```

baldhead69, 2021-05-20:
Hi @david-cermak,

I dont have v4.4-dev-1254-g639e7ad49 installed anymore in my computer, only v4.2.1.

In v4.2.1 the websocket are working "perfectly" and in v4.4-dev-1254-g639e7ad49 the data was corrupted.

I dont change my cell phone app with both versions of esp-idf tests, so i suppose the bug is in the version v4.4-dev-1254-g639e7ad49.

I had the two problems in v4.4-dev-1254-g639e7ad49:
W (00:00:08.807) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now
W (00:00:08.816) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now

and 

corruped data in websocket payload too.

david-cermak, 2021-05-20:
> suppose the bug is in the version v4.4-dev-1254

Yes, there could indeed be a bug in the latest IDF, but to find it, we would need your help. We have many applications which work *perfectly* on the latest `v4.4-dev`, and you seem to have one that fails (and used to work on `v4.2`!). Could you please share some portions of your app or at least a minimum project that could be used to reproduce the issue?

Alvin1Zhang, 2021-05-31:
@baldhead69 Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

baldhead69, 2021-07-14:
Hi @Alvin1Zhang ,

Apparently all is working in v4.2.1 and not working in version v4.4-dev-1254-g639e7ad49.

When will esp32-s3 be made available to the general public ?

Could you provide me with samples of development boards with esp32-s3 ?

Thank's.

david-cermak, 2021-07-26:
> Apparently all is working in v4.2.1 and not working in version v4.4-dev-1254-g639e7ad49.

@baldhead69 Could you please share more details and a small portion of your application code to reproduce the issue as asked in https://github.com/espressif/esp-idf/issues/7027#issuecomment-845299400 ?

Alvin1Zhang, 2021-07-27:
Thanks for reporting and sorry for late reply, regarding ESP32-S3 sample boards, please contact https://www.espressif.com/en/contact-us/get-samples, thanks.

baldhead69, 2021-07-27:
Hi @Alvin1Zhang ,

I send a "sales questions".

Thank's.

david-cermak, 2021-07-27:
@baldhead69 Could you please share more details on this issue? The title says:
```
Issues with mdns server and now with websocket server too
```
but no mdns problem has been reported and the websocket server seems to work correctly on my end.

baldhead69, 2021-07-27:
Hi @david-cermak ,

What you need ?

Full project ?

I can sent to a private email.

david-cermak, 2021-07-27:
@baldhead69 Just the piece of code that sets the https server up in websocket mode, if that's enough to reproduce the problem. I'd believe it would be about 20 lines of code calling the http-server API. This could be shared here.

It would be also great to remove the `mdns` part from the issue title, not only to keep reporting clean, but also for sake of other users who might search for the "open" issues.

> I can sent to a private email.

That might be helpful, too, but let's try with the snippet first. If you're willing to share the project via email, you can send it to my git address (present in any of [my commits](https://github.com/espressif/esp-idf/commits?author=david-cermak) ), e.g.:
```bash
cd esp-idf
git fetch
git show dbbf21 | grep Author
```

baldhead69, 2021-07-27:
@david-cermak 

`
static httpd_handle_t server = NULL;

static void start_webSocket_secure_server()
{    
    httpd_ssl_config_t config = HTTPD_SSL_CONFIG_DEFAULT();    // Considerar usar wolfssl lib, pois usa menos memoria ram(metade da lib mbedTls). 

    config.httpd.core_id = PRO_CPU_NUM;
    config.httpd.max_open_sockets = open_sockets_max;  // 5
    config.httpd.task_priority =  tskIDLE_PRIORITY + 5;  // tskIDLE_PRIORITY + 20;  
    //config.httpd.open_fn = opened_socket_handler;
    config.httpd.close_fn = closed_socket_handler;
    //config.transport_mode = HTTPD_SSL_TRANSPORT_INSECURE;    // da para testar com ws://192.168.0.8/ 
    config.httpd.lru_purge_enable = false;    // The lib limits to "max_open_sockets" openned sockets, but closes one socket and opens another if "lru_purge_enable = true".  

#if enable_ws_Keep_Alive
    //UBaseType_t keepAliveTaskPriority = config.httpd.task_priority + 1;    
    UBaseType_t keepAliveTaskPriority = tskIDLE_PRIORITY + 4;
#endif

    ESP_LOGI(TAG, "Starting WebSocket Secure Server");

       
    extern const unsigned char cacert_pem_start[] asm("_binary_cacert_pem_start");
    extern const unsigned char cacert_pem_end[]   asm("_binary_cacert_pem_end");
    config.cacert_pem = cacert_pem_start;
    config.cacert_len = cacert_pem_end - cacert_pem_start;

    extern const unsigned char prvtkey_pem_start[] asm("_binary_prvtkey_pem_start");
    extern const unsigned char prvtkey_pem_end[]   asm("_binary_prvtkey_pem_end");
    config.prvtkey_pem = prvtkey_pem_start;
    config.prvtkey_len = prvtkey_pem_end - prvtkey_pem_start;

    esp_err_t ret = httpd_ssl_start(&server, &config);
    
    if ( ret != ESP_OK )
    {
        ESP_LOGE(TAG, "Failed to start the server!");
        printf("\nError->httpd_ssl_start: %d\n", ret);
        return;
    }
    			
    printf("\nserver = %p\n\n", server );

    // URI handler for websocket client(s) message reception.
    static const httpd_uri_t get_clients_message =
    {
        .uri       = "/",  
        .method    = HTTP_GET,
        .handler   = get_clients_message_handler,
        .user_ctx  = NULL,
        .is_websocket = true
    };
    httpd_register_uri_handler(server, &get_clients_message);
   
    
#if enable_ws_Keep_Alive
    ret = ESP_FAIL;
    ret = ws_init_keep_alive_task( 1000, ws_clients_list.max_list_size, 30, keepAliveTaskPriority, server );

    if ( ret != ESP_OK )
    {
        printf("\nError->init_Ws_Keep_Alive_Task: %d\n", ret);     
        return;
    }
#endif
}
`

david-cermak, 2021-07-28:
@baldhead69 Thanks for sharing the code. I've been testing it and found no issues. But I used this *very simple handler as your `get_clients_message_handler()`:
```cpp
static esp_err_t get_clients_message_handler(httpd_req_t *req)
{
    if (req->method == HTTP_GET) {
        ESP_LOGI(TAG, "Handshake done, the new connection was opened");
        return ESP_OK;
    }
    httpd_ws_frame_t ws_pkt;
    uint8_t *buf = NULL;
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));

    // First receive the full ws message
    /* Set max_len = 0 to get the frame len */
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_recv_frame failed to get frame len with %d", ret);
        return ret;
    }
    ESP_LOGI(TAG, "frame len is %d", ws_pkt.len);
    if (ws_pkt.len) {
        /* ws_pkt.len + 1 is for NULL termination as we are expecting a string */
        buf = calloc(1, ws_pkt.len + 1);
        if (buf == NULL) {
            ESP_LOGE(TAG, "Failed to calloc memory for buf");
            return ESP_ERR_NO_MEM;
        }
        ws_pkt.payload = buf;
        /* Set max_len = ws_pkt.len to get the frame payload */
        ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "httpd_ws_recv_frame failed with %d", ret);
            free(buf);
            return ret;
        }
    }
    if (ws_pkt.type == HTTPD_WS_TYPE_TEXT) {
        ESP_LOGI(TAG, "Received packet with message: %s", ws_pkt.payload);
        ret = httpd_ws_send_frame(req, &ws_pkt);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "httpd_ws_send_frame failed with %d", ret);
        }
        ESP_LOGI(TAG, "ws_handler: httpd_handle_t=%p, sockfd=%d, client_info:%d", req->handle,
                 httpd_req_to_sockfd(req), httpd_ws_get_fd_info(req->handle, httpd_req_to_sockfd(req)));
        free(buf);
        return ret;
    }
    free(buf);
    return ESP_OK;
}
```

Could you please share your ws handler, too? How do you use the `httpd_ws_recv_frame()` API? There were some changes to allow dynamic payload reception, but the old way should still be supported.

baldhead69, 2021-07-28:
Hi @david-cermak ,

I sent you an email.

baldhead69, 2021-07-28:
Hi @david-cermak ,

The mdns stop of sending multicast packets after some time after esp32 turning on(some hours).

david-cermak, 2021-07-29:
Hi @baldhead69 

Thanks for sharing the project. I've made it compilable supplying some missing symbols (like the list of clients and the array of function pointers). And was able to reproduce an issue (not sure though if that was the issue you reported).
I was able to fix it quickly supplying a payload pointer:
```patch
static esp_err_t get_clients_message_handler(httpd_req_t *req)
{
+    ws_pkt.payload = buf;
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));  // sizeof(buf) = 128 bytes

    if (ret != ESP_OK)
    {
#if webSocketDebug
        printf( "\nhttpd_ws_recv_frame failed with: %d code.\n", ret );
#endif
        return ret;
    }
```

After this change the project seemed to work correctly, didn't run the custom functions, but got the expected reply over ws transport: `Given **it !!!`

As mentioned earlier, there might be some minor inconsistencies of using the ws api between releases, but this was actually present on `v4.2` as well:

https://github.com/espressif/esp-idf/blob/release/v4.2/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L73

Just a note, that if I understand your project correctly, you're implementing manually what has been implemented in this example on `v4.3` and later, including a keep-alive module (it might be easier, as safer, too, to migrate to this type of project)

https://github.com/espressif/esp-idf/tree/release/v4.3/examples/protocols/https_server/wss_server

---
As for the mdns, we've been reported an issue with avahi https://github.com/espressif/esp-idf/issues/7124 causing the mdns library to stop responding (only with avahi and only after the TTL expires), but this is fixed already and backported to all release branches.

baldhead69, 2021-07-30:
@david-cermak,

I don't think you reproduced the problem correctly.

The "ws_pkt.payload = buf;" i initialize into another function.

baldhead69, 2021-07-31:
@david-cermak ,

With regards to Mdns, i checked again that esp32 was not sending the multicast packet to the network and i checked in the router list that esp32 was connected.

baldhead69, 2021-08-03:
@david-cermak

I installed de esp-idf version: v4.4-dev-2359-g58022f859, and the websocket bug yet exist.

```
static esp_err_t get_clients_message_handler(httpd_req_t *req)
{      
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));  // sizeof(buf) = 128 bytes 
   
    printf( "\nhttpd_ws_recv_frame ret: %d code.\n", ret );
    int fd = httpd_req_to_sockfd(req);
    printf("Packet Socket: %d\n", fd );
    
    printf("Packet type: %d\n", ws_pkt.type);

    switch ( ws_pkt.type ) 
    {
        case HTTPD_WS_TYPE_CONTINUE:  printf("Packet type:   CONTINUE");   break;
        case HTTPD_WS_TYPE_TEXT:            printf("Packet type:   TEXT");             break;   
        case HTTPD_WS_TYPE_BINARY:        printf("Packet type:   BINARY");        break;
        case HTTPD_WS_TYPE_CLOSE:          printf("Packet type:   CLOSE");          break;
        case HTTPD_WS_TYPE_PING:            printf("Packet type:   PING");            break;
        case HTTPD_WS_TYPE_PONG:          printf("Packet type:   PONG");           break;
                                            default:          printf("Packet type:   UNKNOWN");  break;
    }
    
    printf("\n");

    ws_pkt.payload[ws_pkt.len] = '\0';    // printf("Got packet with message: %s\n", frame->payload);

    printf("Packet lenght: %d\n", ws_pkt.len);
    
    
    if( ws_pkt.type == HTTPD_WS_TYPE_TEXT )
    {
        printf("Packet messag: %s\n", ws_pkt.payload);    
    }
    else if( ws_pkt.type == HTTPD_WS_TYPE_BINARY )
    {
        printf("Packet messag BIN:\n");
        
        for( int i = 0 ; i < ws_pkt.len ; i++ )
        {        
            printf("buf[%d] = %u\n", i, ws_pkt.payload[i] );
        }
    }
    else
    {
        printf("\n\nPAYLOAD:\n\n");

        for( uint32_t i = 0 ; i < ws_pkt.len ; i++ )
        {
            printf("ws_pkt.payload[%u] = %u\n", i, ws_pkt.payload[i] );   
        }
        
        printf("\n\n");
    }    
     
    return ESP_OK;
   }

```

```
httpd_ws_recv_frame ret: 0 code.
Packet Socket: 55
Packet type: 0
Packet type:   CONTINUE
Packet lenght: 2

PAYLOAD:

ws_pkt.payload[0] = 50
ws_pkt.payload[1] = 0

httpd_ws_recv_frame ret: 0 code.
Packet Socket: 55
Packet type: 0
Packet type:   CONTINUE
Packet lenght: 2

PAYLOAD:

ws_pkt.payload[0] = 0
ws_pkt.payload[1] = 239
```

baldhead69, 2021-08-05:
Some suggestion ????

baldhead69, 2021-08-06:
@Alvin1Zhang

Alvin1Zhang, 2021-08-06:
@baldhead69 Thanks for sharing updates, sorry for replying late, we are still investigate this issue, will update once we have any progress. Thanks.

baldhead69, 2021-08-14:
@Alvin1Zhang,

What is the last stable version of esp-idf ?

In this last stable version, does the WSS (WebSocket Secure) server work ?

Thank's.

david-cermak, 2021-08-16:
@baldhead69 

> The "ws_pkt.payload = buf;" i initialize into another function.

Sorry, didn't see it anywhere in the shared code, and that was the only trouble I experienced.

> With regards to Mdns, i checked again that esp32 was not sending the multicast packet

Here again, we would need more details. Could you please share the packet log?

> Packet type:   CONTINUE

That could be the trouble as the wss server doesn't support fragmented messages, as stated in the docs:

https://github.com/espressif/esp-idf/blob/release/v4.3/examples/protocols/http_server/ws_echo_server/README.md#how-to-use-example

> Please note that the WebSocket HTTP server does not automatically fragment messages.

But that was never supported and doesn't explain why the server *works* with your application on v4.2. Some timing of the other side changed? Could you please tell us something about your protocol? From what I read from your code it looks like you're just calling functions based on the first value in your payload and your frames are relatively small, so the other side should not fragment the packets.

> In this last stable version, does the WSS (WebSocket Secure) server work ?

The WSS server application works on the latest master (as stated above) and is automatically tested on every commit to master. The test is available here: https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/wss_server/wss_server_example_test.py (this tests also the keep-alive mechanism and checks if client gets disconnected if doesn't respond to pings)

As said before, we don't see any problem with the WSS server. I was trying to reproduce your issue with the supplied code but with little success. So far I could conclude that this could be either
1) a bug in esp-idf httpd which only occurs with specific packet sequence sent by the client app
2) some sort of packet fragmentation happening which is not supported
3) some memory corruption in your application
In any case, I would need your help to tell which of these is more likely happening. Could you please share a packet capture with the debug log from the board so we can match the packets with the received data?

baldhead69, 2021-08-16:
> @baldhead69
>
>
> Sorry, didn't see it anywhere in the shared code, and that was the only trouble I experienced.
>
>
> Here again, we would need more details. Could you please share the packet log?

Apparently with esp-idf version v4.4-dev-2359-g58022f859 the mdns is working.
You had commented that there were problems with mdns and avahi.
May be in this esp-idf version this bug has been fixed and also solved my problem.
 
> That could be the trouble as the wss server doesn't support fragmented messages, as stated in the docs: [...]

My client side hasn't changed anything.
The websocket server are corrupting the data.
The correct would be "HTTPD_WS_TYPE_BINARY" and not "HTTPD_WS_TYPE_CONTINUE". 

> But that was never supported and doesn't explain why the server _works_ with your application on v4.2. Some timing of the other side changed? Could you please tell us something about your protocol? [...]

baldhead69, 2021-08-16:
I disabled the TLS and still the bug remains. 
```
httpd_ssl_config_t config = HTTPD_SSL_CONFIG_DEFAULT();  
config.transport_mode = HTTPD_SSL_TRANSPORT_INSECURE; 
```

david-cermak, 2021-08-17:
> The packets are small. The client are not fragmenting the packets.

Then these two options remain
1) bug in the IDF's httpd ws
2) memory corruption in your application

> The websocket server are corrupting the data.

It's very hard to tell, as the application and the IDF are linked together and there's always some interaction. I've played with your setup quite a lot already and haven't found any issue.

> Maybe a spiram/cache problem ?!

very unlikely, as it seems you're able to reproduce this issue quite frequently.

> Apparently with esp-idf version v4.4-dev-2359-g58022f859 the mdns is working.

Thank you for this info. at least we know that the mdns part works!

---

With that said we would have to find out where exactly the memory corruption takes place. Here are my suggestions:
* There are not some many commits between the IDF version that used to work for you and the `v4.4-dev-2359` in the `httpd` component. Could you please try to `git bisect` the offending commit?
* Seems like the memory gets corrupted in the ws payload buffer, it's a good start so you can simply place a watchpoint and see who's responsible for corrupting the data. Could you please try to follow these guidelines https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#finding-heap-corruption and check for the memory corruption? 

If I was able to reproduce your issue, I could check the two points above myself.

* You've also mentioned (in another issue) that you keep serving both ws and http clients, is it correct? Haven't seen that in your sample application, maybe that could be a problem. Could you please share more details on how you setup your httpd with different handlers for http and ws? I'd try to play with that as well to see if any kind of issue could be found.

baldhead69, 2021-08-17:
> * There are not some many commits between the IDF version that used to work for you and the `v4.4-dev-2359` in the `httpd` component. Could you please try to `git bisect` the offending commit?
> * [...]

I will try tomorrow.
 
> * You've also mentioned (in another issue) that you keep serving both ws and http clients, is it correct? Haven't seen that in your sample application, maybe that could be a problem. Could you [...]

I am using only websocket now.
Websocket running on browser: The Javascript running on chrome browser opened multiple sockets, not just websocket socket.
Websocket running on android app: only websocket socket.

david-cermak, 2021-08-17:
> I will try tomorrow.

Thanks!

> I am using only websocket now.
Websocket running on browser: The Javascript running on chrome browser opened multiple sockets, not just websocket socket.
Websocket running on android app: only websocket socket.

Is the issue reproducible with the android app only or need to run the javascript app in the browser? That could be a clue, as some browsers use speculative connections. Still don't understand what could be wrong, but will give it a try and stress test with opening many connections.

One more idea, could you please try to move the buffer variable from the global space to the automatic variables of the handler? Just to see if anything changed, the handlers are be still called from the `httpd` context, though, one after another.

```patch
@@ -19,7 +19,6 @@
 // Number of functions in the function pointer array.
 static const uint32_t web_comm_t_array_size_t = sizeof(func_pointer) / sizeof(func_pointer[0]);
 
-static uint8_t buf[128] = { 0 };
 static httpd_ws_frame_t ws_pkt;
 
 
@@ -29,6 +28,7 @@
 
 static esp_err_t get_clients_message_handler(httpd_req_t *req)
 {      
+    uint8_t buf[128] = { 0 };
     esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));  // sizeof(buf) = 128 bytes 
 
     if (ret != ESP_OK)
````

baldhead69, 2021-08-18:
Hi @david-cermak ,

I did some tests and found that the websocket server receives the data packet corrupted, but i am able to send the data correctly to the client using the function:
"httpd_queue_work( )";  // update the state of the client(s).

baldhead69, 2021-08-18:
> Is the issue reproducible with the android app only or need to run the javascript app in the browser? That could be a clue, as some browsers use speculative connections. Still don't understand what could be wrong, but will give it a try and stress test with opening many connections.

I think it's unnecessary for you to do this.
The problem is happening in both the browser and the android app.

> One more idea, could you please try to move the buffer variable from the global space to the automatic variables of the handler? Just to see if anything changed, the handlers are be still called from the `httpd` context, though, one after another.
> 
> ```diff
> @@ -19,7 +19,6 @@
>  // Number of functions in the function pointer array.
>  static const uint32_t web_comm_t_array_size_t = sizeof(func_pointer) / sizeof(func_pointer[0]);
>  
> -static uint8_t buf[128] = { 0 };
>  static httpd_ws_frame_t ws_pkt;
>  
>  
> @@ -29,6 +28,7 @@
>  
>  static esp_err_t get_clients_message_handler(httpd_req_t *req)
>  {      
> +    uint8_t buf[128] = { 0 };
>      esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));  // sizeof(buf) = 128 bytes 
>  
>      if (ret != ESP_OK)
> ```

```
static esp_err_t get_clients_message_handler(httpd_req_t *req)
{   
    httpd_ws_frame_t pkt;
    uint8_t bu[128] = { 0 };

    memset(&pkt, 0, sizeof(httpd_ws_frame_t));
    pkt.payload = bu;

    esp_err_t ret = httpd_ws_recv_frame(req, &pkt, sizeof(bu)); 

   
    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("httpd_ws_recv_frame ret: %d code."), ret );
    int fd = httpd_req_to_sockfd(req);
    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet Socket: %d"), fd );
    
    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type: %d"), pkt.type );

    switch( pkt.type ) 
    {
        case HTTPD_WS_TYPE_CONTINUE: ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type:   CONTINUE")); break;
        case HTTPD_WS_TYPE_TEXT:     ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type:   TEXT"));     break;   
        case HTTPD_WS_TYPE_BINARY:   ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type:   BINARY"));   break;
        case HTTPD_WS_TYPE_CLOSE:    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type:   CLOSE"));    break;
        case HTTPD_WS_TYPE_PING:     ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type:   PING"));     break;
        case HTTPD_WS_TYPE_PONG:     ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type:   PONG"));     break;
                        default:     ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet type:   UNKNOWN"));  break;
    }
    
    pkt.payload[pkt.len] = '\0';   

    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet lenght: %d"), pkt.len );
    
    
    if( pkt.type == HTTPD_WS_TYPE_TEXT )
    {
        ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet messag: %s"), pkt.payload);    
    }
    else if( pkt.type == HTTPD_WS_TYPE_BINARY )
    {
        ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Packet messag BIN:"));
        
        for( int i = 0 ; i < pkt.len ; i++ )
        {        
            ESP_LOGI( WSS_SERVER_TAG, LOG_USER("bu[%d] = %u"), i, pkt.payload[i] );
        }
    }
    else
    {
        ESP_LOGI( WSS_SERVER_TAG, LOG_USER("PAYLOAD:"));

        for( uint32_t i = 0 ; i < pkt.len ; i++ )
        {
            ESP_LOGI( WSS_SERVER_TAG, LOG_USER("pkt.payload[%u] = %u"), i, pkt.payload[i] );   
        }        
    }    
     
    return ESP_OK;
}
```

baldhead69, 2021-08-18:
```java
 @Override
        public void onOpen(WebSocket webSocket, @NotNull Response response)
        {

            Log.d("onOpen: ", "WebSocket Connection Opened !!!" );

            byte open_connection = 69;
            byte send_Server_State = 1;

            webSocket.send( ByteString.of( open_connection ) );
            webSocket.send( ByteString.of( send_Server_State ) );

            webSocket.send( ByteString.of( (byte)0x55 ) );
            webSocket.send( ByteString.of( (byte)0xAA ) );

        }
```

baldhead69, 2021-08-18:
I am trying to capture the http(ws) or https(wss) from my android phone on wireshark running on windows, but wireshark are only capturing IGMPv3 and MDNS packets.

Wireshark running on windows 11 pro. 

Some suggestion on how to capture http/s packets on wireshark.

david-cermak, 2021-08-18:
> Partially worked.

Don't want to just to conclusions, but this really points to the application that corrupts the data

> Some suggestion on how to capture http/s packets on wireshark.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html#espressif-wireshark-user-guide

If you're seeing only multicasts, perhaps you're capturing from a node that's neither the sender nor the receiver of these http packets? In that case you could:
* run the client from the capturing device
* capture in promiscuous mode/capture air packets
* run a hot spot on the capturing device and connect both the server and the client to it.

baldhead69, 2021-08-18:
> > Partially worked.
> 
> Don't want to jump to conclusions, but this really points to the application that corrupts the data

I disabled the entire application.
I only left the websocket server active.
Theoretically, if the problem was the application, the websocket server should work, but it still has the same problem.

baldhead69, 2021-08-19:
If i don't send any data for the server, when the connection open, the server shows incoming data that does not exist.

```c
I (00:00:36.589) WSS_SERVER: Opened Socket Function:
I (00:00:36.590) WSS_SERVER: hd = 0x3ffd51b0
I (00:00:36.591) WSS_SERVER: Opened socket = 55
I (00:00:36.597) WSS_SERVER: info: 1
I (00:00:36.602) WSS_SERVER: Opened WebSocket = 55
I (00:00:36.609) WSS_SERVER: open_connection: 55
I (00:00:36.615) WSS_SERVER: <LOG INIT>
I (00:00:36.619) WSS_SERVER: <LOG INIT>
I (00:00:36.624) CLIENTS LIST: ***************** Lista *******************
I (00:00:36.632) CLIENTS LIST: Connected wsClients: 1
I (00:00:36.639) CLIENTS LIST: fd_list[0] = 55
I (00:00:36.645) CLIENTS LIST: fd_list[1] = no_socket
I (00:00:36.652) CLIENTS LIST: fd_list[2] = no_socket
I (00:00:36.659) CLIENTS LIST: fd_list[3] = no_socket
I (00:00:36.665) CLIENTS LIST: fd_list[4] = no_socket
I (00:00:36.672) CLIENTS LIST: ********************************************
I (00:00:36.680) WSS_SERVER: </LOG END>
I (00:00:36.684) WSS_SERVER: </LOG END>
I (00:00:37.299) MY_APP: MALLOC_CAP_INTERNAL = 202879
I (00:00:37.299) MY_APP: MALLOC_CAP_SPIRAM = 4132439

W (00:00:41.693) httpd_txrx: httpd_sock_err: error in recv : 11
W (00:00:41.693) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
I (00:00:41.697) WSS_SERVER: httpd_ws_recv_frame ret: -1 code.
I (00:00:41.704) WSS_SERVER: Packet Socket: 55
I (00:00:41.710) WSS_SERVER: Packet type: 0
I (00:00:41.716) WSS_SERVER: Packet type:   CONTINUE
I (00:00:41.723) WSS_SERVER: Packet lenght: 0
I (00:00:41.729) WSS_SERVER: PAYLOAD:
I (00:00:42.299) MY_APP: MALLOC_CAP_INTERNAL = 203035
I (00:00:42.299) MY_APP: MALLOC_CAP_SPIRAM = 4132439
```

david-cermak, 2021-08-19:
> I disabled the entire application.Theoretically, if the problem was the application, the websocket server should work, but it still has the same problem.

*Remove* rather than *disable*, but we already did this exercise and found no issue. These kind of problems are very hard to find and decide whether the culprit is the user part or the IDF part. Therefore we usually ask users to provide a minimal, reproducible example (i believe I asked for that in the very first reply).

> After esp32 hard reset by me, the server takes about 5 to 10 seconds to boot.

This is weird, but very unlikely has any connection to the issue you're reporting. Is it the http server which takes that long to start? Checked the code you shared and didn't see any such delay.
Again, it's very hard to judge where the problem lies, so I'd suggest breaking your project down to smaller pieces to see if you reproduce the init problem separately.

> If i don't send any data for the server, when the connection open, the server shows incoming data that does not exist.

I haven't noticed any received data when the connection stayed open with no data sent when testing your project. The wireshark log should reveal if any data are coming or not. Might your clients pinging the server, perhaps?

---

As mentioned above, we still wanted to follow these three points:
1) `git bisect`
2) debug mem-corruption
3) wireshark log

Of course (repeating myself again), a minimal reproducible example would be very helpful

baldhead69, 2021-08-20:
> I haven't noticed any received data when the connection stayed open with no data sent when testing your project. The >wireshark log should reveal if any data are coming or not. Might your clients [...]

In my android app i disabled the ping to test.
Javascript running on chrome dont send ping to server, because client websocket in browser lib dont have support for this. 
As far as i know the websocket browser lib only responds with pongs(in background) to the pings sent by the server, but my server it's not sending ping to client/s.
And another thing, if the client pings the server, this ping was not supposed to show up in the uri handler.

> Of course (repeating myself again), a minimal reproducible example would be very helpful
I just need to test with the wifi initialization implemented in esp-idf examples.
After this test i will send a email to you with a minimum reproducible code.

baldhead69, 2021-08-20:
> If i don't send any data for the server, when the connection open, the server shows incoming data that does not exist.
>
> ```c
> I (00:00:36.589) WSS_SERVER: Opened Socket Function:
> I (00:00:36.590) [...]

Today i tested with ESP32-PICO-DevKitM-2 board and the problem persists.

baldhead69, 2021-08-20:
Hi @david-cermak 

I sent "a minimal reproducible example code" to your email.   

Even me using the wifi example code from esp-idf the problem persist.

david-cermak, 2021-08-20:
@baldhead69 Thanks for sharing the project! Finally, I was able to reproduce the issue!

The problem is that the ws handler is also called upon http-get method, so that we misread the second byte of the WS protocol manually in the handler and read the first one instead. That's why the packet types and lengths were wrong.
This functionality was introduced in 
* this commit https://github.com/espressif/esp-idf/commit/a1d5cfc26023c64e973d5b891677a53302e993a0
* requested here https://github.com/espressif/esp-idf/issues/6597

You can fix it in your code adding this condition:

https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L70-L73

in your handler.

Sorry I haven't noticed you're also missing this condition in the first version you supplied, but that was largely incomplete so I filled the missing parts myself.

---

Also, please re-read my *second* post in this issue back from May

https://github.com/espressif/esp-idf/issues/7027#issuecomment-844743481

where I guessed that this had been the very problem!

baldhead69, 2021-08-20:
> @baldhead69 Thanks for sharing the project! Finally, I was able to reproduce the issue!
> 
> The problem is that the ws handler is also called upon http-get method, so that we misread the second byte of the WS protocol manually in the handler and read the first one instead. That's why the packet types and lengths were wrong.
> This functionality was introduced in
> 
> * this commit [a1d5cfc](https://github.com/espressif/esp-idf/commit/a1d5cfc26023c64e973d5b891677a53302e993a0)
> * requested here [httpd_server: websocket callback for successfull handshake / open (IDFGH-4796) #6597](https://github.com/espressif/esp-idf/issues/6597)

> 
> You can fix it in your code adding this condition:
> 
> https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L70-L73
> 
> in your handler.
> 
> Sorry I haven't noticed you're also missing this condition in the first version you supplied, but that was largely incomplete so I filled the missing parts myself.
> 
> Also, please re-read my _second_ post in this issue back from May
> 
> [#7027 (comment)](https://github.com/espressif/esp-idf/issues/7027#issuecomment-844743481)
> 
> where I guessed that this had been the very problem!

I tested the "solution".

With local variables the "solution" works.
```c 
static esp_err_t get_clients_message_handler(httpd_req_t *req)
{
    httpd_ws_frame_t pkt;
    uint8_t bu[128] = { 0 };

    memset(&pkt, 0, sizeof(httpd_ws_frame_t));
    pkt.payload = bu;  
}
```

With static global variables the "solution" dont works.
```c
static uint8_t buf[128] = { 0 };
static httpd_ws_frame_t ws_pkt;

static uint8_t state_buf[128] = { 0 };
static httpd_ws_frame_t ws_server_state_pkt;

tatic void init_frame_variables()
{    
    // Receiver and transmitter frame
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_pkt.payload = buf;

    // Transmitter only frame
    memset(&ws_server_state_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_server_state_pkt.payload = state_buf;
    ws_server_state_pkt.final = true;   
    ws_server_state_pkt.type = HTTPD_WS_TYPE_BINARY;
}
```

I think that when the server called the uri handler, the handshake was already finished, i am corret ?

Why the first call to uri handler(when websocket client open connection) are of type "req->method == HTTP_GET" ?

Why the other calls to uri handler are of type "req->method == HTTP_DELETE" ?

david-cermak, 2021-08-21:
Hi @baldhead69 

Here's the complete diff of your project to make it working
```patch
diff -u components/communication/my_wss_server/src/webSocket_server.c
@@ -122,6 +122,11 @@
     return ESP_OK;
 */
 
+    if (req->method == HTTP_GET) {
+        return ESP_OK;
+    }
+    ws_pkt.payload = buf;
+    ws_pkt.len = 0;
 
     esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));  // sizeof(buf) = 128 bytes
```

As for the feature requests, they are evaluated by component owner's team based on customer requests, if we have more customers, more likely the feature will be implemented (You know my opinion (and reasoning) on your suggestion of calling `open_fd()` after the handshake).

I have to admit, though, that the changes to cause your problems are *breaking* and ~~should~~ must  **not** have been made between minor releases, no matter how experimental the ws-server feature is. Let me apologize for the inconvenience, we always try to pay particular attention to anything that may break existing application, but this changeset slipped through the net, sorry for that.

baldhead69, 2021-08-22:
Hi @david-cermak ,

So i assume this approach is correct for adding a websocket client to a list, right ?
"add_ws_client_to_list(sockfd)"
```c
if( req->method == HTTP_GET )
{
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Handshake done, the new connection was opened"));        
    int sockfd = httpd_req_to_sockfd(req);
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("Handshake Socket: %d"), sockfd); 
    **add_ws_client_to_list(sockfd);**
    printf("\n");    
    return ESP_OK;
}
```

Here the closed socket it was supposed to be websocket("HTTPD_WS_CLIENT_WEBSOCKET"), but it is returning 1 "HTTPD_WS_CLIENT_HTTP".
```c
// typedef void (*httpd_close_func_t)(httpd_handle_t hd, int sockfd);

static void closed_socket_handler(httpd_handle_t hd, int sockfd)
{    
    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Closed socket Function:") );
    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("hd = %p"), hd );
    ESP_LOGI( WSS_SERVER_TAG, LOG_USER("Closed socket = %d"), sockfd );
        
    // if( sockfd == websocket_Socket )
    httpd_ws_client_info_t info = httpd_ws_get_fd_info(hd, sockfd);
 
    ESP_LOGI(WSS_SERVER_TAG, LOG_USER("info: %d"), info );

    if( info != HTTPD_WS_CLIENT_WEBSOCKET )
        return;

    remove_ws_client_from_list(sockfd);
}
```

david-cermak, 2021-08-23:
@baldhead69 Unfortunately none of these is possible if you wanted to keep the list of ws clients only. I would suggest listing **all**  clients and sorting the ws clients out only if any action (pinging, sending async messages) needed.
* the URI handler is called before the handshake for the ws clients, but also for standard http clients, so the condition above is true also if a common http request takes place with the same uri.
* `httpd_ws_get_fd_info()` returns the ws type only if 
    - ws handshake is completed
    - and at the same time the connection is not being closed

https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_ws.c#L479

The latter is *mostly* true if your clients close the connection in a clean way (sending `CLOSE` frame) which most clients do.

Please understand that the ws server support in IDF is very simple. It's not a full featured websocket server, just a httpd that understands ws frames and some of the heavy-lifting is left for the user code.

baldhead69, 2021-08-24:
Hi @david-cermak ,

I follow the example code"https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c".

Partially code of function "static esp_err_t echo_handler(httpd_req_t *req)".  

The ws_pkt.len are return buggy values.

```c
I (00:03:10.493) esp_https_server: performing session handshake
I (00:03:12.016) TAG: Handshake done, the new connection was opened
I (00:03:12.044) TAG: frame len is 1
I (00:03:15.503) TAG: frame len is 2
I (00:03:20.116) TAG: frame len is 5
I (00:03:20.118) TAG: frame len is 4
W (00:03:23.494) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
E (00:03:23.494) TAG: httpd_ws_recv_frame failed to get frame len with ESP_ERR_INVALID_STATE
I (00:03:23.500) WSS_SERVER: Closed socket handler:
```

I am using binary frame "HTTPD_WS_TYPE_BINARY".
```c
static esp_err_t get_clients_message_handler(httpd_req_t *req)
{   
   
    if (req->method == HTTP_GET)
    {
        ESP_LOGI("TAG", "Handshake done, the new connection was opened");
        return ESP_OK;
    }

    httpd_ws_frame_t ws_pkt;    
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    
    /* Set max_len = 0 to get the frame len */
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
    if (ret != ESP_OK) {
        ESP_LOGE("TAG", "httpd_ws_recv_frame failed to get frame len with %s", esp_err_to_name(ret));
        return ret;
    }

    ESP_LOGI("TAG", "frame len is %d", ws_pkt.len);

    return ESP_OK;    
```

baldhead69, 2021-08-24:
@david-cermak ,

As a suggestion could have a function that returns the payload size directly:
```c
static esp_err_t websocket_uri_handler( httpd_req_t *req )
{   
    if (req->method == HTTP_GET)
    {
        ESP_LOGI("TAG", "Handshake done, the new connection was opened");
        return ESP_OK;
    }

    size_t payload_size = httpd_ws_get_payload_size(req); 
}
```

david-cermak, 2021-08-24:
@baldhead69 The original API works good, but you have to read the payload after checking the length. The same way it's outlined in the [example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c). Otherwise next payload check reads the actual payload data from the network and reads incorrectly.

```cpp
  // read length
  httpd_ws_recv_frame(req, &ws_pkt, 0); 
  // log length
  ESP_LOGI( WSS_SERVER_TAG, LOG_USER("length: %d"), ws_pkt.len );
  // read payload
  httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);  // provided sizeof(buf) > ws_pkt.len
```

baldhead69, 2021-08-24:
@david-cermak 

Except the first payload packet that has payload lenght 1 the rest of the payloads lenght is 4.
In addition, an error occurs after some receptions.

david-cermak, 2021-08-25:
Hi @baldhead69 

I wonder if you're really asking support questions or just randomly shuffle the lines in your handler to post a log proving that *it doesn't work*.
Please understand that GitHub issues are mainly used to track bugs in ESP-IDF and thus we rely on submitters to create clean and meaningful reports and keep the conversation to the point.
This helps the IDF support team and our community to handle issues smoothly and pays back to all users when searching for active or older issues, bugs and corresponding fixing commits.
What do you think about this report, what would anyone think when reading through this issue? Especially when experiencing trouble with mDNS or the WS server, or both?

About your [query on the binary frames](https://github.com/espressif/esp-idf/issues/7027#issuecomment-904345634) and *"The ws_pkt.len returning buggy values"* :
Here's the complete diff to make [your snippet](https://github.com/espressif/esp-idf/issues/7027#issuecomment-904345634) working:

```patch
diff -u components/communication/my_wss_server/src/webSocket_server.c
@@ -73,8 +73,6 @@
         ESP_LOGE("TAG", "httpd_ws_recv_frame failed to get frame len with %s", esp_err_to_name(ret));
         return ret;
     }
-    ws_pkt.payload = buf;
-    httpd_ws_recv_frame(req, &ws_pkt, sizeof(buf));
     ESP_LOGI("TAG", "frame len is %d", ws_pkt.len);
 
     return ESP_OK;
```

As mentioned above, the websocket support in httpd is very simple and we have to read the data continuously in the user handler (need to read the payload and set the `ws_pkt.payload`).

To my understanding, there are no issues with websocket server nor in mnds.
There are some incompatibilities between the `v4.3` and current master, though.
I'd suggest updating the documentation on how to correctly write a handler. Since there's no real documentation of this feature, besides this [Websocket server](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#websocket-server) section and reference to the project and code, I will add a paragraph to the [ws echo example](https://github.com/espressif/esp-idf/tree/f65c824/examples/protocols/http_server/ws_echo_server#how-to-use-example)
highlighting the need for checking the request method and setting a valid length and payload in the `httpd_ws_frame_t` before reading frames.

baldhead69, 2021-08-25:
@david-cermak 

I commented about the payload size because i decided to do some experiments (learning) with dynamic allocation of the httpd_ws_frame_t structure and with the payload buf.

I call a function that allocate the httpd_ws_frame_t and payload buffer and return a frame_t pointer.

The free allocated memory not show in example.

Example: More or less like this.
```c
inside uri handler
{
httpd_ws_frame_t* ws_frame_mem = NULL;

ret = allocate_ws_frame_memory(ws_pkt.len, &ws_frame_mem); if( ret != ESP_OK ) { ESP_LOGE(WSS_SERVER_TAG, "allocate_ws_frame_memory ERROR: %s\n", esp_err_to_name(ret)); return ret; }
}

static esp_err_t allocate_ws_frame_memory(size_t buf_size, httpd_ws_frame_t** allocated_ws_frame_memory) { if( buf_size == 0 ) return ESP_FAIL; *allocated_ws_frame_memory = calloc(1, sizeof(httpd_ws_frame_t)); if(*allocated_ws_frame_memory == NULL) return ESP_ERR_NO_MEM; uint8_t* buf = calloc(buf_size, sizeof(uint8_t)); if (buf == NULL) { free(*allocated_ws_frame_memory); return ESP_ERR_NO_MEM; } (*allocated_ws_frame_memory)->fragmented = false; (*allocated_ws_frame_memory)->final = true; (*allocated_ws_frame_memory)->payload = buf; (*allocated_ws_frame_memory)->len = buf_size; (*allocated_ws_frame_memory)->type = HTTPD_WS_TYPE_BINARY; ESP_LOGI(WSS_SERVER_TAG, LOG_USER("*allocated_ws_frame_memory: 0x%p"), *allocated_ws_frame_memory); ESP_LOGI(WSS_SERVER_TAG, LOG_USER("buf: 0x%p"), buf); ESP_LOGI(WSS_SERVER_TAG, LOG_USER("(*allocated_ws_frame_memory)->payload: 0x%p\n"), (*allocated_ws_frame_memory)->payload); return ESP_OK; }
```

baldhead69, 2021-08-28:
@david-cermak ,

I took two tests:
* Send ping with payload and send ping without payload.

First test: Send ping with payload.
```c
I (00:00:28.062) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:28.070) WSS_SERVER: Handshake Socket: 55

I (00:00:28.162) WSS_SERVER: URI HANDLER:
I (00:00:28.166) WSS_SERVER: req->method = 0

W (00:00:28.172) httpd_txrx: httpd_sock_err: error in recv : 104
W (00:00:28.178) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
I (00:00:28.186) WSS_SERVER: httpd_ws_recv_frame failed with: ESP_FAIL
```
[image]
[image]

Second test: Send ping without payload.
```c
I (00:00:57.224) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:57.233) WSS_SERVER: Handshake Socket: 55

I (00:00:57.325) WSS_SERVER: URI HANDLER:
I (00:00:57.328) WSS_SERVER: req->method = 0

W (00:00:57.382) httpd_ws: httpd_ws_recv_frame: Failed to receive the second byte
I (00:00:57.382) WSS_SERVER: httpd_ws_recv_frame failed with: ESP_FAIL
```

[image]
[image]

baldhead69, 2021-08-28:
@david-cermak ,

I already took the opportunity to test the websocket close control frame.

The websocket close control frame dont call uri handler:
```c
I (00:00:23.257) WSS_SERVER: URI HANDLER:
I (00:00:23.258) WSS_SERVER: req->method = 1

I (00:00:23.258) WSS_SERVER: Handshake done, the new connection was opened
I (00:00:23.267) WSS_SERVER: Handshake Socket: 55

I (00:00:23.359) WSS_SERVER: Closed socket handler:
I (00:00:23.363) WSS_SERVER: hd = 0x3ffd5348
I (00:00:23.369) WSS_SERVER: Closed socket = 55
```

[image]

david-cermak, 2021-08-30:
@baldhead69 Please be kind to us and create a separate issue for the control frames. We will close this one, once we update the docs.

There might be some bugs in processing of ping frames with no payload, and passing the close frame to the handler. It is unrelated to this issue.

baldhead69, 2021-08-30:
@david-cermak 

> @baldhead69 Please be kind to us and create a separate issue for the control frames. We will close this one, once we update the docs.

OK, i will open a new issue.
 
> There might be some bugs in processing of ping frames with no payload, and passing the close frame to the handler. It is unrelated to this issue.

There are some bugs in processing of ping frames with payload too.

closed 2021-09-10 by espressif-bot