espressif/esp-idf #7028: Unable to send files from Android phone to ESP using SPP

shark8594, 2021-05-17:
I am trying to send an mp3 file from my Samsung phone (Could not get the HCI snoop logs on the phone :/) to ESP-Lyra using SPP, and it seems to fail after successful pairing. However using an SPP BT terminal app on the phone I am able to send the serialized data. Wanted to know if any changes are needed to the stock bt_spp_acceptor example to receive files from the Android devices without custom apps that open RFCOMM sockets and send data through it. Log below:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6892
load:0x40078000,len:13072
load:0x40080400,len:3896
entry 0x40080688
I (29) boot: ESP-IDF v4.1-dirty 2nd stage bootloader
I (29) boot: compile time 22:00:21
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x19b6c (105324) map
I (152) esp_image: segment 1: paddr=0x00029b94 vaddr=0x3ffbdb60 size=0x03400 ( 13312) load
I (158) esp_image: segment 2: paddr=0x0002cf9c vaddr=0x40080000 size=0x00404 (  1028) load
I (160) esp_image: segment 3: paddr=0x0002d3a8 vaddr=0x40080404 size=0x02c70 ( 11376) load
I (173) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x65bac (416684) map
I (336) esp_image: segment 5: paddr=0x00095bd4 vaddr=0x40083074 size=0x112a0 ( 70304) load
I (377) boot: Loaded app from partition at offset 0x10000
I (377) boot: Disabling RNG early entropy source...
I (378) cpu_start: Pro cpu up.
I (381) cpu_start: Application information:
I (386) cpu_start: Project name:     bt_spp_acceptor_demo
I (392) cpu_start: App version:      v4.1-dirty
I (397) cpu_start: Compile time:     May 16 2021 20:57:45
I (404) cpu_start: ELF file SHA256:  33718e1dfe5e42d1...
I (410) cpu_start: ESP-IDF:          v4.1-dirty
I (415) cpu_start: Starting app cpu, entry point is 0x400812fc
I (0) cpu_start: App cpu up.
I (425) heap_init: Initializing. RAM available for dynamic allocation:
I (432) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (438) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (444) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (450) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (456) heap_init: At 3FFCAD38 len 000152C8 (84 KiB): DRAM
I (463) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (469) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (475) heap_init: At 40094314 len 0000BCEC (47 KiB): IRAM
I (482) cpu_start: Pro cpu start user code
I (500) spi_flash: detected chip: gd
I (500) spi_flash: flash io: dio
W (501) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (511) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (577) BTDM_INIT: BT controller compile version [219866f]
I (577) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (677) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
D (1127) BT_BTC: btc_transfer_context msg 0 0 0 0x0

D (1127) BT_BTC: btc_thread_handler msg 0 0 0 0x0

D (1137) BT_BTC: btc_main_call_handler act 0

D (1137) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
[... 608 lines cut ...]
D (28707) BT_BTC: btc_thread_handler msg 1 8 5 0x3ffbb42c

D (28717) BT_BTC: btc_dm_upstreams_cback  ev: 5

D (28717) BT_BTC: btc_dm_sec_cback : unhandled event (5)

D (31797) BT_HCI: Receive packet event_code=0x5

D (31797) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (31797) BT_BTC: btc_transfer_context msg 1 8 6 0x3ffcfbc0

I (31807) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
D (31807) BT_BTC: btc_transfer_context msg 1 8 8 0x3ffcfbc0

D (31817) BT_BTC: btc_thread_handler msg 1 8 6 0x3ffbabe0

D (31827) BT_BTC: btc_dm_upstreams_cback  ev: 6

D (31827) BT_BTC: btc_dm_sec_cback : unhandled event (6)

D (31837) BT_BTC: btc_thread_handler msg 1 8 8 0x3ffbacfc

D (31837) BT_BTC: btc_dm_upstreams_cback  ev: 8

Executing action: monitor

Wth-Esp, 2021-05-17:
Hi,

You can use VFS mode BT SPP acceptor, which is much like the socket.
Thanks

shark8594, 2021-05-17:
Thanks for the quick response. I tried the VFS acceptor demo and it seems to have the same issue, but works with the SPP terminal app. Attached is the log, it seems like it is not reaching the ESP_SPP_OPEN_EVT event and an l2cap_disconnect() is called. 
[ESP_BT_log.txt](https://github.com/espressif/esp-idf/files/6494628/ESP_BT_log.txt)

xiongweichao, 2021-05-24:
Hi, @shark8594 

Could you please provide **sdkonfig**?

Thanks

xiongweichao, 2021-05-25:
Hi, @shark8594 

Mobile phones do not use SPP to transfer files. If you want to play music, you can use A2DP.

Thanks

Alvin1Zhang, 2021-05-31:
@shark8594 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

shark8594, 2021-05-31:
SPP does not seem to work well for file transfer, the only way is to write a custom app that serializes the data and sends it over SPP.

closed 2021-05-31 by shark8594