espressif/esp-idf #7029: tjpgd library does not exist in ESP32-S2 ROM.

nopnop2002, 2021-05-17:
**Is your feature request related to a problem? Please describe.**

tjpgd library does not exist in ESP32-S2 ROM.

**Describe the solution you'd like**

I want you to include the tjpgd library in the ESP32-S2 ROM.

**Describe alternatives you've considered**

The tjpeg library uses a lot of memory, but small size images are also available in esp32s2.

**Additional context**

None.

igrr, 2021-05-31:
Hi @nopnop2002, what issues are you facing when linking tjpgd library into your application? If you observe some performance issues when tjpgd is placed into Flash, you can selectively move some of the functions to IRAM to improve performance, using [linker script fragments](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/linker-script-generation.html).
If that still doesn't work, please explain the problem you are facing.

Here is an example of tjpgd being included into the application code for the ESP32-S2, in the esp32-camera project:

https://github.com/espressif/esp32-camera/blob/901c05b431ea615b11fbc65b39a2ed286abe8df8/CMakeLists.txt#L39-L45
https://github.com/espressif/esp32-camera/blob/901c05b431ea615b11fbc65b39a2ed286abe8df8/target/esp32s2/private_include/tjpgd.h
https://github.com/espressif/esp32-camera/blob/901c05b431ea615b11fbc65b39a2ed286abe8df8/target/esp32s2/tjpgd.c

nopnop2002, 2021-05-31:
@igrr 

Thank you for response.

The reason why this library is not available by default in ESP32S2 (tjpgd must be included into the application code) is strange.

negativekelvin, 2021-05-31:
>I want you to include the tjpgd library in the ESP32-S2 ROM.

It's not possible to change the ROM.  S2 has a small ROM capacity.  C3 and S3 have those functions in ROM.

nopnop2002, 2021-05-31:
>It's not possible to change the ROM. S2 has a small ROM capacity. C3 and S3 have those functions in ROM.

OK.
I understood.

nopnop2002, 2022-11-14:
In ESP-IDF Ver4.4, esp32s2 does not include tjpeg component.

This is due to the small ROM size of esp32s2.

In ESP-IDF Ver5.0, the tjpeg component has been moved to the IDF Component Registry.

Still can't we use the tjpeg component with esp32s2?

nopnop2002, 2022-11-14:
I found that tjpeg can be used with ESP-IDF Ver5 + esp32s2

closed 2022-11-14 by nopnop2002