espressif/esp-idf #7030: assert rwble.c 178, param 00000000 00000000

u-tec-com, 2021-05-17:
i am using ESP-C3 with ble mesh and mesh gatts proxy.

got error
assert rwble.c 178, param 00000000 00000000

how to fix it?

InfiniteYuan, 2021-05-18:
@u-tec-com Please provide more detailed information based on the issue template.

Alvin1Zhang, 2021-05-31:
@u-tec-com Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

liangzc1, 2021-07-09:
我在C3上面碰到了同样的问题，如果我OTA升级时不关掉蓝牙则必现，调用esp_bt_controller_disable()关闭蓝牙，该问题则变成偶现的commit是
commit e9cf9e2978d0187b0f83a9978391217e39946517 (HEAD -> release/v4.3, tag: v4.3-beta3, origin/release/v4.3)

log如下：

[19:45:16.995]收←◆[0;31mE (10140) app_engine.c(41): MQTT_PUB topic: /led/91808545b024efce/status/up, msg: {"did": "91808545b024efce", "uptime":"10140", "rssi":"-38"}[0m

[19:45:22.598]收←◆[0;31mE (15750) tools.c(50): ---MEMINFO [15750] (HTTP) : [0m
[0;31mE (15750) tools.c(58): heap free: 39.25Kb used: 54.25Kb change: -54.25Kb largest: 16.00Kb min-heap: 37.05Kb  40196 33792[0m

[19:45:23.910]收←◆[0;31mE (17060) main.c(94): Erase partition ota_0  0x10000 0x1b0000[0m

[19:45:27.276]收←◆[0;31mE (20420) app_engine.c(41): MQTT_PUB topic: /led/91808545b024efce/status/up, msg: {"did": "91808545b024efce", "uptime":"20420", "rssi":"-39"}[0m

[19:45:32.512]收←◆[0;31mE (25660) main.c(94): Partition erased, Downloading firmware...[0m

[19:45:32.905]收←◆[0;31mE (26050) tools.c(50): ---MEMINFO [26050] (HTTP) : [0m

[19:45:33.206]收←◆[0;31mE (26050) tools.c(58): heap free: 12.94Kb used: 80.56Kb change: -26.31Kb largest: 2.00Kb min-heap: 8.67Kb  13828 10824[0m

[19:45:33.806]收←◆assert rwble.c 178, param 00000000 00000000

打印后芯片自动重启

InfiniteYuan, 2021-07-12:
Please update to latest IDF release/v4.3. We have already fix it on commit id [4657069afdc036bc930073edd6ea17efdae21c27](https://github.com/espressif/esp-idf/commit/4657069afdc036bc930073edd6ea17efdae21c27).

Alvin1Zhang, 2021-08-16:
@liangzc1 @u-tec-com Would you please help share if any updates for the issue? Thanks.

Alvin1Zhang, 2021-12-07:
Thanks for reporting, please help try the fix as mentioned in https://github.com/espressif/esp-idf/issues/7030#issuecomment-877929386, feel free to reopen if the issue still happens.

closed 2021-12-07 by Alvin1Zhang

lipanpan1030, 2023-04-06:
ESP V4.3.1 重置设备后上电起来 卡住了.

相关日志如下. 完整日志见附件
15:55:45.328  I (00:00:17.064) MZ_BLE_PERIPHERAL: start_mesh_ble_adv
15:55:45.328  DIAG0 1127c3a0
15:55:45.333  DIAG1 11270093
15:55:45.333  BB DIAG0: 020e4027
15:55:45.339                              BB DIAG1: b9004622
15:55:45.344                              BB DIAG2: 00000000
15:55:45.344                              BB DIAG3: 8e89bed6
15:55:45.350                              BB DIAG4: 32000000
15:55:45.356                              BB DIAG5: 00000000
15:55:45.356  EM BASE ERROR: b00
15:55:45.359  assert rwble.c 261, param 00200200 00000000
15:55:50.236  E (7255) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
15:55:50.242  E (7255) task_wdt:  - IDLE (CPU 0)
15:55:50.248  E (7255) task_wdt: Tasks currently running:

[rwble重启后发生崩溃问题反馈乐鑫.txt](https://github.com/espressif/esp-idf/files/11168499/rwble.txt)

xiewenxiang, 2023-04-10:
@lipanpan1030 
可以提供复现代码吗？ 如果无法提供代码，请描述一下复现的步骤和流程