espressif/esp-idf #7035: Wifi - Connected but not getting IP Address

nikhilavrio, 2021-05-17:
I (3974) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 2
W (3974) phy_init: saving new calibration data because of checksum failure, mode(0)
I (3992) wifi:mode : sta (24:6f:28:60:10:12)
System STA EVENT START
Connecting to wifi...
I (3994) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
I (6040) wifi:new:<11,2>, old:<1,0>, ap:<255,255>, sta:<11,2>, prof:1
I (6041) wifi:state: init -> auth (b0)
I (6045) wifi:state: auth -> assoc (0)
I (6053) wifi:state: assoc -> run (10)
I (6161) wifi:connected with AvrioTest, aid = 2, channel 11, 40D, bssid = 84:d8:1b:b9:a7:30
I (6161) wifi:security: WPA2-PSK, phy: bgn, rssi: -48
I (6165) wifi:pm start, type: 1

I (6254) wifi:AP's beacon interval = 102400 us, DTIM period = 1

espxiehang, 2021-05-18:
Hi @nikhilavrio , Can you use the master version and open the DHCP debug log for testing.

Component config  ---> LWIP  --->  Debug  --->  [ * ] Enable DHCP state tracking
                                                                            [ * ] Enable DHCP debug messages

Alvin1Zhang, 2021-05-31:
@nikhilavrio Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

ralphgod3, 2021-06-29:
I am having the same problem as @nikhilavrio.
the logs are with dhcp state tracking and dhcp debug messages enabled esp-idf version is latest master as of writing (8e3e65a)

I (478) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (564) wifi station: ESP_WIFI_MODE_STA
I (574) wifi:wifi driver task: 3ffbffb0, prio:23, stack:6656, core=0
I (574) system_api: Base MAC address is not set
I (574) system_api: read default base MAC address from EFUSE
I (594) wifi:wifi firmware version: ff5f4ea
I (594) wifi:wifi certification version: v7.0
I (594) wifi:config NVS flash: enabled
I (594) wifi:config nano formating: disabled
I (604) wifi:Init data frame dynamic rx buffer num: 32
I (604) wifi:Init management frame dynamic rx buffer num: 32
I (614) wifi:Init management short buffer num: 32
I (614) wifi:Init dynamic tx buffer num: 32
I (624) wifi:Init static rx buffer size: 1600
I (624) wifi:Init static rx buffer num: 10
I (624) wifi:Init dynamic rx buffer num: 32
I (634) wifi_init: rx ba win: 6
I (634) wifi_init: tcpip mbox: 32
I (644) wifi_init: udp mbox: 6
I (644) wifi_init: tcp mbox: 6
I (644) wifi_init: tcp tx win: 5744
I (654) wifi_init: tcp rx win: 5744
I (654) wifi_init: tcp mss: 1440
I (664) wifi_init: WiFi IRAM OP enabled
I (664) wifi_init: WiFi RX IRAM OP enabled
I (674) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (774) wifi:mode : sta (fc:f5:c4:30:b4:4c)
I (774) wifi:enable tsf
I (774) wifi station: wifi_init_sta finished.
I (784) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (784) wifi:state: init -> auth (b0)
I (794) wifi:state: auth -> assoc (0)
I (794) wifi:state: assoc -> run (10)
I (804) wifi:connected with RALPH-PC 4796, aid = 9, channel 1, BW20, bssid = 3e:f0:11:da:68:e4
I (804) wifi:security: WPA2-PSK, phy: bgn, rssi: -53
I (814) wifi:pm start, type: 1

dhcp_start(netif=0x3ffbda30) st1
dhcp_start(): restarting DHCP configuration
dhcp_start(): starting DHCP configuration
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(94b461d8)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 500 msecs
I (904) wifi:AP's beacon interval = 102400 us, DTIM period = 3
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(94b461d8)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 1000 msecs
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(94b461d8)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 2000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_fine_tmr(): request timeout

the dhcp request timeout just keeps repeating.

espxiehang, 2021-07-21:
Hi @ralphgod3  ,AP does not respond to DHCP discover sent by ESP32. You can check the DHCP behavior of AP.

ahawh87, 2021-07-24:
I have the same issue, all other devices (Mobile phone & laptop) connect to the home network without any issue, Just the ESP sometimes do this issue.

nikhilavrio, 2021-07-25:
Check your power supply once.

I fixed my powers supply and it got working. But not sure if it comes up
again.

TianaESP, 2021-11-16:
@nikhilavrio  may I know if it works well now? If so, we will close the issue first. Thanks.

closed 2021-12-07 by Alvin1Zhang