espressif/esp-idf #7038: No messages shown by any node

L30n4rd0-code, 2021-05-17:
I have been testing this example for a couple of weeks, my nodes connect correctly and create an organized network with the root node and four child nodes but there are no messages shown in the serial monitor, whether of a root node or a child node. If I correctly understand, the root node sends messages to all the child nodes, so is in the child node's serial monitor that I should see such messages. How could I manage to have those messages shown in the serial monitor? Any ideas about it?

Alvin1Zhang, 2021-05-18:
@L30n4rd0-code Thanks for reporting, which example you refer to? Thanks.

L30n4rd0-code, 2021-05-18:
I'm refering to the _Internal_communication_ example.

shenjun7, 2021-07-07:
The bug has been fixed in gitlab and is waiting to be synchronized to github.

Alvin1Zhang, 2021-11-24:
Thanks for reporting, and sorry for slow turnaround, fix is available at https://github.com/espressif/esp-idf/commit/d93849baf86b6de5f08a03ae6aa7006dac58a93b, feel free to reopen.

closed 2021-11-24 by Alvin1Zhang