espressif/esp-idf #7039: gsm0710 support

aep, 2021-05-17:
Hi,

edf-idf doesn't appear to have a way to use tcp/ip offloading with connected modems, so we're restricted to using PPP.
To query GPS or signal info during a PPP session, you need to use gsm0710 mux mode, which is currently not implemented.

it's a fairly simple protocol that wraps all commands in a MUX header so you can keep using text based AT commands in parallel.
I'll have to implement it anyway, but if there's any interest in publishing it upstream, i'll try to keep to the coding standard

aep, 2021-05-17:
this is more invasive than expected because the current PPP driver uses uart_enable_pattern_det_baud_intr
i'd have to rewrite most of the uart handler.
actually i wonder if this should be part of uart instead, like it is on linux.

david-cermak, 2021-05-20:
This is a duplicate of https://github.com/espressif/esp-idf/issues/4688

There's been some progress on the CMUX implementation in the [`esp-modem`](https://components.espressif.com/component/espressif/esp_modem) component, which is a managed component, external one, not part of IDF.
You can evaluate this feature, installing the esp-modem using [IDF component manager](https://components.espressif.com):
```
pip install --upgrade  idf-component-manager
```

closed 2021-05-20 by aep

jonathandreyer, 2021-05-22:
Hi @aep, nice to hear that you are also interested with the MUX feature. As explained by David, there are already two ideas of implementation in issue #4688.

@david-cermak Is it possible to add this progress in the opened issue? I have taken a look to your link, but the document is bit small. Is it possible to access to source code?

aep, 2021-05-23:
yeah, could not find the source code for esp-modem and the "Install" button on the website does nothing.
idf-component-manager also doesnt itself contain any urls as far as i can see, so that's a dead end too.

i had to make a new thing instead. https://github.com/aep/bg9x

AxelLin, 2021-05-26:
> This is a duplicate of #4688
>
> There's been some progress on the CMUX implementation in the [`esp-modem`](https://components.espressif.com/component/espressif/esp_modem) component, which is a managed [...]

Hi @david-cermak 
This is new to me.
After installing idf_component_manager, how to get the esp-modem component source code?

Thanks,
Axel

david-cermak, 2021-05-26:
We don't have a good solution (yet) to deploy examples, so currently the component is available only if set as a dependency to some idf project.

To deploy this component, please add this yaml, name it `idf_component.yml`
```yaml
targets:
  - esp32
dependencies:
  espressif/esp_modem:
    version: "0.1.7"
```
to your main component in any idf example (e.g. here: [examples/get-started/hello_world/main](https://github.com/espressif/esp-idf/tree/c13afea635adec735435961270d0894ff46eef85/examples/get-started/hello_world/main) ) and type 
```
idf.py build
```

(this will download the esp-modem component including examples, tests, docs to this folder `managed_components/espressif__esp_modem` if component manager was installed)

jonathandreyer, 2021-06-06:
@david-cermak Do you plan to release the source code of this component in a near future? As I use PlatformIO, I can't download it as you have mentioned. Probably, I can install IDF in a virtual environment, but I have come concerns about this option.

jonathandreyer, 2021-06-27:
Hi everyone, sometime I check the website of the IDF component registry. I have discovered that now, it is possible to download all versions of the component. To simplify the follow-up of the iteration of this component, I have created another repository with commits for each release: https://github.com/jonathandreyer/esp-modem

mapedraza, 2021-07-28:
> Hi everyone, sometime I check the website of the IDF component registry. I have discovered that now, it is possible to download all versions of the component. To simplify the follow-up of the [...]

Checking your folder, there is only the cpp example, there is any example using just C?

jonathandreyer, 2021-08-06:
@david-cermak Sorry to ping you from this issue, but the URL indicate in your above comment ([here](https://github.com/espressif/esp-idf/issues/7039#issuecomment-844951625)) seems not working. It is normal?
Best regards,
Jonathan Dreyer

david-cermak, 2021-08-16:
@jonathandreyer Sorry for the late reply
* I think it's a good idea to publish the source somewhere, but we haven't yet decided if this would be a separate repository of a collection of external components, so for now the source it available only via component manager
* The component registry was migrated at that time, should be available now.

@mapedraza 
The first example `pppos_client` is C only. The component is internally implemented in C++17, but a plain C only API is provided https://components.espressif.com/component/espressif/esp_modem

 > `examples/pppos_client` – simple client implemented in C

jonathandreyer, 2021-09-07:
Hi @david-cermak Sorry to ping you again, but I did not find another elegant way to ask question regarding the esp_modem component. Do I need to open another issue?
The question is related to the PPPoS example of this repository versus examples from esp_modem component. The example located from this repository will be maintained in parallel of the esp_modem component or is it better to switch to the esp_modem component?
Best regards,
Jonathan Dreyer

david-cermak, 2021-09-08:
Hi Jonathan, 

I'd suggest migrating to the new `esp_modem` component. The example itself will be maintained in `v5.0` (and later), but the internal implementation of the [`modem`](https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client/components/modem) in examples will be replaced by the official managed component.

regards,
David

jonathandreyer, 2021-09-08:
@david-cermak Thanks for your reply. Ok, I understand, but when we have some questions regarding components where we can ask those questions?
Best regards,
Jonathan Dreyer

david-cermak, 2021-10-06:
FYI: `esp_modem` component sources have been made public and moved to [`esp-protocols`](https://github.com/espressif/esp-protocols) repository.