espressif/esp-idf #7042: ninja failed after build exit code 1

angelo-dising, 2021-05-18:
after executing the idf.py i receive the next text

Executing action: all (aliases: build)
Running ninja in directory /home/angelo/Desktop/esp/esp-idf/examples/get-started/hello_world/build
Executing "ninja all"...
ninja: error: '/home/angelo/Desktop/esp/esp-idf/components/esp_wifi/lib/esp32s2/libcoexist.a', needed by 'hello-world.elf', missing and no known rule to make it
ninja failed with exit code 1

it says that there is a missing file but there is no such one even on the original ifd repo
im on ubuntu 20.04 , is a very recent installation

Alvin1Zhang, 2021-05-18:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

igrr, 2021-05-18:
@angelo-dising can you try running `git submodule update --init`? The missing file is inside the submodule, perhaps it is not initialized or outdated.

Alvin1Zhang, 2021-05-31:
@angelo-dising Thanks for reporting. Would you please help share if any updates for this issue? Thanks.

huotanzai, 2021-06-01:
我也遇到了同样的问题。

Dazza0, 2021-09-29:
@angelo-dising @huotanzai This issue is most likely caused by your git submodules in ESP-IDF being out of date.

Please try running:
-  `git submodule update --init --recursive` to make sure all the submodules are up to date
- Then run `idf.py fullclean` then `idf.py build` to rebuild the project

I'm closing this issue as it does not appear to be a bug. If so, please feel free to reopen.

closed 2021-09-29 by Dazza0

tianrking, 2022-12-22:
same problem 

ninja: error: '/home/w0x7ce/Desktop/AAA/micro_ros_espidf_component/libmicroros.a', needed by 'int32_publisher.elf', missing and no known rule to make it
ninja failed with exit code 1

with idf.py 4.3 ubuntu22.04