espressif/esp-idf #7043: cant overload bootloader_support component

chegewara, 2021-05-18:
Hi,
i am trying to change factory reset pin level to use HIGH instead of LOW (btw it would be nice it selectable from menuconfig as feature), which requires to overload 2 components:
- bootloader
- bootloader_support

To do so i am having copy of both components in project components folder, but `bootloader_support` is used from esp-idf:
```
/home/chegewara/upwork/ota-over-ble-syrp/components/bootloader/subproject/main/bootloader_start.c: In function 'selected_boot_partition':
/home/chegewara/upwork/ota-over-ble-syrp/components/bootloader/subproject/main/bootloader_start.c:82:13: error: too many arguments to function 'bootloader_common_check_long_hold_gpio'
         if (bootloader_common_check_long_hold_gpio(CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET, CONFIG_BOOTLOADER_HOLD_TIME_GPIO, CONFIG_BOOTLOADER_PIN_LEVEL_FACTORY_RESET) == 1) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/chegewara/upwork/ota-over-ble-syrp/components/bootloader/subproject/main/bootloader_start.c:18:
/home/chegewara/esp/esp-idf/components/bootloader_support/include/bootloader_common.h:81:22: note: declared here
 esp_comm_gpio_hold_t bootloader_common_check_long_hold_gpio(uint32_t num_pin, uint32_t delay_sec);
 ```

projectgus, 2021-05-31:
Hi @chegewara,

Sorry for the slow reply.

We'll take the request for configurable levels on board, thanks (a PR would also be welcome if you happen to have one!)

I think to override bootloader_support component in the bootloader requires copying it to `project_dir/bootloader/subproject/components/` rather than `project_dir/components`. This may result in a different bootloader_support being built for app vs bootloader,but if you only need to override the bootloader behaviour then this may be acceptable.

We know this is a shortcoming in the build system bootloader implementation for now. We're working on some better options for overriding parts of bootloader functionality.

Angus

chegewara, 2021-05-31:
Hi @projectgus 
I think it is acceptable solution and since it is very rare case to override bootloader_support we can leave it this way or give it a low priority.
In my case the problem is solved anyway.

Thanks

closed 2021-07-16 by espressif-bot