espressif/esp-idf #7044: Uart IRAM optimization crashing

formus14, 2021-05-18:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- IDF version: v4.2

## Problem Description

If the **CONFIG_UART_ISR_IN_IRAM** SdkConfig option is enabled, the interrupt driver may crash because some resources may not be allocated in internal ram depending on another configuration (CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL).

If **CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL** is very small, like 4 Bytes, some components in the uart_driver_install function (components/driver/uart.c) will not be allocated in the internal ram.

Following allocation have to be fixed to work properly in all cases
```
p_uart_obj[uart_num]->xQueueUart = xQueueCreate(queue_size, sizeof(uart_event_t)); 
p_uart_obj[uart_num]->rx_ring_buf = xRingbufferCreate(rx_buffer_size, RINGBUF_TYPE_BYTEBUF);
p_uart_obj[uart_num]->tx_ring_buf = xRingbufferCreate(tx_buffer_size, RINGBUF_TYPE_NOSPLIT);
```

**Those 3 allocations have to be replaced by xQueueCreateStatic and xRingbufferCreateStatic when the CONFIG_UART_ISR_IN_IRAM option is enabled**

for reference , i2c driver has the driver installation in a proper way
i2c_driver_install function (components/driver/i2c.c) use xQueueCreateStatic on internal memory case **ESP_INTR_FLAG_IRAM** was enabled!
as shown below from function i2c_driver_install ()
```
            if ( !(intr_alloc_flags & ESP_INTR_FLAG_IRAM) ) {
                p_i2c->cmd_evt_queue = xQueueCreate(I2C_EVT_QUEUE_LEN, sizeof(i2c_cmd_evt_t));
            } else {
                p_i2c->evt_queue_storage = (uint8_t *)heap_caps_calloc(I2C_EVT_QUEUE_LEN, sizeof(i2c_cmd_evt_t), MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
                if ( p_i2c->evt_queue_storage == NULL ) {
                    ESP_LOGE(I2C_TAG, I2C_DRIVER_MALLOC_ERR_STR);
                    goto err;
                }
```

uart_driver_delete has t be also modified to free the statically generated queue and ringbuffer resources that shall be used case  **ESP_INTR_FLAG_IRAM** was enabled!

### Code to reproduce this issue

- Enable CONFIG_UART_ISR_IN_IRAM in the SdkConfig
-  Set CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL to 4 
- Use any uart example, Crash will occur

Alvin1Zhang, 2021-05-19:
Thanks for reporting.

thomasweichselbaumer, 2021-07-08:
Any new informations about this bug?

formus14, 2021-08-02:
@Alvin1Zhang  Issue exists in IDF 4.2.2 as well .. 

Lost packets on the uart occur when the **CONFIG_UART_ISR_IN_IRAM** is unticked as well! 

any updates on that issue ?

Alvin1Zhang, 2021-08-09:
@formus14 Thanks for reporting and sorry for the very slow turnaround. Thanks for being patient while waiting for the fix, we had a fix under internal reviewing and will be back ported to release/4.2. The issue will be closed automatically once the fix is available on GitHub. Thanks.

closed 2021-08-11 by espressif-bot