espressif/esp-idf #7045: esp_http_client_perform can be connecting endlessly when is_async = true

jef-sure, 2021-05-18:
----------------------------- Delete below -----------------------------

## Environment

- Development Kit:      ESP32-Wrover-Kit
- Chip is ESP32-D0WDQ6 (revision 1)

- IDF version: 4.2
- Using an IDE?:  Eclipse

## Problem Description

I copied https_async example from [esp_http_client_example.c](https://github.com/espressif/esp-idf/blob/master/examples/protocols/esp_http_client/main/esp_http_client_example.c):
```c
    while (1) {
        err = esp_http_client_perform(client);
        if (err != ESP_ERR_HTTP_EAGAIN) {
            break;
        }
    }
```

I spent several evenings to understand why sometimes after several hours of successful work this loop sometimes doesn't break out. It looks like sometimes after some network error while connecting `esp_http_client_perform(client);` returns always `ESP_ERR_HTTP_EAGAIN`. 

### Expected Behavior

No unendless loop.

### Steps to reproduce

My project tries to retrieve current weather from [openweathermap](https://openweathermap.org/) every minute. 

### Workaround 

I added timeout to the loop: [my http_client_request_get](https://github.com/jef-sure/dgx_clock/blob/main/main/http_client_request.c#L55)

shubhamkulkarni97, 2021-06-07:
Hi @jef-sure, thanks for reporting this issue.

If you are facing issue after network issue is occurred, can you try adding `.keep_alive_enable = true` in `esp_http_client_config_t`?

This will enable keep-alive timeout on the connection and report error to upper layer when network goes down.

Hope this fixes the issue.

Thanks,
Shubham

shubhamkulkarni97, 2021-06-14:
@jef-sure, Any updates on the issue?

jef-sure, 2021-06-14:
I think the problem comes from this condition: https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/esp_http_client.c#L1028
```
                    if (client->is_async && err == ESP_ERR_HTTP_CONNECTING) {
                        return ESP_ERR_HTTP_EAGAIN;
                    }
```
But I was unable to fine exact reason for this condition. 
The problem is not easy to trigger, sometimes I had to wait for 3 days. So, I made my workaround and reported.

shubhamkulkarni97, 2021-07-14:
@jef-sure I was not able to reproduce this issue in esp_http_client example. Could you provide a branch to reproduce this issue in esp_http_client example?

Alvin1Zhang, 2021-07-19:
@jef-sure Thanks for reporting, would you please help share any updates for the issue? Thanks.

jef-sure, 2021-07-19:
I have no updates -- I just use my workaround. As I said, the problem is not easy to trigger, I have no reliable case to reproduce it.

Alvin1Zhang, 2021-07-19:
@jef-sure Thanks for sharing the updates, feel free to reopen.

peter-pakanun, 2022-07-19:
I've recently come across this exact same issue.

If ESP lose connection to Wi-Fi It will get out of the loop with `ESP_ERR_HTTP_MAX_REDIRECT` error code.
```
[UPLOADER] Posting reading
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
E (425646) HTTP_CLIENT: Error, reach max_redirection_count count=0
E (425646) HTTP_CLIENT: Error response
[UPLOADER] Debug Reading: [28673] ESP_ERR_HTTP_MAX_REDIRECT
[UPLOADER] Post Reading http error: ESP_ERR_HTTP_MAX_REDIRECT
```

But if ESP still connected to Wi-Fi but the router lose connection to the internet I'll trigger this endless loop.

```
...
[UPLOADER] Debug Reading: [0] ESP_OK
[UPLOADER] Reading Posted with Status 200

> Router disconnected from WAN.

[UPLOADER] Posting reading
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN

> Router reconnected to WAN.

[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
...
```
Note that while this is happening, `esp_http_client_perform` also taking longer time to return (around ~5 seconds)

I can reproduce this issue 5 times out of 5 by unplugging my fiberoptic cable.

Running separate client *after* the reconnected return `ESP_OK` as expected, but the running connection still return `ESP_ERR_HTTP_EAGAIN` (and taking 5 seconds to run)

```
> Router disconnected from WAN.

[UPLOADER] Posting reading
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN

> Router reconnected to WAN.

[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Posting logs
[UPLOADER] Debug Logs: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Logs: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Logs: [0] ESP_OK
[UPLOADER] Logs Posted successfully, 1 entries posted.
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
[UPLOADER] Debug Reading: [28679] ESP_ERR_HTTP_EAGAIN
...
```

Enable keep-alive timeout didn't help.

## Environment
Board: M5Stamp C3 (ESP32-C3)
IDF: 4.4.1
IDE: PlatformIO

ibrahimuslu, 2025-03-24:
I  have discovered that this problem occurs when .async = true set in http config.

bryghtlabs-richard, 2025-06-25:
@Alvin1Zhang , impacts our 5.1.2-based fork. Will try to make a reproducible case.

bryghtlabs-richard, 2025-06-25:
Can reproduce on supported ESP-IDF 5.1.6, @Alvin1Zhang, please reopen.

### Initial discovery conditions: "Stress Test" wifi network

- 150 kbps down
- 100 kbps up
- drop every 10th packet
- esp_http_client connected to nginx nginx 1.18 server that allows 100 requests/response per HTTP connection
- Forked ESP-IDF 5.1.2, or 5.1.6
- Local network: Internet <-> ISP <-> ISP Router <-> Dev PC with bridging <-> Stress Test WiFi <-> ESP32-WROOM-32 stick

It seems this sometimes triggers some sort of sequencing difference. It usually works and doesn't get stuck. But once it gets stuck, it never recovers and only logs:

```
D (257182) esp-tls: select() timed out
D (257182) HTTP_CLIENT: Connection not yet established
D (257182) HTTP_CLIENT: Begin connect to: https://194.71.11.165:443
D (257186) esp-tls: connecting...
...~1x esp_http_client config timeout delay here
D (258194) esp-tls: select() timed out
D (258194) HTTP_CLIENT: Connection not yet established
D (258194) HTTP_CLIENT: Begin connect to: https://194.71.11.165:443
D (258198) esp-tls: connecting...
```

It's something strange for sure.

bryghtlabs-richard, 2025-06-25:
Seems to me like a bug in esp_tls.c `esp_tls_low_level_conn()`.

During the `ESP_TLS_CONNECTING` phase, it calls select(), but never checks SO_ERROR. In Test3 above, SO_ERROR eventually results in 113, EHOSTUNREACH. Hooking this up causes the connection failure to bubble up to the example client, which returns `ESP_ERR_HTTP_CONNECT` - seems reasonable to me. Took 18 seconds after starting the connection, sample size of 1.