espressif/esp-idf #7046: Pairing / Bonding error with CC2640R2F

twomas, 2021-05-19:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-D0WD-V3 (revision 3)
- IDF version (run ``git describe --tags`` to find it): v4.2.1
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
Interoperability issue with TI CC2640R2F
TI is central and trying to connect towards ESP peripheral configured with "just works"
Pairing fails every time.
It seems that it takes so much time to answer the Pairing Random Packet from frame 461 to 2794
So the question is about this delay. Why it takes almost 10 seconds?

## Debug Logs
See cfa file in TI3_5_Stack_Dongle_ESP_PairingFailure.zip

[TI3_5_Stack_Dongle_ESP_PairingFailure.zip](https://github.com/espressif/esp-idf/files/6507121/TI3_5_Stack_Dongle_ESP_PairingFailure.zip)

[image]

xiewenxiang, 2021-06-29:
I have test this issue with TI `CC2640R2F`:

 - IDF version: `v4.2.1`
 - Test example: `gatt_security_server` 
 - Test devices: `ESP32-WROOM-32` and `CC2640R2F`

But the reason for the pairing fail seems to lie on the `TI(CC2640R2F)` device side, Although sometimes `ESP32` is a little slower to respond, it is also within the protocol(30s), this does not result in a pairing fail. 

I have done a number of tests , and provided capture logs, found that the failure should be because the `CC2640R2F` repeated retransmission.

Sometimes, `ESP32` send response a bit slower. We will further analyze the reasons and give you a timely reply ASAP.

[image]
[image]
[image]
[image]
[image]

xiewenxiang, 2021-06-30:
Hi @twomas 

We have made a further analysis for this problem. In order to avoid the loss of the packet, we did the test in the shielding box. Please check the attachment.

The packet where the problem starts is on `frame#301`:

- `Frame#295`   *Master*->*Slave*    **SN:0**   **NESN:0**
- `Frame#296`   *Slave*->*Master*    **SN:0**   **NESN:1**
- `Frame#297`   *Master*->*Slave*    **SN:1**   **NESN:1**
- `Frame#298`   *Slave*->*Master*    **SN:1**   **NESN:0**
- `Frame#299`   *Master*->*Slave*    **SN:0**   **NESN:0**
- `Frame#300`   *Slave*->*Master*    **SN:0**   **NESN:1**
- `Frame#301`   *Master*->*Slave*    **SN:1**   **NESN:0**       `TI want ESP retransmission , because the NESN set to 0`.
- `Frame#302`   *Slave*->*Master*    **SN:0**   **NESN:0**       `ESP retransmission`
- `Frame#303`   *Master*->*Slave*    **SN:0**   **NESN:1** 
- `Frame#304`   *Slave*->*Master*    **SN:1**   **NESN:1**
- `Frame#305`   *Master*->*Slave*    **SN:1**   **NESN:1**   `TI want ESP retransmission , because the NESN set to 1`.
- `Frame#306`   *Slave*->*Master*    **SN:1**   **NESN:0**   `ESP retransmission`
- `Frame#307`   *Master*->*Slave*    **SN:0**   **NESN:1**   `TI want ESP retransmission , because the NESN set to 1`.
- `Frame#308`   *Slave*->*Master*    **SN:1**   **NESN:1**   `ESP retransmission`
- `Frame#309`   *Master*->*Slave*    **SN:1**   **NESN:1**   `TI want ESP retransmission , because the NESN set to 1`.
- `Frame#310`   *Slave*->*Master*    **SN:1**   **NESN:0**   `ESP retransmission`

The following is still all the case of retransmission, you can check the attachment, So maybe TI needs to take a look at this issues.

xiewenxiang, 2021-06-30:
[capture_6.zip](https://github.com/espressif/esp-idf/files/6737878/capture_6.zip)

zhp0406, 2023-09-11:
Have you resolved your issue?

closed 2025-03-13 by Alvin1Zhang