espressif/esp-idf #7047: PPPoS unable to set authentication,will always fail.

persan666, 2021-05-19:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-beta3-195-g6be10fab0
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [Yes (Eclipse)]
- Power Supply:         [Battery]

## Problem Description

Unable to set authentication for PPPoS.
The call to 'esp_netif_ppp_set_auth' will always fail and return invalid parameter even though the parameters are correct. 
The evaluation of macro '_IS_NETIF_POINT2POINT_TYPE' evaluates to true if the parameters are correct.

This is the implementation which not work:
```cpp
esp_err_t esp_netif_ppp_set_auth(esp_netif_t *netif, esp_netif_auth_type_t authtype, const char *user, const char *passwd)
{
    if (_IS_NETIF_POINT2POINT_TYPE(netif, PPP_LWIP_NETIF)) {
        return ESP_ERR_ESP_NETIF_INVALID_PARAMS;
    }
#if PPP_AUTH_SUPPORT
    lwip_peer2peer_ctx_t *ppp_ctx = (lwip_peer2peer_ctx_t *)netif->related_data;
    assert(ppp_ctx->base.netif_type == PPP_LWIP_NETIF);
    pppapi_set_auth(ppp_ctx->ppp, authtype, user, passwd);
#else
    ESP_LOGE(TAG, "%s failed: No authorisation enabled in menuconfig", __func__);
    return ESP_ERR_ESP_NETIF_IF_NOT_READY;
#endif
    return ESP_OK;
}
```
And should be changed to something like this (Notice the '!' before the macro '_IS_NETIF_POINT2POINT_TYPE'):
```cpp
esp_err_t esp_netif_ppp_set_auth(esp_netif_t *netif, esp_netif_auth_type_t authtype, const char *user, const char *passwd)
{
    if (!_IS_NETIF_POINT2POINT_TYPE(netif, PPP_LWIP_NETIF)) {
        return ESP_ERR_ESP_NETIF_INVALID_PARAMS;
    }
#if PPP_AUTH_SUPPORT
    lwip_peer2peer_ctx_t *ppp_ctx = (lwip_peer2peer_ctx_t *)netif->related_data;
    assert(ppp_ctx->base.netif_type == PPP_LWIP_NETIF);
    pppapi_set_auth(ppp_ctx->ppp, authtype, user, passwd);
#else
    ESP_LOGE(TAG, "%s failed: No authorisation enabled in menuconfig", __func__);
    return ESP_ERR_ESP_NETIF_IF_NOT_READY;
#endif
    return ESP_OK;
}
```

### Expected Behavior
Should not return error if the parameters are correct
### Actual Behavior
return error if the parameters are correct

RobNuttall, 2021-05-19:
We're also seeing this issue on 4.3-beta3

AxelLin, 2021-05-19:
@david-cermak 
This needs an urgent fix.

AxelLin, 2021-06-04:
v4.3-rc still includes this bug.

closed 2021-06-06 by espressif-bot

AxelLin, 2021-06-15:
Unfortunately, the v4.3 release does **not** include this fix.

persan666, 2021-06-16:
I have postpone my own software release just due to this bug. Now I have to deliver a dirty one for my customers unfortunately.

What was the reason for not including this fix in the 4.3 release?