espressif/esp-idf #7049: ESP32S2 - fail to configure MBEDTLS_EXTERNAL_MEM_ALLOC

oachkatzalschwoaf, 2021-05-19:
## Environment

- Development Kit:      none
- ESP32-S2-WROVER
- IDF version: 4.2.1
- Build System:  cmake
- Compiler version: xtensa-esp32s2-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
   
- Operating System:     Windows
- Using an IDE: Visual studio code - Espressif IDF 1.0.3

## Problem Description

sdkconfig: CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC can`t be configured for ESP32S2

### Expected Behavior

``CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y``  in sdkconfig

### Actual Behavior

``CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC`` is always reverted to  ``CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC``

### Steps to reproduce

1. open example "HTTPS_REQUEST" (in my case with `VSC` and extension `Espressif IDE`)

=>Espressif IDF in VSC automaticly use ESP32S2

2. open SDK Configuration editor

3. press SAVE
 > sdkconfig gets configured:

```
CONFIG_IDF_TARGET="esp32s2"
CONFIG_IDF_TARGET_ESP32S2=y

... 
# CONFIG_ESP32S2_SPIRAM_SUPPORT is not set
...
CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC=y

```

based on ``sdkconfig.ci`` it should be (however i use ESP32S2 => therefore disabled SPIRAM_SUPPORT disabled might be ok?):
```
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
```

4. enable ``CONFIG_IDF_TARGET_ESP32S2`` 
 and set 
mbedTLS - Memory allocation strategy to ``external SPIRAM``  (`` CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC`` )

ERROR:
```
{"version":2,"save":"c:\\esp\\esp-idf\\examples\\protocols\\https_request\\sdkconfig"}
Saving config to c:\esp\esp-idf\examples\protocols\https_request\sdkconfig...

{"version": 2, "values": {}, "ranges": {}, "visible": {}}

{"version": 2, "set": { "MBEDTLS_EXTERNAL_MEM_ALLOC": true }}

---------------------------ERROR--------------------------

Error: The following config symbol(s) were not visible so were not updated: MBEDTLS_EXTERNAL_MEM_ALLOC

-----------------------END OF ERROR-----------------------
{"version": 2, "values": {}, "ranges": {}, "visible": {}, "error": ["The following config symbol(s) were not visible so were not updated: MBEDTLS_EXTERNAL_MEM_ALLOC"]}

{"version":2,"save":"c:\\esp\\esp-idf\\examples\\protocols\\https_request\\sdkconfig"}
Saving config to c:\esp\esp-idf\examples\protocols\https_request\sdkconfig...

{"version": 2, "values": {}, "ranges": {}, "visible": {}}
```

it seems there is no way to use external PSRAM for MbedTLS with ESP32S2 as it always gets reverted 
=> for my remaining Project PSRAM is recognized and works, but i need it for https

[sdkconfig.LOG](https://github.com/espressif/esp-idf/files/6510394/sdkconfig.LOG)

negativekelvin, 2021-05-20:
https://github.com/espressif/esp-idf/commit/cf4b9ef854651596a6014ac47e209d426b901c29

mahavirj, 2021-05-20:
@oachkatzalschwoaf Please check if commit linked above by @negativekelvin fixes your problem. We will backport similar fix for `release/v4.2` branch as well, it shall appear in v4.2.2 official release.

oachkatzalschwoaf, 2021-05-20:
thanks, my issue is fixed by: cf4b9ef854651596a6014ac47e209d426b901c29

mahavirj, 2021-06-07:
Fixed with https://github.com/espressif/esp-idf/commit/06bf491f1324568862c2ae4e9ca15f23425c5618 on v4.2 branch

closed 2021-06-07 by mahavirj