espressif/esp-idf #7051: vscode error links not working with idf build system

qt1, 2021-05-20:
The vscode environment gives a very good way to find the location of compilation errors.
This is done by ctrl-clicking the file location in the output or clicking lines in the "problems" pane.
/home/baruch/esp/hello_world/.vscode/tasks.json
This is a very important tool in the write-compile-debug work cycle.

Unfortunately the paths reported by cmake builds are relative to the CMakeLists file and not to the project root and therefoe vscode does not find the files.

Reconstruction:

1.  Copy hello_world example.
2. Create .vscode folder
3. Copy the attached file into .vscode and rename back to tasks.json [tasks.json.txt](https://github.com/espressif/esp-idf/files/6512166/tasks.json.txt)
4. source ../esp-idf/export.sh
5. run ````code .````
6. build using the script, e.g. press ctrl-shitf-B

At this point the program should compile OK.

Now create an error by adding the line

    #error find me
  
And build using ctrl-shift-B

The build fails.

ctrl-click the reported lines do not open the source.
Trying to open the "problems" also does not work correctly (file not found)

Please advise.
Thanks

projectgus, 2021-05-20:
Hi @qt1,

> Unfortunately the paths reported by cmake builds are relative to the CMakeLists file and not to the project root and therefoe vscode does not find the files.

It looks like what's happening is that the path is relative to the build directory (where CMake & ninja sub-processes are run) and not the project root.

Have you tried using the [VSCode ESP-IDF Integration](https://github.com/espressif/vscode-esp-idf-extension/) instead?

chegewara, 2021-05-20:
Yes, it is useful and problematic at the same time. But you can use to it quick. What i do is to CTRL + click, then delete 2 dots and slash and voila, you are in the line of code that failed..

brianignacio5, 2021-05-20:
@qt1 In our vscode extension for ESP-IDF, we have a [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) document to help you configure the `c_cpp_properties.json`.

The thing with the relative paths is a problem matcher configuration issue in your tasks.json. In our extension we provide [template tasks.json](https://github.com/espressif/vscode-esp-idf-extension/blob/master/templates/.vscode/tasks.json) that should target those build problems with correct links to the source file.

Please consider trying [our extension](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension&ssr=false#overview) which provide such build, flash and other tasks as well as many other [features](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/FEATURES.md)

Alvin1Zhang, 2021-05-31:
@qt1 Thanks for reporting. Would you please help share if any updates or details for this issue? Thanks.

qt1, 2021-05-31:
I have tried the extension.
Unfortunately it is very intrusive. I had too much clutter and I could not work with other python projects even after removing it.
It seems like it is designed for someone that only works with the idf, but not convenient for someone that uses vscode for many other types of work.. Maybe a separate user or even a docker environment will give the appropriate separation..

The bottom line - I still do not have a simple working vs setting file for idf

Thanks

closed 2021-06-07 by Alvin1Zhang