espressif/esp-idf #7053: What is the proper way to init/deinit between STA and AP mode

wwll123, 2021-05-20:
## Environment

- Module or chip used:  esp32-wrover-b
- IDF version: v4.1
- Build System: Make
- Compiler version: esp-2020r2
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

ESP32 sometimes would be reboot during large file downloading. 
This issue would only happen if ESP32 first enter AP mode and then enter STA mode.
So I suspect the issue is because of AP/STA mode init/deinit process.

### Expected Behavior

No reboot during large file downloading.

### Actual Behavior

ESP32 sometimes would be reboot during large file downloading after entering STA mode from AP mode. 

### Code to reproduce this issue
```c
// Two netif handler. One for STA mode and one for AP mode.
// Once they are inited. They would not be destroyed.
esp_netif_t* StaModeHandler;
esp_netif_t* ApModeHandler;
```

The interchange process of AP and STA mode would be:
Power up => WifiTaskInit() => WifiSTAModeDeinit() => WifiAPModeInit() => WifiAPModeStartServer() => WifiAPModeStopServer() => WifiAPModeDeinit() => WifiSTAModeInit()

WifiTaskInit()
```c
ESP_ERROR_CHECK(esp_netif_init());
ESP_ERROR_CHECK(esp_event_loop_create_default());
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
```

WifiSTAModeDeinit()
```c
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_START, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_CONNECTED, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_LOST_IP, &WifiEventHandler));
esp_wifi_stop();
esp_wifi_deinit();
```

WifiAPModeInit()
```c
if (ApModeHandler == NULL)
{
	esp_netif_config_t netif_config = ESP_NETIF_DEFAULT_WIFI_AP();
	ApModeHandler = esp_netif_new(&netif_config);
	assert(ApModeHandler);
}
esp_netif_attach_wifi_ap(ApModeHandler);
esp_wifi_set_default_wifi_ap_handlers();
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_AP_START, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_AP_STOP, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_AP_STACONNECTED, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_AP_STADISCONNECTED, &WifiEventHandler, NULL));
esp_netif_dhcps_stop(ApModeHandler);
esp_netif_ip_info_t ApIpInfo;
memset(&ApIpInfo, 0, sizeof(esp_netif_ip_info_t));
IP4_ADDR(
	&ApIpInfo.ip,
	DEFAULT_WIFI_AP_IP_1,
	DEFAULT_WIFI_AP_IP_2,
	DEFAULT_WIFI_AP_IP_3,
	DEFAULT_WIFI_AP_IP_4);
IP4_ADDR(
	&ApIpInfo.gw,
	DEFAULT_WIFI_AP_GW_1,
	DEFAULT_WIFI_AP_GW_2,
	DEFAULT_WIFI_AP_GW_3,
	DEFAULT_WIFI_AP_GW_4);
IP4_ADDR(
	&ApIpInfo.netmask,
	DEFAULT_WIFI_AP_NM_1,
	DEFAULT_WIFI_AP_NM_2,
	DEFAULT_WIFI_AP_NM_3,
	DEFAULT_WIFI_AP_NM_4);
esp_netif_set_ip_info(ApModeHandler, &ApIpInfo);
esp_netif_dhcps_start(ApModeHandler);
esp_wifi_set_mode(WIFI_MODE_AP);
esp_wifi_set_config(ESP_IF_WIFI_AP, &WifiApConfig);
esp_wifi_start();
```
WifiAPModeStartServer() 
```c
httpd_ssl_start(&HttpsServerHandler, &SerConfig);
httpd_register_uri_handler(HttpsServerHandler, &GetInfo);
```

WifiAPModeStopServer()
```c
if(HttpsServerHandler != NULL)
{
	httpd_ssl_stop(HttpsServerHandler);
	HttpsServerHandler = NULL;
}
if(ApModeHandler != NULL)
{
	esp_netif_dhcps_stop(ApModeHandler);
}
```

WifiAPModeDeinit()
```c
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_AP_START, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_AP_STOP, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_AP_STACONNECTED, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_AP_STADISCONNECTED, &WifiEventHandler));
esp_wifi_stop();
esp_wifi_deinit();
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));
```
WifiSTAModeInit()
```c
if(StaModeHandler == NULL)
{
	esp_netif_config_t netif_config = ESP_NETIF_DEFAULT_WIFI_STA();
	StaModeHandler = esp_netif_new(&netif_config);
	assert(StaModeHandler);
}
esp_netif_attach_wifi_station(StaModeHandler);
esp_wifi_set_default_wifi_sta_handlers();
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_START, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_CONNECTED, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_LOST_IP, &WifiEventHandler, NULL));
esp_wifi_set_mode(WIFI_MODE_STA);
esp_wifi_set_config(ESP_IF_WIFI_STA, &WifiStaConfig);
esp_wifi_start();
```

xueyunfei998, 2021-05-31:
hi @wwll123 

There is something wrong with the flow in these functions.

You can refer to the following:

esp_netif_t * ApModeHandler = NULL;
esp_netif_t * StaModeHandler = NULL;

wifi_init() {
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

}

Wifi_STA_Mode_Deinit() {

StaModeHandler = 
  if (StaModeHandler == NUll) {
  return;
  }
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_START, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_CONNECTED, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_LOST_IP, &WifiEventHandler));
    esp_err_t err = esp_wifi_stop();
    if (err == ESP_ERR_WIFI_NOT_INIT) {
        return;
    }
    ESP_ERROR_CHECK(err);
    ESP_ERROR_CHECK(esp_wifi_deinit());
    ESP_ERROR_CHECK(esp_wifi_clear_default_wifi_driver_and_handlers(StaModeHandler));
    esp_netif_destroy(StaModeHandler);
    StaModeHandler = NULL;
}

Wifi_AP_Mode_Deinit() {

  if (ApModeHandler == NUll) {
  return;
  }
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_START, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_CONNECTED, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &WifiEventHandler));
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_LOST_IP, &WifiEventHandler));
    esp_err_t err = esp_wifi_stop();
    if (err == ESP_ERR_WIFI_NOT_INIT) {
        return;
    }
    ESP_ERROR_CHECK(err);
    ESP_ERROR_CHECK(esp_wifi_deinit());
    ESP_ERROR_CHECK(esp_wifi_clear_default_wifi_driver_and_handlers(ApModeHandler));
    esp_netif_destroy(ApModeHandler);
    ApModeHandler = NULL;

}

Wifi_STA_Mode_init() {
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_WIFI_STA();
    StaModeHandler = esp_netif_new(&cfg);
    assert(StaModeHandler);
    esp_netif_attach_wifi_station(StaModeHandler);
    esp_wifi_set_default_wifi_sta_handlers();

ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_START, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_CONNECTED, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_LOST_IP, &WifiEventHandler, NULL));
esp_wifi_set_mode(WIFI_MODE_STA);
esp_wifi_set_config(ESP_IF_WIFI_STA, &WifiStaConfig);
esp_wifi_start();

} 

Wifi_AP_Mode_init {

esp_netif_ip_info_t ApIpInfo;
memset(&ApIpInfo, 0, sizeof(esp_netif_ip_info_t));
IP4_ADDR(
	&ApIpInfo.ip,
	DEFAULT_WIFI_AP_IP_1,
	DEFAULT_WIFI_AP_IP_2,
	DEFAULT_WIFI_AP_IP_3,
	DEFAULT_WIFI_AP_IP_4);
IP4_ADDR(
	&ApIpInfo.gw,
	DEFAULT_WIFI_AP_GW_1,
	DEFAULT_WIFI_AP_GW_2,
	DEFAULT_WIFI_AP_GW_3,
	DEFAULT_WIFI_AP_GW_4);
IP4_ADDR(
	&ApIpInfo.netmask,
	DEFAULT_WIFI_AP_NM_1,
	DEFAULT_WIFI_AP_NM_2,
	DEFAULT_WIFI_AP_NM_3,
	DEFAULT_WIFI_AP_NM_4);

    esp_netif_inherent_config_t esp_netif_config = _g_esp_netif_inherent_ap_config();
   esp_netif_config.ip_info = (esp_netif_ip_info_t*)&ApIpInfo;
    esp_netif_config_t netif_config = {
        .base = &esp_netif_config,
        .stack = ESP_NETIF_NETSTACK_DEFAULT_WIFI_AP
    };
    ApModeHandler = esp_netif_new(&netif_config);
    assert(ApModeHandler);
    esp_netif_attach_wifi_station(ApModeHandler);
    esp_wifi_set_default_wifi_ap_handlers();

ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_START, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_CONNECTED, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &WifiEventHandler, NULL));
ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_LOST_IP, &WifiEventHandler, NULL));
esp_wifi_set_mode(WIFI_MODE_AP);
esp_wifi_set_config(ESP_IF_WIFI_AP, &WifiApConfig);
esp_wifi_start();

}

xueyunfei998, 2021-08-06:
hi @wwll123 

Is there any problem? If not, I will close the problem.

closed 2021-11-17 by Alvin1Zhang