espressif/esp-idf #7054: Invalid Netkey Index after power reset in onoff client!!

sumathi98, 2021-05-20:
## Environment

- Development Kit      :  ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32U
- IDF version              :  v4.4.0
- Build System           :   idf.py
- Compiler version     :   8.4.0
- Operating System   :   Linux
- Using an IDE?         :   No
- Power Supply          :   external 5V

## Problem Description

     Iam working on ble mesh and i have issue in sending message to server from client after power reset. Iam using esp32 example ble mesh node on/off client.
     I have modified onoff_client example by creating multiple elements and gpio interrupt for controlling server based on switch press.
     I can provision the node and control the led on the server node from client, It works fine. once I reset the board Iam unable to transmit message from client. Though netkey is stored in flash, Iam getting below error.

## Debug Logs
<27>[0;31mE (12969) BLE_MESH: Invalid NetKeyIndex 0x0000<27>[0m<\r><\n>
<27>[0;31mE (12969) BLE_MESH: Failed to send client message 0x00008203<27>[0m<\r><\n>
<27>[0;32mI (12969) EXAMPLE: Generic client, event 1, error code -22, opcode is 0x8203<27>[0m<\r><\n>
<27>[0;32mI (12979) EXAMPLE: ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT<27>[0m<\r><\n>
<27>[0;32mI (12989) EXAMPLE_NVS: Store, key "onoff_client", length 6<27>[0m<\r><\n>
<27>[0;32mI (12989) EXAMPLE_NVS: Store, data: 00 00 00 00 01 93 <27>[0m<\r><\n>

chegewara, 2021-05-20:
Did you enable `persistent storage` in menuconfig for ble mesh?

sumathi98, 2021-05-20:
No I haven't enable persistent storage, Can you provide the path to enable in menuconfig.

chegewara, 2021-05-20:
Just use `/` to enable searching and search for `persistent`.
`BLE_MESH_SETTINGS`

sumathi98, 2021-05-20:
Thank you, but If persistent storage is enabled in menuconfig device is not advertising!!

chegewara, 2021-05-20:
Then it may be different issue. I didnt try with v4.4, but with older idf version ble mesh worked to me.
Maybe try `erase_flash`, just in case.

sumathi98, 2021-05-20:
Thank you solved the issue

closed 2021-05-20 by sumathi98