espressif/esp-idf #7055: AWS OTA job failure on esp-idf

srivardhan777, 2021-05-20:
Hello everyone,

I am trying to run an ota job from aws on esp32 (while running an amazon freeRTOS repo), then after the whole number of file blocks received I am getting an error and the job is getting failed. Below was the error I am getting

------------------------------------------------------------------------------------------------------------------------------------------------------------------------
I (86654) ota_pal: prvPAL_SetPlatformImageState, 3
W (86654) ota_pal: Set image as invalid!
I (86654) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
1540 8650 [OTA Agent Task] [prvIngestDataBlock] Error (1:0x000000) closing OTA file.
1541 8650 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
1542 8650 [OTA Agent Task] [prvProcessDataMessage] Aborting due to IngestResult_t error -2
W (86674) esp_ota_ops: otadata partition is invalid, factory/ota_0 is boot partition
E (86694) ota_pal: currently executing firmware not marked as valid, abort
1543 8654 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"FAILED","statusDetails":{"reason":"0x01000000: 0xfffffffe"}}
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------

My nvs partition is as follows:
----------------------------------------------------------------------
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x10000,24K,
otadata,data,ota,0x16000,8K,
phy_init,data,phy,0x18000,4K,
factory,app,factory,0x20000,1200K,
ota_0,app,ota_0,0x150000,1600K,
ota_1,app,ota_1,0x2e0000,200K,
file,data,spiffs,0x312000,24K,
storage,data,nvs,0x318000,64K,

mahavirj, 2021-05-21:
@srivardhan777 

>  ota_0,app,ota_0,0x150000,1600K,
>  ota_1,app,ota_1,0x2e0000,200K,

Any specific reason to keep non-symmetrical sizes for OTA partitions? What happens if you use default OTA demo from AFR?

>  (while running an amazon freeRTOS repo)

Please consider reporting issues at https://github.com/aws/amazon-freertos/issues (with more information as requested in issue template there).

srivardhan777, 2021-05-22:
Nothing serious about the non-symmetrical sizes, focused much on the ota_0 as the tasks I am going to give runs mostly on it and nothing goes to ota_1. Thank you for suggesting the amazon-freertos issues repo.

mahavirj, 2021-05-24:
> focused much on the ota_0 as the tasks I am going to give runs mostly on it and nothing goes to ota_1.

OTA update will write to second partition, and then active partition will be swapped to `ota_1`. Hence, it is important to have symmetrical sizes for both partitions. You may find article at [useful](https://medium.com/the-esp-journal/ota-updates-framework-ab5438e30c12) in this context.

mahavirj, 2021-05-24:
@srivardhan777 I am closing this issue, please feel free to reopen if there are more questions.

closed 2021-05-24 by mahavirj