espressif/esp-idf #7056: wifi mesh warning, mesh_schedule.c [WND-RX]1200 ms timeout

shiweichn, 2021-05-21:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROVER-E]
- IDF version :v 4.2-477-g6b66930fb
- Build System:         [idf.py]
- Operating System:     [macOS]
- Using an IDE?: yes, clion.
- Power Supply:         [USB]

## Problem Description
I am developing a product based on Wi-Fi mesh. My program ran normally at first, but after running for a while, some warning messages appeared. I don't know the meaning of these warning messages and whether it affects my program. Please help me, this question is urgent.

The following is the log output by the program. This problem seems to cause the node to be unable to send data to the root node.

W (58432004) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40847, no_wnd_count:4, timeout_count:0
W (58434614) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40847, no_wnd_count:7, timeout_count:0
W (58440094) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40851, no_wnd_count:10, timeout_count:0
W (58441294) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40851, no_wnd_count:10, timeout_count:1
W (58442494) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40851, no_wnd_count:10, timeout_count:2
W (58445854) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40856, no_wnd_count:10, timeout_count:0
W (58450084) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40856, no_wnd_count:13, timeout_count:0
W (58452344) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40860, no_wnd_count:0, timeout_count:0
W (58456044) mesh: [mesh_schedule.c,3072] [WND-RX]1200 ms timeout, seqno:0, xseqno:40860, no_wnd_count:10, timeout_count:0

shiweichn, 2021-12-30:
?

shenjun7, 2022-07-18:
This log marks the upstream status of the data．
If the parent switch occurs, the child node will be temporarily affected. After the network is stable, it will return to normal.
If this problem is caused by the excessive amount of upstream data, you can use esp_mesh_set_xon_qsize() to increase the qsize. You can also ease the pressure on upstream data by reducing the maximum number of connections.
If the ps function is turned on, the node is only awake 10% of the time by default during the cycle. You can increase the duty cycle or turn off the ps function.

closed 2022-07-27 by Alvin1Zhang