espressif/esp-idf #7057: Console example failed to set large string in NVS

k4keval, 2021-05-21:
Hello,

I am trying to execute a console example that calls nvs_set command on ESP32-S2 WROVER board.
The amount of data that I am trying to store in NVS is around 2048 bytes.
But when I try to write the data, it seems that the complete chunk is not getting stored on the NVS memory.
Here is screen shot of my application logs for the same:
[image]

Now while retrieving the same data from NVS I'm able to read even lesser data than what was written.
Here is a screen shot for the receiving cycle:
[image]

Request you to suggest a proper way of writing data chunks of such large size.

0xjakob, 2021-05-25:
Hi @k4keval, thanks for posting the issue!

Could you please post the log as text? This makes it much easier to search and reproduce the exact actions leading to the issue.

I strongly suggest to use blob format for large chunks of binary data. Strings have the problem that they *must* fit into one NVS page which is only around 4000 bytes. Even worse, if the pages are fragmented already, there must be a gap which fits the whole string. If this isn't the case, then the string won't be written. Blobs will be split into smaller chunks if the space is fragmented.

That being said, it looks a bit as if the string gets only half-written. But currently, I'm not sure whether it's an issue with NVS itself of the console component.

0xjakob, 2021-05-25:
I just reproduced this with a simple example using the same commands:
```
esp32> nvs_set certificate str -v <some long text in base64>
I (818390) cmd_nvs: Value stored under key 'certificate'
esp32> nvs_get certificate str
<some truncated version of that text>
```

We'll take a look at this.

k4keval, 2021-05-27:
Hello @0xjakob!
I have a couple more observations when trying to write in blob format in svn:
1. If I try to write a large chunk of data as value(in single shot) in nvs_set command, some of the non-numeric characters are converted to junk characters and errors as shown below:
_tools> nvs_set certificate blob -v 1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234767890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234767890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABC▒EF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890▒BCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234767890ABCDEF12345678;0ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890▒BCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234767890ABCDEF1234567890ABCDEF1234567890ABCD▒F1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890▒BCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABC▒EF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234767890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCD▒F1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890▒BCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF12345678;0ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890▒BCDEF
E (7327477) cmd_nvs: Blob data must contain even number of characters
E (7327477) cmd_nvs: error while set_value_in_nvs - ESP_ERR_NVS_TYPE_MISMATCH
Command returned non-zero error code: 0x1 (ERROR)
tools>_

2. If I write a large chunk of data as value(by gradually adding characters) in nvs_set command, some errors are still shown as below:
_tools> nvs_set certificate blob -v 1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF12345678 [... cut ...] CDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF
E (7452827) cmd_nvs: Blob data must contain even number of characters
E (7452827) cmd_nvs: error while set_value_in_nvs - ESP_ERR_NVS_TYPE_MISMATCH
Command returned non-zero error code: 0x1 (ERROR)
tools>_

Please share your insights for the same as well.

k4keval, 2021-06-01:
Hello,
Is there any other update on this particular issue?
Your support is very much appreciated.
Thank you. :-)

0xjakob, 2021-06-01:
@k4keval Sorry for the long delay!

I had a look at setting strings. There are two issues with setting strings:
1. The command buffer is way too small. The [default in the example is 256](https://github.com/espressif/esp-idf/blob/master/examples/system/console/main/console_example_main.c#L104) bytes.
1. Something goes wrong when reading a large blob of characters in the `nvs_set` command.

The first issue can be easily fixed. Just increase the size. Of course, the ESP32 only has a quite limited amount of memory. But 6 or 8K should still not be a big issue.

We need to further investigate the second issue. We will let you know about progress until the end of this week.

k4keval, 2021-06-02:
@0xjakob ,
Thank you for you suggestion.
After applying the changes mentioned in point 1, I was able to write bigger chunk of data to NVS.
However the data gets corrupted while writing and the same corrupted is received while nvs_get(already know)
```
tools> nvs_namespace cloud
I (38228) cmd_nvs: Namespace set to 'cloud'
tools> nvs_set certificate str -v 234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF12345678;0ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF12345678;0ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABC▒EF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCD▒F1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDE▒1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABC▒EF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890A▒CDEF1234567890ABCDEF1234567890ABCDEF
I (74468) cmd_nvs: Value stored under key 'certificate'
tools> nvs_get certificate str
I (110688) cmd_nvs: 1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF12345678;0ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF12345678;0ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF123456790ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDEF1034567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDEF123456790ABCDEF1234567890ABCDEF1234567890ABCEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890BCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF123567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF124567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCDEF1234567890ABCDE1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ACDEF1234567890ABCDEF1234547890ABCDEF1234567890ABCD1234567890ABCDE1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ACDEF1234567890ABCDEF1234567890ACDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDE1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ABCDEF1234567890ACEF1234567890ABCDEF1234567890ABCDEF
tools>
```
Here, what I'm trying to write is 1234567890ABCDEF data 128 times, i.e. 2048 characters.
The same problem is observed when I try to write the data as string or blob.

Thank you.

k4keval, 2021-06-04:
@0xjakob 
Is there any API other than nvs_set_str() to write a string to NVS?
In a different approach, I'm trying to write to NVS with a custom program without bringing console into picture.
Since nvs_set_str() is used by the console example, I am looking for a different API that can be used by my application to write to NVS.
Please let me know if there is any other way to do it as well.
Appreciate your support for this specific issue.

Thank you.

0xjakob, 2021-06-04:
@k4keval According to my investigation, the issue is not with the `nvs_set_str()` function. It rather happens during reading from console. You should still be able to write nvs data via the nvs functions with another app using an alternative approach to read the data from a source.

0xjakob, 2021-06-04:
What is your use case, btw? Do you need to write this data only once or frequently? If you want to store a certificate, writing it to a custom partition instead of NVS might be the better approach.

0xjakob, 2021-06-04:
@k4keval It works, even with string, if you increase the [maximum command length of the console](https://github.com/espressif/esp-idf/blob/master/examples/system/console/main/console_example_main.c#L104) *and* the [RX buffer size of the uart driver](https://github.com/espressif/esp-idf/blob/master/examples/system/console/main/console_example_main.c#L95) in the example. 

I set both to 8000 and then succeeded. It took a while until the certificate had been read via the console, though. I tested with a [certificate from the esp-idf repo](https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_x509_bundle/certs/Digicert_global_root.pem). As already mentioned, if you want to just store a certificate, it's probably better to put it into a custom partition.

0xjakob, 2021-06-04:
Have to correct myself: currently, our provisioning/manufacturing tools are made for NVS, not raw partitions. So it's probably better to store the certificate as string or blob.

k4keval, 2021-06-07:
Thank you @0xjakob for all the information.

Both kinds of used cases are to be implemented at my end. Writing data once as well as repeatedly.
I tried to increase the [maximum command length of the console](https://github.com/espressif/esp-idf/blob/master/examples/system/console/main/console_example_main.c#L104) and the [RX buffer size of the uart driver](https://github.com/espressif/esp-idf/blob/master/examples/system/console/main/console_example_main.c#L95) in the example.
As a result, over 4k data was written to NVS but it was getting corrupted. (perhaps because of console bug you mentioned above)

So now, I'll proceed with the `nvs_set_str()` function and test by avoid writing larger data over the console.

0xjakob, 2021-06-07:
@k4keval That's great to hear! Did you have a look at the [NVS partition generator utility](https://github.com/espressif/esp-idf/tree/master/components/nvs_flash/nvs_partition_generator)? This tools can create nvs partition images which you can directly flash onto your device during manufacturing. We will follow up with checking whether there's an issue with the uart driver.

k4keval, 2021-06-07:
@0xjakob 
Yes using the above mentioned utility, I created bin file from the csv and verified the behaviour of nvs_set for sting and blob data types.
Thank you for the information.

closed 2021-07-16 by espressif-bot