espressif/esp-idf #7059: esp-idf toolchain setup on Ubuntu 21

kindmartin, 2021-05-21:
Hi all there / here at this github. I m trying to setup a toolchain on unbuntu21 as a prerequisite to compile a ESP32 Micropython fork. but even if i read all docs /guides seems this 2021 something changed on python support so the indicated scripts just do no works.

I followed this setup guide

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html
but as soon I try to add prerequisites a invalid python / n

_sudo apt-get install git wget libncurses-dev flex bison gperf python python-pip python-setuptools cmake ninja-build ccache libffi-dev libssl-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python-is-python2' instead of 'python'
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

E: Package 'python-pip' has no installation candidate_

I do have pip
_martin@ubuntu:~$ python -m pip --version
pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)_

I tryied other guides. reinstalled ubuntu, etc./ 

Im not sure if this is a bug o more a request/need to update documentation for setup, but since seems some script are base on python2.7 witch has lost support since january this year probablyt his isuue im reporting is related to that fact.

<  M

dobairoland, 2021-05-21:
Hi @kindmartin. On the page you've linked by yourself, I see `python3 python3-pip python3-setuptools` and not `python python-pip python-setuptools` as shown in your report. Is it possible that you have used and older version of the documentation? Please try to installed the packages from that page.

dobairoland, 2021-06-11:
I'm closing this for lack of activity. I assume an old documentation page was used. Please let me know if I'm wrong.

closed 2021-06-11 by dobairoland