espressif/esp-idf #7061: ESP32S3 Beta2 not able to start BLE MESH onoff server model

nitinbhosaleemb, 2021-05-22:
ESP-ROM:esp32s3-20200522
Build:May 22 2020
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0100,len:0x1788
load:0x403b6000,len:0x7f8
load:0x403ba000,len:0x2aa0
entry 0x403b61a0
I (21) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (21) boot: compile time 21:58:59
I (21) boot: chip revision: 0
I (25) boot.esp32s3: SPI Speed      : 80MHz
I (30) boot.esp32s3: SPI Mode       : DIO
I (35) boot.esp32s3: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
W (45) bootloader_random: RNG for ESP32-S3 not currently supported
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=2b4f8h (177400)                                                                                                              map
I (132) esp_image: segment 1: paddr=0003b520 vaddr=3fc912c0 size=0418ch ( 16780)                                                                                                              load
I (136) esp_image: segment 2: paddr=0003f6b4 vaddr=40374000 size=00964h (  2404)                                                                                                              load
I (138) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8c0e8h (573672)                                                                                                              map
I (256) esp_image: segment 4: paddr=000cc110 vaddr=40374964 size=0c950h ( 51536)                                                                                                              load
I (268) esp_image: segment 5: paddr=000d8a68 vaddr=600fe000 size=00010h (    16)                                                                                                              load
I (274) boot: Loaded app from partition at offset 0x10000
I (275) boot: Disabling RNG early entropy source...
W (277) bootloader_random: RNG for ESP32-S3 not currently supported
I (295) cpu_start: Pro cpu up.
I (295) cpu_start: Starting app cpu, entry point is 0x40375218
I (0) cpu_start: App cpu up.
I (311) cpu_start: Pro cpu start user code
I (312) cpu_start: cpu freq: 160000000
I (312) cpu_start: Application information:
I (316) cpu_start: Project name:     sensor_server
I (322) cpu_start: App version:      1
I (326) cpu_start: Compile time:     May 21 2021 21:43:08
I (332) cpu_start: ELF file SHA256:  8513e18041803023...
I (338) cpu_start: ESP-IDF:          HEAD-HASH-NOTFOUND
I (344) heap_init: Initializing. RAM available for dynamic allocation:
I (351) heap_init: At 3FC9BF98 len 00044068 (272 KiB): D/IRAM
I (358) heap_init: At 3FCE0000 len 0000F044 (60 KiB): STACK/DRAM
I (364) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (371) spi_flash: detected chip: gd
I (375) spi_flash: flash io: dio
W (379) spi_flash: Detected size(8192k) larger than the size in the binary image                                                                                                              header(2048k). Using the size in the binary image header.
I (393) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (403) EXAMPLE: Initializing...
I (433) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                                                                                             n: 0| Intr:0
I (433) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                                                                                             n: 0| Intr:0
I (443) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldow                                                                                                             n: 0| Intr:0
W (453) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (463) BTDM_INIT: BT controller compile version [7477a1a]
I (463) phy_init: phy_version 200,7b3a9df,Feb 25 2021
I (753) system_api: Base MAC address is not set
I (753) system_api: read default base MAC address from EFUSE
I (753) BTDM_INIT: Bluetooth MAC: 0x7c:df:a1:42:59:31

I (853) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
E (853) BT_HCI: CC evt: op=0x200b, status=0xc
E (853) BLE_MESH: start_le_scan, Invalid status 12
E (863) BLE_MESH: starting scan failed (err -1)
E (863) BT_HCI: CC evt: op=0x2008, status=0xc
E (873) BLE_MESH: set_adv_data, Invalid status 12
E (873) BLE_MESH: Failed to set adv data
E (883) BLE_MESH: Start advertising failed: err -1
W (883) BLE_MESH: Failed to send adv packet
E (893) BT_HCI: CC evt: op=0x2008, status=0xc
E (893) BLE_MESH: set_adv_data, Invalid status 12
E (903) BLE_MESH: Failed to set adv data
I (903) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (913) EXAMPLE: BLE Mesh sensor server initialized
E (5893) BT_HCI: CC evt: op=0x2008, status=0xc
E (5893) BLE_MESH: set_adv_data, Invalid status 12
E (5893) BLE_MESH: Failed to set adv data
E (5893) BLE_MESH: Start advertising failed: err -1
W (5903) BLE_MESH: Failed to send adv packet
E (5913) BT_HCI: CC evt: op=0x2008, status=0xc
E (5913) BLE_MESH: set_adv_data, Invalid status 12
E (5923) BLE_MESH: Failed to set adv data
E (10923) BT_HCI: CC evt: op=0x2008, status=0xc
E (10923) BLE_MESH: set_adv_data, Invalid status 12
E (10923) BLE_MESH: Failed to set adv data
E (10933) BLE_MESH: Start advertising failed: err -1
W (10933) BLE_MESH: Failed to send adv packet
E (10943) BT_HCI: CC evt: op=0x2008, status=0xc
E (10943) BLE_MESH: set_adv_data, Invalid status 12
E (10953) BLE_MESH: Failed to set adv data
E (15953) BT_HCI: CC evt: op=0x2008, status=0xc
E (15953) BLE_MESH: set_adv_data, Invalid status 12
E (15953) BLE_MESH: Failed to set adv data
E (15963) BLE_MESH: Start advertising failed: err -1
W (15963) BLE_MESH: Failed to send adv packet
E (15973) BT_HCI: CC evt: op=0x2008, status=0xc
E (15973) BLE_MESH: set_adv_data, Invalid status 12
E (15983) BLE_MESH: Failed to set adv data

nitinbhosaleemb, 2021-06-07:
is there any solution?

Campou, 2021-06-08:
Hi @nitinbhosaleemb 

Sorry for the late reply.

Please help to enable the following option in the menuconfig, i.e.
`idf.py menuconfig` -> `Component config` -> `Bluetooth` -> `Bluedroid Options` -> `Enable BLE 4.2 features`.

This should resolve the above issue.

Thanks.

nitinbhosaleemb, 2021-06-09:
resolved thanks!!

closed 2021-06-09 by nitinbhosaleemb