espressif/esp-idf #7062: ERROR: Platform MINGW64_NT-6.1-7601-x86_64 appears to be unsupported

yangshunhuai, 2021-05-23:
## Environment

- Development Kit:      [none]
- IDF version (run ``git describe --tags`` to find it):
    // v4.0.2
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32].
- Using an IDE?: [No]

## Problem Description
Trying to install esp-idf in MSYS2, it says that it is not supported
//Detailed problem description goes here.
### Expected Behavior
It will be installed properly.
### Actual Behavior
It shows: `ERROR: Platform MINGW64_NT-6.1-7601-x86_64 appears to be unsupported`
### Steps to reproduce
1. Open MSYS2 64-bit and go to the esp-idf folder.
2. Run install.sh
3. `ERROR: Platform MINGW64_NT-6.1-7601-x86_64 appears to be unsupported` come out.

yangshunhuai, 2021-05-23:
Well, after reading the docs carefully, I found that I should use the `MSYS2 32-bit`.
I tried for another time using `MSYS2 32-bit` in the Start menu, but I was greeted by another warning:
`ERROR: Platform MINGW32_NT-6.1-7601-x86_64 appears to be unsupported`.

dobairoland, 2021-05-23:
Hi @yangshunhuai. I'm sorry to tell you that the 64-bit version of MINGW was never (and is not planned to be) supported. MINGW32 is supported in ESP-IDF v3.3 and earlier and got deprecated in v4.0. I don't recommend to start to use it for new designs or if you are starting with ESP-IDF right now. 

Please read the [get started](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html) guide for setting up a supported environment. You can use Windows command line or PowerShell on Windows. The [Windows Tool installer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer) can install and set up everything.

yangshunhuai, 2021-05-24:
I was trying to compile a [lvgl for MicroPython](https://github.com/lvgl/lv_micropython) binary for an ESP32 module, and in its build guide, it says that if I want to build that, I should set up the toolchain and the ESP-IDF SDK. I referred to the MicroPython build guide and it told me to get the SDK by cloning the v4.0.2 version of ESP-IDF:`git clone -b v4.0.2 --recursive https://github.com/espressif/esp-idf.git`
Then I tried to install it in MSYS2, but failed.

Why don’t I install that by using the `install.bat`? Well, because I was compiling lvgl and it requires to use GNU Make:
```bash

make -C mpy-cross
make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC_SPIRAM deploy
```
But I can’t run GNU Make without MSYS2! Therefore, I have to use ESP-IDF under MSYS2.

Isn’t there a way to solve this problem?

If there is really no way to solve the problem, A Linux VM will work anyway.

dobairoland, 2021-05-24:
It looks like `lvgl` can be built with cmake (e.g. https://github.com/espressif/esp-dev-kits/tree/master/esp32-s2-hmi-devkit-1/components/lvgl) and Cmake support for Micropython has been added recently (https://github.com/micropython/micropython/releases/tag/v1.15). So I guess it just needs a little time until all "glued" together.

I don't think any action is required to be done in ESP-IDF to support this process.

yangshunhuai, 2021-05-24:
OK. There is nothing wrong with ESP-IDF.
Thank you for your suggestions @dobairoland .

closed 2021-05-24 by yangshunhuai